lvgl/lvgl #492: delete character / text?

epikao, 2018-10-28:
Hello

I can set a text, for example a number:
lv_label_set_text(txt, "45");

but how can I delete this text, number now?

thanks

kisvegabor, 2018-10-28:
Hi,

Just set the test again with an empty string `lv_label_set_text(txt, "");`

epikao, 2018-10-29:
hm, but that doesn't delete the previous set text, see:

```
void button1()
{
num++;

        lv_label_set_text(txt,"  ");
        lv_obj_set_pos(txt, 50,50);
        sprintf(buf, "#ff0000 %d", num);
        lv_label_set_text(txt, buf);
        lv_obj_set_pos(txt, 50,50);
}
```

epikao, 2018-10-29:
oh, I need to create second text object... but with "   " it doesn't delete the previous set characters

kisvegabor, 2018-10-30:
I don't understand. What do you expect to happen exactly? Please, write it down step by step.

kisvegabor, 2018-10-31:
`lv_label_set_text()` should change the text. So 
```c
lv_label_set_text(txt, "1");
lv_label_set_text(txt, "2");
```
should result `"2"`.

This simple example is working there?

turoksama, 2018-11-01:
@kisvegabor 
I think the problem is that ```num++; ``` isn't called periodically, because``` void button1()``` is initialized once.
You should set action function of that button, with ```num++;``` in that action function. So, when you press the button, it will be increased, and you'll see 2,3,4,,,etc.

epikao, 2018-11-02:
please see attached picture, which should show/explain better my problem:

[image]

epikao, 2018-11-02:
press button first time i want show 1, press button second time I want show 2 etc. so to show 2 properly I need to delete the 1 from before.... but how delete the 1?

embeddedt, 2018-11-02:
> oh, I think, I need to create second text object

If you simply create a *new* label every time, you will see overlapping text like this. Are you using the same label object?

embeddedt, 2018-11-02:
@epikao You **do not** need to create a new text object. Simply call `lv_label_set_text` on the old one.

epikao, 2018-11-02:
I'm not sure if somebody understand what I'm looking for?, see below my whole button code. I want simply count up a number be pressing a button....
so press first time button => 1 is show
press second time button => 2 is show, but behind, and that is my problem I still have the 1 ... and don't know how to delete this 1
press third time button => 3 is show, but behind I have 1 and 2... etc :-(

void intbutton1()
{  
   if ( (millis() - lastDebounceTime) > debouncedelay) {

        //background();
        Serial.print("buff1:");
        Serial.print(button1);
        Serial.print("\n");

        /*Create anew style*/
        static lv_style_t style_txt;
        lv_style_copy(&style_txt, &lv_style_plain);
        style_txt.text.letter_space = 2;
        style_txt.text.line_space = 1;
        style_txt.text.color = LV_COLOR_HEX(0x606060);
        lv_obj_t * txt = lv_label_create(lv_scr_act(), NULL);
        lv_obj_set_style(txt, &style_txt);
        lv_label_set_recolor(txt, 1);
        //lv_label_set_text(txt, "#ff0000 "SYMBOL_LEFT);

        num ++; //new Value

        lv_label_set_text(txt,"");
        sprintf(buf, "#ff0000 %d", num);
        lv_label_set_text(txt, buf);
        lv_obj_set_pos(txt, 50,50);
      
      lastDebounceTime = millis(); //set the current time 
    }//close if(time buffer)
}

embeddedt, 2018-11-02:
@epikao 

```c
lv_obj_t * txt = lv_label_create(lv_scr_act(), NULL);
```
Should not be called multiple times in the loop.

epikao, 2018-11-02:
> lv_obj_t * txt = lv_label_create(lv_scr_act(), NULL);

I already tried to move these parts outside, for example in the setup() function or global - but so far I know it makes problem with the arduino compiler... but I will try it asap again.

kisvegabor, 2018-11-04:
> I already tried to move these parts outside, for example in the setup() function or global - but so far I know it makes problem with the arduino compiler...

@epikao 
Please update `lvgl` from the `master` branch. There were updates to solve hard faults for some compiler.

If the issue still persists you should:
1. Declare `lv_obj_t * txt` as global variable
2. In `intbutton1`: `if(txt == NULL) txt = lv_label_create(lv_scr_act(), NULL);`. It will create a label only if ti not exists yet

It will create the label only once. Anyway, if you create multiple labels on top of each other, you will get multiple get multiple labels on top of each other. So don't create a new label every time.

epikao, 2018-11-04:
if I put code below in the setup function, I still have the same problem:

```
*Create anew style*/
        static lv_style_t style_txt;
        lv_style_copy(&style_txt, &lv_style_plain);
        style_txt.text.letter_space = 2;
        style_txt.text.line_space = 1;
        style_txt.text.color = LV_COLOR_HEX(0x606060);
        lv_obj_t * txt = lv_label_create(lv_scr_act(), NULL);
        lv_obj_set_style(txt, &style_txt);
        lv_label_set_recolor(txt, 1);
```

kisvegabor, 2018-11-04:
Ok, let's try this in your setup:
```c
lv_obj_t * test_txt = lv_label_create(lv_scr_act(), NULL);
lv_label_set_text(test_txt, "1");
lv_label_set_text(test_txt, "2");
```

And do nothing with labels in the loop.
Does it result `"2"`?

epikao, 2018-11-19:
so how I delete now this 2 and write a 3 at the same position?

embeddedt, 2018-11-19:
@epikao I don't think you understand how labels work. You can simply call `lv_label_set_text` again and it changes the value. **Do not** call `lv_label_create` more than once for the same label.

epikao, 2018-11-20:
>Do not call lv_label_create more than once for the same label.

unfortunately that does not work with arduino.. I get always an error if I don't call "lv_label_create" see error below. 
And if I do "lv_obj_t * test_txt = lv_label_create(lv_scr_act(), NULL);" global the program will be compiled but after load nothing happens (black display). I think arduino does not understand if do this function global.

And If I call for example in the setup follow:
lv_obj_t * test_txt = lv_label_create(lv_scr_act(), NULL);
lv_label_set_text(test_txt, "1");
lv_label_set_text(test_txt, "2");
than in the button function:
lv_obj_t * test_txt = lv_label_create(lv_scr_act(), NULL);
lv_label_set_text(test_txt, "3");

the functions does not delete the previous text... so it's simply overwritten and looks like my pictures above :-(

See the error without the create function in button function:

``````
lvgl_test:306: error: 'test_txt' was not declared in this scope
     lv_label_set_text(test_txt, "3");

                       ^

'test_txt' was not declared in this scope
```

epikao, 2018-11-21:
> Above setup:
> 
> lv_obj_t * my_label;
> int num = 0;

"lv_obj_t * my_label;" above setup makes the display not work, it stay's black, not start anything.

epikao, 2018-11-21:
I found that is not the global variable the problem, is more the function "lv_label_set..." (which I use in my button function):
```
        num ++; //new Value
        sprintf(buf, "#ff0000 %d", num);
        //lv_label_set_text(my_label, buf);
```
as soon I activate "lv_label..." I can only call the button function once and than the whole program crashs, freeze...

embeddedt, 2018-11-21:
@epikao Would you mind showing the whole file? It would be easier than trying to debug from several code snippets.

epikao, 2018-11-21:
see more or less the whole code:

`#include <lvgl.h>
#include <SPI.h>

const int LED = 13;
const int CS0 = 10;
const int DC = 9;
const int RES = 8;
const int BKL_PWM = 23;
const int TEST = 7;

//Touch settings
const int INT = 20;
const int RES2 = 21;
byte buffer[200];

//Button settings
const int button1 = 0;

volatile unsigned long lastDebounceTime = 0, debouncedelay = 300;

SPISettings settingsA(30000000, MSBFIRST, SPI_MODE0);
IntervalTimer tickTimer;

int num = 0;
char buf[32];
lv_color_t c_test[1000];
uint16_t i;
lv_obj_t * my_label;

void my_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t *color_array)
{
...
  lv_flush_ready();
}

void hal_init(void)
{
    ...
	tickTimer.begin(tick,1000); //
}

void writecommand(uint8_t a)
{....
}

void writedata(uint8_t a)
{...
}

void initST7789V()
{
  reset();
....
}

void reset(void)
{ ...
}

void setup() {
  #ifdef DEBUG
  Serial.begin(9600);
  #endif
  
  pinMode(LED,OUTPUT);
  pinMode (CS0, OUTPUT);
  digitalWrite(CS0, 1);
  pinMode (DC, OUTPUT);
  pinMode (RES, OUTPUT);
  pinMode (BKL_PWM, OUTPUT);
  digitalWrite(BKL_PWM, 1);
  pinMode (TEST, OUTPUT);

  pinMode(button1, INPUT_PULLUP);

  attachInterrupt(digitalPinToInterrupt(button1), intbutton1, FALLING);

  SPI.begin();
  SPI.beginTransaction(settingsA);
  initST7789V();
  delay(10);
  
 my_label = lv_label_create(lv_scr_act(), NULL);
  lv_init();
  
  hal_init();
  delay(10);

  LV_IMG_DECLARE(dmb1);
  LV_IMG_DECLARE(dmb2);
  LV_IMG_DECLARE(dmb3);

  lv_obj_t * img_src = lv_img_create(lv_scr_act(), NULL); /*Crate an image object*/
  lv_img_set_src(img_src, &dmb2);  /*Set the created file as image (a red fl  ower)*/
  lv_obj_set_pos(img_src, 0, 0);      /*Set the positions*/
  lv_obj_set_drag(img_src, true);

          /*Create anew style*/
        //static lv_style_t style_txt;
        //lv_style_copy(&style_txt, &lv_style_plain);
        //style_txt.text.letter_space = 2;
        //style_txt.text.line_space = 1;
        //style_txt.text.color = LV_COLOR_HEX(0x606060);
        //lv_obj_t * txt = lv_label_create(lv_scr_act(), NULL);
        //lv_obj_set_style(txt, &style_txt);
        //lv_label_set_recolor(txt, 1);
        //lv_obj_t * txt = lv_label_create(lv_scr_act(), NULL);
  /*
  lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
  lv_label_set_text(label, "Hello Arduino!");
  lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);
  */
//lv_obj_t * test_txt = lv_label_create(lv_scr_act(), NULL);
//lv_label_set_text(test_txt, "1");
//lv_label_set_text(test_txt, "2");

}
void intbutton1()
{  
    //lv_obj_t * test_txt = lv_label_create(lv_scr_act(), NULL);
    //lv_label_set_text(test_txt, "3");
   if ( (millis() - lastDebounceTime) > debouncedelay) {

        //background();
        Serial.print("buff1:");
        Serial.print(button1);
        Serial.print("\n");

        /*Create anew style*/
        /*
        static lv_style_t style_txt;
        lv_style_copy(&style_txt, &lv_style_plain);
        style_txt.text.letter_space = 2;
        style_txt.text.line_space = 1;
        style_txt.text.color = LV_COLOR_HEX(0x606060);
        lv_obj_t * txt = lv_label_create(lv_scr_act(), NULL);
        lv_obj_set_style(my_label, &style_txt);
        lv_label_set_recolor(my_label, 1);
        //lv_label_set_text(txt, "#ff0000 "SYMBOL_LEFT);
*/
        num ++; //new Value
        sprintf(buf, "#ff0000 %d", num);
        //lv_label_set_text(my_label, buf);
        //lv_obj_set_pos(my_label, 50,50);
      
      lastDebounceTime = millis(); //set the current time 
    }//close if(time buffer)
}

void tick()
{
  Serial.print("tick");
  Serial.print("\n");
  lv_tick_inc(1); //it crashs after several loops
}

void loop() {
  // put your main code here, to run repeatedly:

  lv_task_handler();
  /*If 'lv_tick_inc(5)' is not called in a Timer then call it here*/
  delay(5);             /*Wait a little*/
}`

embeddedt, 2018-11-21:
You can't create a label before you call `lv_init`!

epikao, 2018-11-21:
I think there are still some problems with Arduino, especially the tick function together with other function... don't know?? I use branch 5.2

embeddedt, 2018-11-21:
@epikao Is the tick function actually being called?

epikao, 2018-11-21:
yes (but only if lv_label is uncommented)

embeddedt, 2018-11-21:
@epikao Can you show `lv_conf.h`? I think something has changed/broken since you first opened this issue.

epikao, 2018-11-21:
[lv_conf.txt](https://github.com/littlevgl/lvgl/files/2605941/lv_conf.txt)

turoksama, 2018-11-22:
> It should work:
> 
> Above `setup`:
> 
> ```c
> lv_obj_t * my_label;
> int num = 0;
> ```
> 
> In `setup`
> 
> ```c
> my_label = lv_label_create(lv_scr_act(), NULL);
> ```
> 
> In `loop`:
> 
> ```c
> char buf[32];
> sprintf(buf, "num: %d\n", num);
> num++;
> lv_label_set_text(my_label, buf);
> ```

I think you should carefully read and check what kisvegabor said above. 
It should be helpful.

epikao, 2018-11-22:
> I think you should carefully read and check what kisvegabor said above.
> It should be helpful.

you can see my code above, it's exactly according.
`my_label = lv_label_create(lv_scr_act(), NULL);` I have put now after hal/lv init.
`char buf[32];` I have global.. but should not be the problem??

Also I have put the code "loop" in my button function... but that should not be a problem?!

kisvegabor, 2018-11-22:
You have this:

```c
my_label = lv_label_create(lv_scr_act(), NULL);
lv_init();
```

`lv_init` should be called first!

epikao, 2018-11-26:
> lv_init should be called first!

as mentioned, I already did that... but that still not works :-(  .. (I have modify my code above, so maybe somebody can check again, if there is still something wrong??)

Anyway does somebody already could do / test that with arduino / teensy?

embeddedt, 2018-11-27:
@epikao I would try creating a new project. In October it sounded like LittlevGL was running normally, but somehow something appears to have been broken between then and now.

xiaoloveyao, 2018-12-11:
What means of label? you can't change label casually. I advise you use lv_ta_create() function instead of lv_label_create() function... and the number will update without delete ex-number.

embeddedt, 2018-12-11:
@xiaoloveyao Actually a label can be changed by calling `lv_label_set_text` again.

kisvegabor, 2019-01-05:
I suppose the issue has solved finally so I close this issue.

closed 2019-01-05 by kisvegabor