lvgl/lvgl #493: image with teensy 3.2 / arduino

epikao, 2018-10-28:
Hello,

just wanted create again an image object (in a previous project from me it worked) but now maybe because of the new librarie 5.2?!, I got follow errors, i'm confused:

  LV_IMG_DECLARE(picture);

  lv_obj_t * img_src = lv_img_create(lv_scr_act(), NULL); /*Crate an image object*/
  lv_img_set_src(img_src, &picture);  /*Set the created file as image (a red fl  ower)*/
  lv_obj_set_pos(img_src, 0, 0);      /*Set the positions*/
  lv_obj_set_drag(img_src, true);

Compiler ERror:

C:\Users\xx\Documents\Arduino\libraries\lvgl\src\picture.c:988:7: error: unknown type name 'lv_img_t'

 const lv_img_t picture = {

       ^

etc.

kisvegabor, 2018-10-28:
The structure of the images is changed in `v5.2` therefore you need to generate the images again with the image converter tool.

epikao, 2018-10-29:
ok. thanks I didn't realize that

closed 2018-10-29 by epikao