lvgl/lvgl #496: how to use ttf font?

baidxi, 2018-10-29:
in my project need to display chinese font ,i want to use ttf font for chinese font

kisvegabor, 2018-10-30:
Just create a C array from the TTF font using: https://littlevgl.com/ttf-font-to-c-array

baidxi, 2018-11-03:
> Just create a C array from the TTF font using: https://littlevgl.com/ttf-font-to-c-array

no no no ,I'm want use freetype2 library for use ttf font,not create a C array for it.

embeddedt, 2018-11-03:
In that case you need to create a custom `lv_font_t` structure. See https://github.com/littlevgl/lvgl/issues/107#issuecomment-371821673.

embeddedt, 2018-11-12:
@baidxi Be warned, however, that this is pretty inefficient on a small embedded system. I don't know what the specs of your system are but I advise you to also look at #473 .

kisvegabor, 2018-11-14:
As #473 is very similar, let's continue the discussion there.

closed 2018-11-14 by kisvegabor