lvgl/lvgl #497: Suggestion - Whether line meter can support colors and size for a selected portion?

ajithpv, 2018-10-29:
To understand the idea in a better way, I'm using the below image:
[image]

I would like to know how to achieve this kind of feature using LittleVGL. I hope using the "[line meter widget](https://littlevgl.com/object-types/line-meter-lv_lmeter)" is close to the solution but, really don't know whether this supports the color and size adjustment to individual blocks or portion of the meter.

It would be good to add this feature to the LittleVGL to enhance the GUI for audio, automotive solutions etc.

kisvegabor, 2018-10-30:
Hi,

It looks very impressive! However, unfortunately, you can't do it with the line meter. Are you interested in adding this new object type?

ajithpv, 2018-11-02:
I must say that, I'm new to littlevgl. However, I will definitely give a try.

kisvegabor, 2018-11-04:
Great! :)

Feel free to ask if you stuck somewhere!