lvgl/lvgl #498: Missing function to reset points in a serie and to remove a serie from a chart

AloyseTech, 2018-10-29:
A function like lv_chart_remove_serie would be usefull to remove a serie from a chart. In the same idea, a function to remove the data from a serie would be useful, like lv_serie_remove_reset and/or lv_serie_remove_point( * serie, uint32_t pointIndex). At the moment the only way I found to reset the serie is to set every points to zero, but this lead to unwanted effects.

kisvegabor, 2018-10-30:
Hi,

It sounds reasonable! A PR is welcome.

> At the moment the only way I found to reset the serie is to set every points to zero, but this lead to unwanted effects.

What "unwanted effects" did you experience?

AloyseTech, 2018-10-30:
The unwanted effect is that the first line drawn between two point will start at 0. Which will appear as a step on the chart.

[image]

AloyseTech, 2018-10-30:
#507

kisvegabor, 2018-11-02:
As the PR is merged I close this issue!

Thank you very much for the contribution!

closed 2018-11-02 by kisvegabor