lvgl/lvgl #499: Disabling 'exit from edit' mode when group as only one object

AloyseTech, 2018-10-29:
Hi,

I'm experimenting with the touchles navigation. It seems that when the group has only one object I'm still able to go from edit to select mode. Would it be possible to stay in edit mode if there is only one object? For exemple I have a screen whith only one list as a menu. I force the edit mode at launch but if I long-press the encoder, the focus is not on the button items but on the list itself which is not desirable. 

I observe the same behavior when I have a messagebox with one button. I have freeze the focus on the button (previously I call 	lv_group_set_editing(group, true)), but if I long press, the focus goes to the messagebox itself even if there is no other object.

kisvegabor, 2018-10-31:
Hi,

it sounds reasonable to me. Would you like to implement it?

kisvegabor, 2018-11-21:
As the PR is merged I close this issue.
Thank you for this improvement!

closed 2018-11-21 by kisvegabor