lvgl/lvgl #500: Double buffering

Masi2017, 2018-10-29:
Hi All,
To speed up drawing on my LCD I am gonna use double buffering. I am not sure what needs to be done in addition to set the LV_VDB_DOUBLE  to 1? where "lv_flush_ready" should be called? I am gonna use DMA and once data is sent to LCD completely call  "lv_flush_ready" , is that correct or I need to do more?

classic-gentleman, 2018-10-30:
Please note that double buffering per se does not speed up anything. It does however allow you to draw another buffer if the DMA engine of your MCU can be made to work as a background task. Otherwise it is simply freeing up CPU time by doing a potentially faster DMA transfer instead of some form of programmed I/O.

So yes, you are basically correct, but it's not magical. DMA can be faster than PIO (not necessarily!), and depending on the case interleaving the rendering and the flushing is the actual trick. Is your MCU running any form of RTOS? Does it offer asynchronous callbacks for DMA transfers?

Masi2017, 2018-10-30:
@classic-gentleman Thanks for responding. My MCU is ESP32 and it is running on free RTOS. In TFT_FLUSH I'll copy the VDB buffer content to DMA and once data is transferred in DMA callback I should call lv_flush_ready?

classic-gentleman, 2018-10-30:
Right. So you're using FreeRTOS. Is your display connected by SPI? You may set a tasks that waits for a xQueue for a ready buffer and shuttles it to a task that uses the SPI master driver to flush the buffer.

That piece of code you quoted means the LV task is busy waiting for ANY of the frame buffers to transition to the flushed state (and therefore is free for reuse).

May I ask which display you're using? Because it might just be that the work is already done for you on the ESP IOT Solution repository. It's less work for you to perhaps port your driver according to the template they already laid rather than rolling out your own code.

Masi2017, 2018-10-30:
That would be great! I am using ER-TFT035-6, lcd driver is ILI9488 . I'm using it in 8080 , 8 bits parallel mode and I am gonna use I2S DMA.

classic-gentleman, 2018-10-31:
Bad news is that the sample I2S LCD driver of ESP-IOT-Solution doesn't have a callback for the transfer end. Good news is that you can probably hack the component yourself since the source is open. Weird news is that it's an ISR, so a little care has to be taken... You can probably forget all of that because you can transfer a full screen very quickly if you set your I2S to 20MHz and as lvgl uses partial updates you can get away easily with synchronous DMA on the I2S. Or you might experiment with using one of the input only pins as an interrupt receiving the TE signal and use it as an equivalent of the DMA end callback. I haven't explored this possibility much yet.

(Dirty secret: I have the same screen as you. Last Friday I finished porting to the display to eIOTs and am in the process of porting the FT6236 cap-touch controller too, ETA is tomorrow. So perhaps we should exchange notes in the near future :smile:)

Masi2017, 2018-10-31:
@classic-gentleman Thanks for all the information. I've already ported FT6236 and I can read registers over I2C also get touch point but I am not able to get gesture ID – it always returns 0 , maybe we could have some discussion on that once you port it. I'm happy to hear that you are using the same LCD and MCU, It's really good to share the information . keep in touch.

classic-gentleman, 2018-11-19:
> @classic-gentleman Thanks for all the information. I've already ported FT6236 and I can read registers over I2C also get touch point but I am not able to get gesture ID – it always returns 0 , maybe [...]

Hey @Masi2017, what's up? I still haven't been able to configure the I2S device so I resorted to bit-banging (and would love some hand here). But the FT6236 works just okay. What field/registers are you using for gesture ID? I suspect you're doing more work than necessary, especially considering you probably won't use multitouch with LittlevGL. I'm pasting part of my code here so you can cross-check with yours.

```
#define FT6236_MAX_TOUCH_POINTS 1

typedef enum {
    FT6236_EVENT_PRESS_DOWN = 0b00,
    FT6236_EVENT_LIFT_UP    = 0b01,
    FT6236_EVENT_CONTACT    = 0b10,
    FT6236_EVENT_NO_EVENT   = 0b11
} ft6236_event_t;

#pragma pack(push, 1)

struct ft6236_touchpoint {
	uint8_t xhi    : 4;
	uint8_t        : 2;
	ft6236_event_t event : 2;
	uint8_t xlo    : 8;
	uint8_t yhi    : 4;
	uint8_t id     : 4;
	uint8_t ylo    : 8;
	uint8_t weight : 8;
	uint8_t        : 4;
	uint8_t area   : 4;
};

/* This packet represents the register map as read from offset 0 */
typedef struct ft6236_packet {
	uint8_t          : 4;
	uint8_t dev_mode : 3;
	uint8_t          : 1;
	uint8_t gest_id  : 8;
	uint8_t touches  : 4;
	uint8_t          : 4;
	struct ft6236_touchpoint points[FT6236_MAX_TOUCH_POINTS];
} ft6236_packet_t;

#pragma pack(pop)

bool ex_tp_read(lv_indev_data_t *data)
{
	static lv_coord_t x = 0xFFFF, y = 0xFFFF;
	ft6236_packet_t packet;

	data->state = LV_INDEV_STATE_REL;
	data->point.x = x;
	data->point.y = y;

	iot_ft6236_read(dev, 0, sizeof(packet), (void *)&packet);

	if (packet.touches > 0 && packet.points[0].id != 0x0F) {
		data->point.x = (packet.points[0].xhi << 8) | packet.points[0].xlo;
		data->point.y = (packet.points[0].yhi << 8) | packet.points[0].ylo;

		if (packet.points[0].event == FT6236_EVENT_PRESS_DOWN || packet.points[0].event == FT6236_EVENT_CONTACT) {
			data->state = LV_INDEV_STATE_PR;
		}

		x = data->point.x;
		y = data->point.y;
	}

	return false;
}
```

Masi2017, 2018-11-27:
Hi @classic-gentleman , Thanks for the code, Yes you're right , technically I don't need to use multi touch with lvgl. I just wanted to make sure if this works in case in future I need to do some stuff on my own ( like zoom in and out ) . I am reading Gest_ID ( address 0x01) , but always it is 0 even for example if I scroll down or up.
About the I2s , one of my colleague has done some work on that and unfortunately I haven't had the chance to have a look on that part of the code , just using it.

kisvegabor, 2018-11-28:
Yes, LittlevGL doesn't support multi touch because there is nothing to use it for. Zoom and rotate could be a good purpose but they are not implemented.

classic-gentleman, 2018-12-11:
Hi @Masi2017,

I'm polling the device (in other words, using the regular `bool tp_read(lv_indev_data_t *data)` function). I can easily see the pair of touch points but could never get a gesture ID other than `0x0`. From what I could gather (from the Chinese Android tablets community, where this touch controller is often used), this is a firmware feature that depends on the panel vendor (probably calibration parameters) and it's likely the default firmware is devoid of it...

Masi2017, 2018-12-18:
Hi @classic-gentleman Thanks , yes gesture ID always is zero and as you have mentioned , this could be a firmware feature.

closed 2019-01-17 by stale[bot]

mvturnho, 2020-03-18:
> Hey @Masi2017, what's up? I still haven't been able to configure the I2S device so I resorted to bit-banging (and would love some hand here). **_But the FT6236 works just okay_**. What [...]

@classic-gentleman Hi, I am working on a lvgl project and have the capactive touch display. Could you help me out with the driver for this FT6236 device?

Sorry for using this issue, but I would not know how to contact you. Thanks so much