lvgl/lvgl #501: bug with VDB allocation

ImmortalBuka, 2018-10-30:
it impossible to define LV_VDB_ADR as variable address.

embeddedt, 2018-10-30:
See https://github.com/littlevgl/lvgl/blob/04239f92872cfec584af548633f7092488eb8112/lv_conf_templ.h#L47

> LV_VDB_ADR_INV: to replace it later with `lv_vdb_set_adr()`

embeddedt, 2018-10-30:
You can then use `lv_vdb_set_adr` to set the address as you wish. However I doubt that the VDB address can be changed while the GUI is running.

ImmortalBuka, 2018-10-30:
thanks.
and one more question
when i tried to place work_mem_int into external RAM, it caused to crash system (stm32f429), but work with internal RAM
what i do change? (LV_MEM_ATTR i'd set __attribute__ ((section (".sdram"))) )

kisvegabor, 2018-11-04:
`LV_MEM_ATTR` is used only by the array of `lv_mem_alloc` to allocate memory in it.

Try to add `attribute ((section (".sdram"))) )` in `lv_vdb.c` dircetly where `vdb_buf` is declared.

ImmortalBuka, 2018-11-06:
but in this case lv_mem_alloc returned wrong address (after 5 or 6 execute times).

kisvegabor, 2018-11-07:
Sorry, I misread it. 
So you would like to move the work memory to SRAM (which is used by `lv_mem_alloc`). I understand now. In this case you really should use `LV_MEM_ATTR`.

To place the work memory to a specific address you need to modify `lv_mem.c`: https://github.com/littlevgl/lvgl/blob/master/lv_misc/lv_mem.c#L82

Replace this line with `work_mem = (uint8_t*)SDRAM_ADDRESS_TO_USE`
You can delete the declaration of `work_mem_int` array.

I didn't test it but is should work.

ImmortalBuka, 2018-11-09:
It works, thx for help)

closed 2018-11-09 by ImmortalBuka