lvgl/lvgl #509: Disabling cursor on text area when focused but not editable (set active only in edit mode)

AloyseTech, 2018-10-31:
Hi,

I think the cursor should be hidden when a Text Area is focused in a group but not yet selected for edition.

One might consider the text area editable as soon as the cursor is active and blinking.

Currently the cursor is visible both in focused and edition mode.
The signal LV_SIGNAL_FOCUS should not enable the cursor. The cursor should be set to visible when the signal LV_SIGNAL_GET_EDITABLE is caught. It seems, however, that there is no signal indicating the end of edition that could be used to disable the cursor...

kisvegabor, 2018-11-01:
Hi,

> It seems, however, that there is no signal indicating the end of edition that could be used to disable the cursor...

`LV_SIGNAL_FOCUS` is sent when the edit mode has changed too. See `lv_group_set_editing`. So in the FOCUS signal we can check the edit state to decide what to do with the cursor.

But what if a virtual keyboard (`lv_kb`) is writing to the Text area. In this case, the keyboard is in focus so cursor won't be displayed. Now, `lv_kb` clears the `LV_CURSOR_HIDDEN` flag to show the otherwise hidden cursor.

AloyseTech, 2018-11-01:
Do you plan tu add the edit flag in each editable object? We could maybe imagine a way to let the text area “know” when it is edited but not in focus.

kisvegabor, 2018-11-04:
I think the edit flag is not required because we can do something similar like the in `lv_ddlist`: https://github.com/littlevgl/lvgl/blob/master/lv_objx/lv_ddlist.c#L528

AloyseTech, 2018-11-04:
The same solution as in the ddlist could be used, indeed. Do you plan to implement this functionality?

kisvegabor, 2018-11-05:
Great! I'll add it in a few days.

kisvegabor, 2018-11-09:
Hi,

I added this feature: https://github.com/littlevgl/lvgl/commit/2a053b86bfe8726a2055ce38e4b3a1cbb6933a0e

AloyseTech, 2018-11-09:
I tested it with my GUI but it does not work as I expected. Currently I have a button "done" to stop editing a spinbox. When it is clicked I call `lv_group_remove_obj(spinbox);` to remove it from my single object group. It seems that LV_SIGNAL_FOCUS is not sent at this point and so the cursor stays visible. The style goes back to normal though.

kisvegabor, 2018-11-09:
The spinbox is the only object in the group?

AloyseTech, 2018-11-10:
Yes it is.

embeddedt, 2018-11-11:
I assume this issue is resolved so I will close it. If you continue to experience problems feel free to reopen the issue.

closed 2018-11-11 by embeddedt