lvgl/lvgl #510: constant buffer size in flushing

Masi2017, 2018-10-31:
Hi All,
I was wondering if it is possible to have lvgl output a full buffer each time (LV_VDB_SIZE) and not allow arbitrary smaller sizes?
I am trying to use DMA and I can do a quick integration if I can keep the buffer size the same between flushes. Essentially I don't want to resize the dma descriptor between flushes and keep it consistant

embeddedt, 2018-10-31:
Maybe #466 could help you.

kisvegabor, 2018-10-31:
Hi,

It doesn't sound reasonable render the full buffer if, let's say, only one pixel has changed. 

Resizing the DMA descriptor really takes so much time? 

Ideas:
- With `LV_VDB_DOUBLE` you can use the buffer directly in `disp_flush` directly for DMA. 
- If you need the VDB to be placed on a special address for DMA, see: #501

Masi2017, 2018-11-01:
Actually there is no big difference on writing one pixel or one row using DMA ( I am using 8bits I2S DMA). It's done in almost same amount of time. I set the VDB_size to 32*480 and I would like to render the full buffer in each flashing. I added a rounder callback function but that only works with full screen but I don't want to redraw the full screen. Is there any way to make sure that always area is equal to VDB_SIZE?

Masi2017, 2018-11-01:
We are using 8bits I2S DMA and there isn't a significant difference in speed to write one pixel or 2 rows. I added a round callback function as you mentioned in #466 but we don't want to render the whole screen . We want to restrict the rendering to a VDB_SIZE. Basically we want to update always exactly one VDB amount with full screen width. Is there any function that we can put this restriction?

kisvegabor, 2018-11-02:
Let's assume you have `LV_VDB_SIZE   (LV_HOR_RES * LV_VER_RES / 4)`
Then your `rounder` function can look like:
```c
    a->x1 = 0;
    a->x2 = LV_HOR_RES - 1;

    uint32_t vdb_rows = LV_VDB_SIZE / LV_HOR_RES;

    /*y1: Round down to 0, 1/4, 2/4 or 3/4 LV_VER_RES*/
    a->y1 = a->y1 / vdb_rows;
    a->y1 = a->y1 * vdb_rows;

    /*y2: Round up to `1 * LV_VER_RES / 4 - 1` or `2 * LV_VER_RES / 4 - 1` or `3 * LV_VER_RES / 4 - 1` or `4 * LV_VER_RES / 4 - 1`*/
    a->y2 = a->y2 / vdb_rows;
    a->y2 = a->y2 * vdb_rows;
    a->y2 += vdb_rows - 1;
```

Please note that because of the divisions it might be slow so some optimization should be done (shift instead of divide if applicable)

embeddedt, 2018-11-02:
@kisvegabor I would recommend replacing `vdb_rows` with a macro. That way most compilers will optimize the division to a shift automatically if it is possible.

closed 2018-11-02 by Masi2017