lvgl/lvgl #511: [Question] Can make a circular list?

robertvxt, 2018-10-31:
As I have a circular list of options (A,BC,D): D <-> A <-> B <-> C <-> D <-> A
With help from lv_roller, I created a list: A <-> B <-> C <-> D but seems it does not support for circular

Please consult me...

Thanks,
Robert

kisvegabor, 2018-11-02:
Hi,

I made some experiments to see how complicated it is to improve lv_roller this way. Unfortunately, I found only odd and buggy "solutions". So I'd rather say that lv_roller doesn't support circular scrolling. :(

embeddedt, 2018-11-02:
This is kind of a hack, but maybe this would work (I haven't tried):

Your current list is: A, B, C, D

1. Make blank items on both ends of the list.
2. When the roller detects that the blank item is selected, it jumps to the opposite end of the list.

BrianPugh, 2018-11-02:
Here's a hacky solution that I'm currently using:

https://github.com/joltwallet/jolt_gui/blob/esp32/jolt_gui_entry.c#L195

Basically my roller contains the numbers 0-9 three times, and whenever i reach a certain index, I reset the position after the animation takes place. So its a hacky infinite loop that works for me.

kisvegabor, 2018-11-04:
@embeddedt  I think it won't work properly when you drag fast and a big "drag throw" is applied. In this case, the roller with e.g. 4 items can turn around multiple times. 

@BrianPugh It seems a good solution if you use only buttons a to navigate. I found more difficulties with the handling of animations and dragging.

embeddedt, 2018-11-04:
@kisvegabor I agree that turnaround could occur multiple times. However, could this be desired behavior?

kisvegabor, 2018-11-06:
> I agree that turnaround could occur multiple times. However, could this be desired behavior?

I thought the intended behavior is to make the roller "infinite". So the support of multiple turnarounds is critictal. Or?

embeddedt, 2018-11-06:
@kisvegabor Exactly.

>  I think it won't work properly when you drag fast and a big "drag throw" is applied. In this case, the roller with e.g. 4 items can turn around multiple times.

Why wouldn't it work properly if the roller is supposed to be infinite? I think I'm not understanding something.

kisvegabor, 2018-11-09:
@embeddedt  The roller is like a page with a scrollable part which is not infinite. With a big drag, you can easily reach the top/bottom position. So the roller needs to reposition itself while the drag happens to look like infinite. And this reposition isn't worked well for me.

robertvxt, 2018-11-18:
Hi,
Thank you all for your attention, I was busy in other stuffs.

I based on @BrianPugh 's idea then it worked properly and of course, I just used buttons to navigation.

embeddedt, 2018-11-18:
@robertvxt Great to hear that it's working! If the issue is resolved please close it.

robertvxt, 2018-11-18:
BTW, @embeddedt , I saw for the first display of roller, there are only 2 options which be displayed while 
the visible row count was set upper than 2. i.e. lv_roller_set_visible_row_count(roller, 3);

Can we change to always display row count per setting lv_roller_set_visible_row_count?

embeddedt, 2018-11-18:
@robertvxt If you adjust the roller size yourself (by disabling `lv_roller_set_hor_fit` and changing the height) I think this behavior might occur. Can you send a code sample so I can reproduce this?

robertvxt, 2018-11-18:
@embeddedt , we can see easily in https://github.com/littlevgl/lvgl/wiki/Roller

embeddedt, 2018-11-18:
@robertvxt Whoops, my mistake there. I don't know how I thought that `hor_fit` was related to height. I'll test the roller issue now.

embeddedt, 2018-11-18:
@robertvxt I tested the roller in the `pc_simulator` and it worked properly. I don't think I can find the issue without more details.

robertvxt, 2018-11-19:
@embeddedt , Please refer to the attachment, there are only 2 rows:
[image]
-> only "Apple" and "Broccoli" displayed. So my question is: can we display 3 row in the initial (first) display of roller (i.e. "**Apple**" and "**Broccoli**" and "**Cabbage**" displayed). I'm not sure this is bug, just clarify the use case!

kisvegabor, 2018-11-19:
@robertvxt As you can see the height of the roller is set to display 3 rows. But the top row is empty because the roller is not circular and "Apple" is the first element.

To see 3 rows of options you need to select the second element. `lv_roller_set_selected(roller, 1, anim_en)`

embeddedt, 2018-11-19:
@robertvxt To build on what @kisvegabor said, I believe the selected element will always be in the middle, so it's not possible to see three options and have the first or last element selected.

robertvxt, 2018-11-19:
Agree, let it be a limitation at this use case
Thanks @kisvegabor  & @embeddedt

closed 2018-11-19 by robertvxt