lvgl/lvgl #512: Use RA8876's SDRAM as frame buffer

techtoys, 2018-10-31:
I am using RA8876 for large screen @1280*720 with ESP32. Preliminary Littlevgl port is working.
Tried LV_VDB_SIZE set to (10*1280) with custom_flush() for whole page, and LV_VDB_SIZE set 0 with custom_fill() by bitblit transfer offered by RA8876 and it is working too.

To enhance its feature I wish to make use of the large 256Mbit SDRAM as frame buffer but I am not sure where to start from lv_conf.h. This SDRAM is wired to RA8876 instead of ESP32 which means256Mbit memory is not addressable from ESP32. Neverthess, it is always possible to use off-screen area from SDRAM as buffer and copy blocks of images by various 2D functions available from RA8876.

The question is, how to accommodate such external SDRAM addressing not in ESP32 space in Littlevgl?
Is it possible to use custom malloc and free for that. I do have something like my_malloc and my_free for this purpose but I don't know where to implement it in lv_conf.h.

embeddedt, 2018-10-31:
I don't think LVGL supports indirect addressing in the manner you've described. However, maybe the GPU interface could help you.

turoksama, 2018-11-01:
I don't think it is a good idea to do so but of course as a workaround :)

embeddedt, 2018-11-01:
I think @kisvegabor is right. Indirectly accessing the SDRAM will probably be too slow in this case if you have to do it 1 pixel at a time.

embeddedt, 2018-11-01:
(I assume, of course, you communicate with this display controller using SPI, I2C, etc)

techtoys, 2018-11-01:
No, if it is new graphic asset of course, pixel-by-pixel rendering is required. However, if it is some sort of preloaded assets to serial flash (interfaced to RA8876 too), there are three possible ways:
1. Copy preloaded assets from serial flash to frame buffer of RA8876 by DMA. This DMA is not the one from host mcu. It is DMA offered by RA8876.
2. Copy preloaded assets from serial flash to off-screen buffer of RA8876 by DMA. When copy is finished, data from one area in SDRAM copied to visible framebuffer area by 2D operation.
3. If data is not preloaded, it is also possible to render the asset in offscreen area (that is rendering in background). Once finished, a single 2D operation copy from offscreen to visible region.

RA8876 operation rely heavily on 2D operation for larger screen. It can output 1366x768 @ 60Hz max. For small mcu, even 1/10 of this screen means 70x1366x2bytes (191kB) is rather demanding. You can say : "get a bigger mcu then". If it is the case, we won't need external graphic controller at all.

This is the reason I asked about indirect memory addressing...

Previously I have implemented a custom malloc and free to allocate memory from SDRAM. It works OK but I don't know if it is possible to "integrate" it within Littlevgl. Will look into it more closely.

BTW, I find vendors are steering to external Serial Flash for asset storage. Not only RA8876, FTDI's BT81x now equipped with Serial Flash for asset storage too. It would be nice for Littlevgl to handle Serial Flash preloaded data.

kisvegabor, 2018-11-02:
I see and I have an idea how to use the preloaded data:
1. You still need to create objects in LittelvGL like lv_btn, lv_img etc.
2. Every object has a `design` function which can be changed by the user with `lv_obj_set_design_func()`
3. In your custom design function, you can instruct the RA8876 to load the button's data to the specified area. It could work for the images in the same way. 

Example design function:
```c
static bool my_btn_design(lv_obj_t * btn, const lv_area_t * mask, lv_design_mode_t mode)
{
    if(mode == LV_DESIGN_COVER_CHK) {
        return false;   /*Ot true is the button fully covers it's area i.e no holes, rounded corners etc.*/
    } else if(mode == LV_DESIGN_DRAW_MAIN) {
        switch(lv_btn_get_state(btn)) {
            case LV_BTN_STATE_REL:
                ra8876_draw_released_btn(&btn->coords);
                break;
            case LV_BTN_STATE_PR:
                ra8876_draw_pressed_btn(&btn->coords);
                 break;
            /*And so on*/
        }
    } else if(mode == LV_DESIGN_DRAW_POST) {
        /*used for special things, probably not required*/
    }

    return true;
}
``` 

In this case of course you should set `LV_VDB_SIZE` to `0`.

techtoys, 2018-11-05:
Thank you Gabor. It is working like overriding the default lv_obj_set_design_func() and it works. In this case it is possible to call native function of Ra8876 outside Littlevgl framework for anything. Will put the code in this forum later.

closed 2018-11-05 by techtoys

kisvegabor, 2018-11-05:
Awesome, thank you!