lvgl/lvgl #514: Suggestion-should add text in the button and imgbtn?

fallstool, 2018-11-01:
now if you want to add some text in the button, you must add a label and then set the text, 
shall we add a label in the button, and just use the function lv_btn_set_text(btn, "some text")?

embeddedt, 2018-11-01:
@fallstool I think the current method was selected because it allows for the most flexibility. For example, in my application, I create a transparent button and then create an image on top of it, giving the impression of an image button. (I am aware of `lv_imgbtn` but I decided not to use it for several reasons)

fallstool, 2018-11-01:
@embeddedt I think the text will show only you call the function. if you did not set the text, the button just same as the current version. In my application, so many button with text setted.

embeddedt, 2018-11-01:
@fallstool A wrapper function is probably the cleanest method then.

closed 2018-11-02 by fallstool

kisvegabor, 2018-11-04:
I was considering a lot to add this feature to the buttons because it's very common to add text to them.

My thoughts: 
1. If a text would be simply drawn on the button some features, like `auto fit` of the button, won't work.
2. It will consume `sizeof(char*)` RAM even if you don't use this feature.
3. However, if a text should be displayed on a fix-sized button it would save the RAM of a label.  

According to this, I think the best would be to create a new object inherited from  `lv_btn` which has only one extra attribute, the text. It could be called: `lv_btnt`or similar.

If you are interested in adding this new object type feel free to re-open this issue!