lvgl/lvgl #516: LV_COLOR_DEPTH problem

harteware07, 2018-11-02:
I updated lvgl yesterday to master and now i have a problem with LV_COLOR_DEPTH in 32 Bit mode. Before I was using 24 Bit color depth. I attach 2 pictures for you to see the difference between 16 and 32 bit color. it seems like it is drawn much too big in 32 bit mode.

Any ideas what i could try? 
[image]
[image]

embeddedt, 2018-11-02:
@harteware07 Looks like you're using STM32F746G-DISCO from the picture. I ran into the same problem. The display driver and framebuffer array also need to be adjusted to accommodate other color depths.

kisvegabor, 2018-11-04:
I agree with @embeddedt . It seems your display driver needs to be reconfigured. If you are using the [example from lv_projects repo](https://github.com/littlevgl/lv_projects/tree/master/stm32f746_discovery_no_os) than check [this function](https://github.com/littlevgl/lv_projects/blob/master/stm32f746_discovery_no_os/hal_stm_lvgl/tft/tft.c#L461)

As you noticed `LV_COLOR_DEPTH 24` was replaced by `32`. Maybe you driver uses this define and waits for `24`?

harteware07, 2018-11-14:
i changed LV_COLOR_DEPTH_24 to LV_COLOR_DEPTH_32 in order to be able to compile, other than that i didnt change anything. 
Could you please give me a hint on what else to change?

embeddedt, 2018-12-11:
@harteware07  There was no activity here for 1 month (plus you appear to have succeeded in getting it working on the discovery board based on your comment in #615) so this issue has been closed. If you continue to have problems please reopen the issue.

closed 2018-12-11 by embeddedt