lvgl/lvgl #517: list callback is not called on touch

Masi2017, 2018-11-02:
Hi All,

I've created a list and added four btns. It used to works fine but since I've updated my lvgl library from ver5.2beta to ver5.2 master, when I touched the brns in the list, action function is not called while it works fine with hardware keypad, do you have any ideas? here is my function to create a list:

lv_obj_t * createOptList(lv_obj_t *parent,const char *txt[], lv_action_t action) {

  uint8_t txtItemNum=findNumItemsPointArr(txt);

  static lv_style_t style_btn_scrl;
  lv_style_copy(&style_btn_scrl, &lv_style_transp_tight);
  style_btn_scrl.body.padding.inner = 2;

  static lv_style_t style_btn_pr;
  lv_style_copy(&style_btn_pr, &lv_style_btn_pr);
  style_btn_pr.body.border.opa = LV_OPA_COVER;
  style_btn_pr.body.border.color = LV_COLOR_ORANGE;
  style_btn_pr.body.border.width = 5;
  style_btn_pr.line.color = LV_COLOR_ORANGE;

  /*Crate the list*/
  lv_obj_t *list2 = lv_list_create(parent, NULL);

//  lv_obj_set_width(list2,LV_HOR_RES-210);
  lv_obj_set_height(list2,txtItemNum*LIST_BTN_HIGH);
  lv_list_set_style(list2, LV_LIST_STYLE_SCRL, &style_btn_scrl);
  lv_list_set_style(list2, LV_LIST_STYLE_BTN_PR, &style_btn_pr);

  for (uint8_t i=0 ; i< txtItemNum ; i++)
  {
    lv_obj_set_free_num(lv_list_add(list2, NULL ,txt[i], action),i);
  }

//  lv_obj_align(list2, NULL, LV_ALIGN_CENTER, 0,(lv_obj_get_height(parent)>(LIST_BTN_HIGH*txtItemNum)? (lv_obj_get_height(parent)-LIST_BTN_HIGH*txtItemNum)/2:5));
//  lv_obj_align(list2, NULL, LV_ALIGN_IN_TOP_LEFT, 5,10);
  lv_obj_align(list2, NULL, LV_ALIGN_CENTER, 0,0);

  listSelIndex=0;

  return list2;
}

here is where I call the above function:

    lv_indev_set_group(kb_indev, activeGroup);

    InboxScreen=lv_obj_create(NULL, NULL);
    lv_scr_load(InboxScreen);

    parent=createBgWindow("",InboxScreen);

    current_obj=createOptList(parent,settingOptStr,settingOptCb);

    lv_obj_set_width(current_obj,LV_HOR_RES-210);
    lv_group_add_obj(activeGroup, current_obj);
    lv_group_focus_obj(current_obj);

    lv_group_set_focus_cb(activeGroup,settingMenuCb);

Masi2017, 2018-11-02:
I added the code below after adding the btn to the list and problem resolved
lv_btn_set_action(btn, LV_BTN_ACTION_PR, action);

I should add that I have created some single btns with CLICK_ACTION and CLICK_ACTION works for them but seems that doesn't work for list btns.

mharizanov, 2018-11-04:
I had similar issue, check my solution: https://github.com/littlevgl/lvgl/issues/521

Masi2017, 2018-11-05:
@mharizanov Thanks , yes I modified my touch driver to return x and y in release and fixed the issue.

closed 2018-11-05 by Masi2017