lvgl/lvgl #518: lv_obj_set_pos() has no effect(pos Y) under tabs(tab_view).

turoksama, 2018-11-03:
Hi there
     I created a tabview in a page(LAYOUT_OFF) as follow:
```
lv_obj_t *page = lv_page_create(scr, NULL);
lv_obj_set_style(page , &t_mfx_page_style);
lv_obj_set_click(page , false);
lv_obj_set_pos(page 0,0);
lv_obj_set_size(page ,600,400);
lv_cont_set_layout(lv_page_get_scrl(page ), LV_LAYOUT_OFF);

lv_obj_t *tbv = lv_tabview_create(page, NULL);
lv_obj_set_size(tbv, 600,400);
lv_obj_set_pos(tbv,0,0);
lv_obj_t *tab1 = lv_tabview_add_tab(tbv, "tab1");
lv_obj_t *tab2 = lv_tabview_add_tab(tbv, "tab2");
lv_tabview_set_sliding(tbv, false);
lv_tabview_set_anim_time(tbv, 0);

lv_obj_t * mc_label = lv_label_create(tab1, NULL);
lv_obj_set_size(mc_label, 20,300);   //has no effect.
lv_obj_set_pos(mc_label,10,100);    //Y offset has no effect.

```
[image]
Hmm... weird.

turoksama, 2018-11-03:
Found:
```
lv_page_set_scrl_fit(tab1 , false, false);
lv_page_set_scrl_height(tab1 , 400);   //Set height of the scrollable part of a page
```
And it works.
Well, looking deep into the library is really necessary.

closed 2018-11-03 by turoksama

kisvegabor, 2018-11-04:
FYI: https://github.com/littlevgl/lvgl/wiki/Page#notes

turoksama, 2018-11-05:
Thanks a lot.