lvgl/lvgl #519: [Question]label align and size

fallstool, 2018-11-03:
Hi:
not current label version, if you set the size and the set the mode, and then set the align, LV_LABEL_ALIGN_LEFT LV_LABEL_ALIGN_CENTER, LV_LABEL_ALIGN_RIGHT only in the top of the area of the label. 
enum {
    LV_LABEL_ALIGN_LEFT,
    LV_LABEL_ALIGN_CENTER,
    LV_LABEL_ALIGN_RIGHT,
};
Here is the test Code
```c
 lv_obj_t * label1 = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_long_mode(label1, LV_LABEL_LONG_DOT);
    lv_obj_set_size(label1, 300, 42);
 	lv_label_set_align(label1, LV_LABEL_ALIGN_CENTER);
    lv_obj_set_style(label1, &lv_style_plain_color);    /*Set a background to clearly see the label size*/
    lv_label_set_body_draw(label1, true);
    lv_label_set_text(label1, "Settings");
    lv_obj_set_pos(label1, 44, 42);
    //the height is not what setted.
	lv_obj_t * label2= lv_label_create(lv_scr_act(), NULL);
  	lv_label_set_long_mode(label2, LV_LABEL_LONG_BREAK);
   	lv_obj_set_size(label2, 300, 50);
   	lv_label_set_align(label2, LV_LABEL_ALIGN_CENTER);
	lv_obj_set_style(label2, &lv_style_plain_color);    /*Set a background to clearly see the label size*/
 	lv_label_set_body_draw(label2, true);
 	lv_label_set_text(label2, "Settings");
	lv_obj_set_pos(label2, 44, 112);
```
should add the the ver align?
enum {
    LV_LABEL_ALIGN_LEFT,
    LV_LABEL_ALIGN_HCENTER,
    LV_LABEL_ALIGN_RIGHT,
    LV_LABEL_ALIGN_TOP,
    LV_LABEL_ALIGN_VCENTER,
    LV_LABEL_ALIGN_BOTTOM,
};

turoksama, 2018-11-03:
Hi,
 first thing came into mind was "LV_LABEL_LONG_BREAK".

kisvegabor, 2018-11-04:
In the case of the `LV_LABEL_LONG_DOT`, you specify the max size of the label. If the text is too long then the parts out of the specified size will be replaced with dots. However, with `LV_LABEL_LONG_BREAK` you can set only the width of the label. The height will be set automatically according to the number of lines. 

The vertical-align is not reasonable if the height is set automatically but there are there are 3 LONG_MODES where the height can be set by the user:
- LV_LABEL_LONG_DOT
- LV_LABEL_LONG_ROLL
- LV_LABEL_LONG_CROP

For these LONG_MODEs the vertical-align would be really useful. I think `lv_label_refr_text` would be a great place to set `ext->offset.y` if required.

fallstool, 2018-11-05:
hi
@kisvegabor in my project I add a new mode LV_LABEL_LONG_NONE, means just the label size what setted, and set the right the positon in the fuction lv_label_refr_text.
many thanks, you can close the issue.

closed 2018-11-05 by kisvegabor