lvgl/lvgl #520: How to modify the text in button when button action arrived?

YipCharles, 2018-11-04:
Hi, I create a label obj from a button obj, and it displays well.
Then, I add a button action function trying to change the text on it, how should I do?
(I had tried "lv_obj_get_child(btn, label);" but failed)

kisvegabor, 2018-11-04:
Hi,

To set an ID you can use `lv_obj_set_free_num(btn, 1 or -1)` and then in the action function `if(lv_obj_get_free_num(btn) < 0) {...} else {...}`

embeddedt, 2018-11-04:
I forgot about `lv_obj_set_free_num`. Much simpler!

turoksama, 2018-11-05:
@kisvegabor Hi, if it is a button matix object, how to set IDs for the matrix?

YipCharles, 2018-11-05:
@kisvegabor Yes, I just found that also.
Then, is that possible to get the text above the button?  That means getting the button's label's text.

kisvegabor, 2018-11-05:
@turoksama `lv_obj_set/get_free_num` works for every object type. But not for the buttons of the button matrix individually because they are not real, unique objects. They are just drawn.

@YipCharles 
Try:
```c
    lv_obj_t * lalel = lv_obj_get_child(btn, NULL);
    const char * txt = lv_label_get_text(label);
```

YipCharles, 2018-11-05:
@kisvegabor 
I mean... like I callback a button function, the only argument passed into is the button handler pointer"btn", can I just use this "btn" to get the text on it ( equal question to the "label's on it")?

closed 2018-11-06 by kisvegabor