lvgl/lvgl #521: Drop-down list not responding to touch

mharizanov, 2018-11-04:
Hello,
I have an issue with ddlist, it won't respond to touch. All other object types work normally, but not dd-list. I  use 5.2, but also tried 5.3 with the same result. I expect that tapping the touch on the dd-list will expand it, and I'll be able to select options using the touch, but none of that happens, it just stays static.

I test it with this basic code:

```
scr = lv_obj_create(NULL, NULL);
lv_scr_load(scr);

/*Create a drop down list*/
lv_obj_t * ddl1 = lv_ddlist_create(scr, NULL);
lv_ddlist_set_options(ddl1, "option 1\noption2\noption3");
lv_obj_align(ddl1, NULL, LV_ALIGN_IN_TOP_RIGHT, -70, 10);
```

I enabled logging to check the output, but there is nothing unusual:
```
Info: Screen create ready 	(...components/littlevgl/lvgl/lv_core/lv_obj.c #180)
Info: container created 	(...components/littlevgl/lvgl/lv_objx/lv_cont.c #108)
Info: container created 	(...components/littlevgl/lvgl/lv_objx/lv_cont.c #108)
Info: page created 	(...components/littlevgl/lvgl/lv_objx/lv_page.c #145)
Info: label created 	(...components/littlevgl/lvgl/lv_objx/lv_label.c #129)
Info: drop down list created 	(...components/littlevgl/lvgl/lv_objx/lv_ddlist.c #144)
```

What are my options to get to the bottom of this?

kisvegabor, 2018-11-04:
@embeddedt  
Yes, thank you. But please note that it is important only in case of touchpads. Using a mouse the released coordinate will be the same as the pressed coordinate naturally.

turoksama, 2018-11-09:
I have the same issue.
I change the return value from "return valid;" to "return false;" and it works fine...  :(

embeddedt, 2018-11-09:
@turoksama Please see my response here: https://github.com/littlevgl/lvgl/issues/537#issuecomment-436807155