lvgl/lvgl #523: STM32F769 Discovery port

kisvegabor, 2018-11-04:
Hi,

I'd like to try LittlevGL on STM32F769 Discovery. Have anybody ported LittlevGL to it?

kl-cruz, 2018-11-06:
It looks similar to STM32F469 DIscovery board, but with Cortex-M7 instead of Cortex-M4. Each board use MIPI DSI interface.
https://github.com/littlevgl/lv_projects/tree/master/stm32f469_discovery_no_os

kisvegabor, 2018-11-09:
Thank you for the info. I tried the STM32F469 but it didn't work out of the box. I will investigate deeper.
Hopefully, only some pins need to be remapped.

kl-cruz, 2018-11-09:
I will try to find some spare time soon. I have one STM32F469 discovery to work with. I think we should check if CubeMX is different between F769 and F469

seyyah, 2018-11-09:
We ported also STM32F767 with CubeMX+Atollic.

seyyah, 2018-11-09:
We'll add these ports (STM32F767, STM32F746 with CubeMX+Atollic) as soon as possible.

kisvegabor, 2018-11-10:
@kl-cruz 
> I will try to find some spare time soon. I have one STM32F469 discovery to work with. I think we should check if CubeMX is different between F769 and F469

How can we check the difference?

@seyyah We'll add these ports (STM32F767, STM32F746 with CubeMX+Atollic) as soon as possible.

Do you have ports to custom boards or development boards?

kl-cruz, 2018-11-12:
IMO it should be enough to compare driver files for MIPI DSI peripheral. API should be the same so for now I can't see big troubles.
I'll look at st documentation in a days

kisvegabor, 2018-11-14:
I think the difference can be in the MSP files. E.g. the backlight can be on a different pin. A can try it on the weekend.

kisvegabor, 2018-12-06:
@seyyah @kl-cruz 
I've just ported LittelvGL to STM32F769 Discovery: https://github.com/littlevgl/lv_projects/tree/master/stm32f769_discovery_no_os

However, it's still not perfect. There are some artifacts on the screen on refresh and I don't know why it is. Can you check it? Maybe you will have an idea

kisvegabor, 2018-12-06:
I've found that the issue is in `tft.c`, `CopyBuffer()` which uses DAM2D to refresh the frame buffer's content.

If I replace it with this code can the issue is gone:
```c
static void CopyBuffer(const uint32_t *pSrc, uint32_t *pDst, uint16_t x, uint16_t y, uint16_t xsize, uint16_t ysize)
{
	uint32_t row;
	uint32_t col;
	for(row = y; row < y + ysize; row++ ) {
		for(col = x; col < x + xsize; col++ ) {
			pDst[row * 800 + col] = *pSrc;
			pSrc++;
		}

	}
}
```

seyyah, 2018-12-07:
> @seyyah We'll add these ports (STM32F767, STM32F746 with CubeMX+Atollic) as soon as possible.
> Do you have ports to custom boards or development boards?

I'm sorry for late answer (ping https://github.com/littlevgl/lvgl/issues/543#issuecomment-445159961). We have both custom boards (based on 767, 5/7/10inch, STMPE811/resistive touch) and discovery/development (429i-disco, 746G-disco) boards.

papadkostas, 2018-12-13:
I've tried to port it few months ago, but had an issue with DMA.
The transition from frame to frame was so slow that i could see the lines changing.
@kisvegabor  have you tried the port on real hardware?

kisvegabor, 2018-12-14:
Thank you for the test in advance.

I've just quickly created a video about it: https://www.youtube.com/watch?v=LRtotWbnpss

papadkostas, 2018-12-14:
@kisvegabor It's working pretty nice from first view. Will read more about it in the documentation since I'm just looking into the demo app. One thing i noticed is that the touch sensor is not that much accurate, but anyway. Also I've noticed in your home page about AviTab, had no idea that this addon for X plane made with littlevgl. So thank you twice and wish you all the best for you and your new family member!! :)

kisvegabor, 2018-12-14:
@papadkostas Thank you for the test and the good wishes! :)

>  Also I've noticed in your home page about AviTab, had no idea that this addon for X plane made with littlevgl. 
If you have a better insight into AviTab and X plane can you suggest a better description?

papadkostas, 2018-12-16:
> better
No, the description is OK. I was just surprised to know that LittlevGL was behind of AviTab :)

papadkostas, 2018-12-16:
That's good :D Yea, X-Plane has big community and it's getting bigger everyday, one of the reasons for this is the free Zibo 737 add-on(a modified version of the default 737 plane) which is really better than most of the available payware add-on and this one uses AviTab.  Anyway, a lot of out of topic. Will check more about lvgl this week :)

kisvegabor, 2018-12-17:
@papadkostas Good to know, thanks :)

Anyway, the porting question seems to be solved so I close this issue. If you find bugs or have improvement proposals feel free to comment here.

closed 2018-12-17 by kisvegabor