lvgl/lvgl #524: Calendar font setting problem

qwert1213131, 2018-11-05:
The default font of the system is lv_font_dejavu_30. When the style part is set to lv_font_dejavu_20, an exception is displayed.
Can you check it out for me? Thank you！

[image]
[image]

@kisvegabor

turoksama, 2018-11-05:
[](https://github.com/littlevgl/lvgl/wiki/Calendar)   may help.

turoksama, 2018-11-05:
[https://github.com/littlevgl/lvgl/wiki/Calendar](url) may help.

kisvegabor, 2018-11-05:
@turoksama 
I don't know why but your link to the documentation shows the issues page. Although the link seems correct.

Maybe this?
https://github.com/littlevgl/lvgl/wiki/Calendar

qwert1213131, 2018-11-05:
@kisvegabor @turoksama
I am using the release 5.2-rc version and the calendar font cannot be set successfully.
  calendar = lv_calendar_create(scr,NULL);

/*Create a style for the current week*/
static lv_style_t style_week_box;
lv_style_copy(&style_week_box, &lv_style_plain);
style_week_box.body.border.width = 1;
style_week_box.body.border.color = LV_COLOR_HEX3(0x333);
style_week_box.body.empty = 1;
style_week_box.body.radius = LV_RADIUS_CIRCLE;
style_week_box.body.padding.ver = 3;
style_week_box.body.padding.hor = 3;
style_week_box.text.font = &lv_font_dejavu_10;

/*Create a style for today*/
static lv_style_t style_today_box;
lv_style_copy(&style_today_box, &lv_style_plain);
style_today_box.body.border.width = 2;
style_today_box.body.border.color = LV_COLOR_NAVY;
style_today_box.body.empty = 1;
style_today_box.body.radius = LV_RADIUS_CIRCLE;
style_today_box.body.padding.ver = 3;
style_today_box.body.padding.hor = 3;
style_today_box.text.color= LV_COLOR_BLUE;
style_today_box.text.font = &lv_font_dejavu_10;

/*Create a style for the highlighted days*/
static lv_style_t style_highlighted_day;
lv_style_copy(&style_highlighted_day, &lv_style_plain);
style_highlighted_day.body.border.width = 2;
style_highlighted_day.body.border.color = LV_COLOR_NAVY;
style_highlighted_day.body.empty = 1;
style_highlighted_day.body.radius = LV_RADIUS_CIRCLE;
style_highlighted_day.body.padding.ver = 3;
style_highlighted_day.body.padding.hor = 3;
style_highlighted_day.text.color= LV_COLOR_BLUE;
style_highlighted_day.text.font = &lv_font_dejavu_10;

/*Apply the styles*/
lv_calendar_set_style(calendar, LV_CALENDAR_STYLE_WEEK_BOX, &style_week_box);
lv_calendar_set_style(calendar, LV_CALENDAR_STYLE_TODAY_BOX, &style_today_box);
lv_calendar_set_style(calendar, LV_CALENDAR_STYLE_HIGHLIGHTED_DAYS, &style_highlighted_day);

turoksama, 2018-11-05:
@kisvegabor 
i don't know why neither, cursor moved to the middle of () automatically.😅

turoksama, 2018-11-05:
maybe you would like to try this out too: 
LV_CALENDAR_STYLE_INACTIVE_DAYS text properties are used to adjust the style of the visible days of previous/next month.

qwert1213131, 2018-11-06:
@turoksama
Thank you!
Already working fine.
This error is caused by forgetting to set other style types.

closed 2018-11-06 by qwert1213131