lvgl/lvgl #525: Issue with 32bpp on ubuntu Simulator

nimkan, 2018-11-05:
[image]

using above picture I created the C array with true color and try using lvgl on pc simulator (32bpp) , It is showing as below. Could you please help me on this ?

[image]

kisvegabor, 2018-11-07:
Please check what happens if you create a button. It should have bluish color.

kisvegabor, 2018-11-30:
I close this issue because there was no activity here for a while .

If you have remarks about this topic feel free to comment here and reopen the issue.

closed 2018-11-30 by kisvegabor