lvgl/lvgl #528: Disable specific indev temporarly

AloyseTech, 2018-11-05:
Hi,

There is already a way to enable/disable a given type of input device. However, it could be interesting to be able to enable/disable specific input device whatever their type.
The existing API is confusing : lv_indev_enable is used to enable/disable device by type.
Maybe it should be changed to lv_indev_enable_type and lv_indev_enable could be used to disable a given input device.

embeddedt, 2018-11-05:
@AloyseTech Such a change would have to be guarded by a macro for backwards compatibility.

AloyseTech, 2018-11-05:
Yes, or it could only make it on 6.0.0. But it seems to be a feature that could be useful before the next major version is available

embeddedt, 2018-12-10:
As per https://github.com/littlevgl/lvgl/issues/543#issuecomment-438566272 the `dev-6.0` branch was removed. Please use `#if LV_API_V6` to guard API-breaking changes.

kisvegabor, 2019-03-02:
In `dev-6.0` I'll change `lv_indev_enable` to disable/enable an input device instead of a type.

kisvegabor, 2019-03-13:
Added in `dev-6.0`.
If you have come comments feel free to add them here and reopen this issue.

closed 2019-03-13 by kisvegabor