lvgl/lvgl #532: ST7735: display issues (animation deformation and DPI settings)

MichaelVLV, 2018-11-05:
I'm using 1.8" 160x128 display with st7735 driver.
Datasheet: [https://www.displayfuture.com/Display/datasheet/controller/ST7735.pdf](url)
I have two problems.
1) DPI settings works incrorrectly (I suppose)
2) Image on display deformated after switching list positions

DPI: 114
[image]
[image]

DPI: 10
[image]
[image]
[image]

For contrillion I'm using hardware buttons.
I have made initialization for it and run basic UI:

Buttons read:
```C
static bool buttons_read(lv_indev_data_t * data)
{
	if(buttons.state_DOWN == true)
	{
		buttons.state_DOWN = false;
		printf("DOWN\n");
		lv_group_send_data(g,LV_GROUP_KEY_RIGHT);

		data->key = 0x00000001;
		data->state = LV_INDEV_STATE_PR;
	}

	if(buttons.state_UP == true)
	{
		buttons.state_UP = false;
		printf("UP\n");
		lv_group_send_data(g,LV_GROUP_KEY_LEFT);

		data->key = 0x00000002;
		data->state = LV_INDEV_STATE_PR;
	}
    return false;
}
```

drv initialisation:
```C
void buttons_drv_reg(void)
{
    lv_indev_drv_t  kp_drv;
    lv_indev_drv_init(&kp_drv);
    kp_drv.type = LV_INDEV_TYPE_KEYPAD;
    kp_drv.read = buttons_read;
    emulated_kp_indev = lv_indev_drv_register(&kp_drv);

    /*Create an object group*/
    g = lv_group_create();

    /*Assig the input device(s) to the created group*/
    lv_indev_set_group(emulated_kp_indev, g);
}
```

test UI:
```C
void lv_test_UI2(void)
{
	/*Crate the list*/
	lv_obj_t * list1 = lv_list_create(lv_scr_act(), NULL);
	lv_obj_set_size(list1, 130, 170);
	lv_obj_align(list1, NULL, LV_ALIGN_IN_TOP_LEFT, 20, 40);

	/*Add list elements*/
	lv_list_add(list1, SYMBOL_FILE, "New", list_release_action);
	lv_list_add(list1, SYMBOL_DIRECTORY, "Open", list_release_action);
	lv_list_add(list1, SYMBOL_CLOSE, "Delete", list_release_action);
	lv_list_add(list1, SYMBOL_EDIT, "Edit", list_release_action);
	lv_list_add(list1, SYMBOL_SAVE, "Save", list_release_action);

	/*Create a label above the list*/
	lv_obj_t * label;
	label = lv_label_create(lv_scr_act(), NULL);
	lv_label_set_text(label, "Default");
	lv_obj_align(label, list1, LV_ALIGN_OUT_TOP_MID, 0, -10);
    lv_group_add_obj(g, list1);
}
```

embeddedt, 2018-11-05:
A couple things to check:

1. The VDB and framebuffer memory are distinct entities and should not be pointing to the same space. If you don't have a framebuffer disregard that.
2. Create one button on your screen of width `LV_DPI`. Then measure whether the drawn button is one inch wide in real life.
3. Do the color depths of LVGL and your display driver match? The skewing on the display looks like an incorrect color depth issue.

turoksama, 2018-11-06:
4. check your porting isusse.

MichaelVLV, 2018-11-06:
> A couple things to check:
>
> 1. The VDB and framebuffer memory are distinct entities and should not be pointing to the same space. If you don't have a framebuffer disregard that.
> 2. Create one button [...]

1. Not sure about framebuffer, but looks like it is not used, my lv_conf.h (see below)
2. To update
3. Looks like everything match (both are 16 bit). 

Here is part from initialization:

```C
/* Interface pixel format: COLMOD */
{0x3A, {0x05}, 1},
```
**0x05 = 0000 0101b**
[image]

And from lv_conf.h:

```C
/*Color settings*/
#define LV_COLOR_DEPTH     16                     /*Color depth: 1/8/16/24*/
```

My full **lv_conf.h**:

```C
/**
 * @file lv_conf.h
 * 
 */

#ifndef LV_CONF_H
#define LV_CONF_H

/*----------------
 * Dynamic memory
 *----------------*/

/* Memory size which will be used by the library
 * to store the graphical objects and other data */
#define LV_MEM_CUSTOM      0                /*1: use custom malloc/free, 0: use the built-in lv_mem_alloc/lv_mem_free*/
#if LV_MEM_CUSTOM == 0
#define LV_MEM_SIZE    (32U * 1024U)        /*Size memory used by `lv_mem_alloc` in bytes (>= 2kB)*/
#define LV_MEM_ATTR                         /*Complier prefix for big array declaration*/
#define LV_MEM_AUTO_DEFRAG  1               /*Automatically defrag on free*/
#else       /*LV_MEM_CUSTOM*/
#define LV_MEM_CUSTOM_INCLUDE <stdlib.h>   /*Header for the dynamic memory function*/
#define LV_MEM_CUSTOM_ALLOC   malloc       /*Wrapper to malloc*/
#define LV_MEM_CUSTOM_FREE    free         /*Wrapper to free*/
#endif     /*LV_MEM_CUSTOM*/

/*===================
   Graphical settings
 *===================*/

/* Horizontal and vertical resolution of the library.*/
#define LV_HOR_RES          (160)//(128)
#define LV_VER_RES          (128)//(160)
#define LV_DPI              10 //114 // http://dpi.lv (160x128 with diagonal 1.8")

/* Size of VDB (Virtual Display Buffer: the internal graphics buffer).
 * Required for buffered drawing, opacity and anti-aliasing
 * VDB makes the double buffering, you don't need to deal with it!
 * Typical size: ~1/10 screen */
#define LV_VDB_SIZE         (20 * LV_HOR_RES)  /*Size of VDB in pixel count (1/10 screen size is good for first)*/
#define LV_VDB_ADR          0                 /*Place VDB to a specific address (e.g. in external RAM) (0: allocate automatically into RAM)*/

/* Use two Virtual Display buffers (VDB) parallelize rendering and flushing (optional)
[... 178 lines cut ...]
#if USE_LV_LIST != 0
#define LV_LIST_FOCUS_TIME  100 /*Default animation time of focusing to a list element [ms] (0: no animation)  */
#endif

/*Drop down list (dependencies: lv_page, lv_label)*/
#define USE_LV_DDLIST    1
#if USE_LV_DDLIST != 0
#define LV_DDLIST_ANIM_TIME     200     /*Open and close default animation time [ms] (0: no animation)*/
#endif

/*Roller (dependencies: lv_ddlist)*/
#define USE_LV_ROLLER    1
#if USE_LV_ROLLER != 0
#define LV_ROLLER_ANIM_TIME     200     /*Focus animation time [ms] (0: no animation)*/
#endif

/*Slider (dependencies: lv_bar)*/
#define USE_LV_SLIDER    1

/*Switch (dependencies: lv_slider)*/
#define USE_LV_SW       1

#endif /*LV_CONF_H*/

```

MichaelVLV, 2018-11-06:
> Hi, probably the issue is in your `disp_flush` function. Please send it too.
> […](#)

here is flush:
```C
void st7735b_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_map)
{
	uint8_t data[4];

	/*Column addresses*/
	st7735b_send_cmd(0x2A);
	data[0] = (x1 >> 8) & 0xFF;
	data[1] = x1 & 0xFF;
	data[2] = (x2 >> 8) & 0xFF;
	data[3] = x2 & 0xFF;
	st7735b_send_data(data, 2);

	/*Page addresses*/
	st7735b_send_cmd(0x2B);
	data[0] = (y1 >> 8) & 0xFF;
	data[1] = y1 & 0xFF;
	data[2] = (y2 >> 8) & 0xFF;
	data[3] = y2 & 0xFF;
	st7735b_send_data(data, 2);

	/*Memory write*/
	st7735b_send_cmd(0x2C);

	uint32_t size = (x2 - x1 + 1) * (y2 - y1 + 1);

	/*Byte swapping is required*/
	uint32_t i;
	uint8_t * color_u8 = (uint8_t *) color_map;
	uint8_t color_tmp;
	for(i = 0; i < size * 2; i += 2) {
		color_tmp = color_u8[i + 1];
		color_u8[i + 1] = color_u8[i];
		color_u8[i] = color_tmp;
	}

	while(size > ST7735B_HOR_RES) {

		st7735b_send_data((void*)color_map, ST7735B_HOR_RES * 2);
		//vTaskDelay(10 / portTICK_PERIOD_MS);
		size -= ST7735B_HOR_RES;
		color_map += ST7735B_HOR_RES;
	}

	st7735b_send_data((void*)color_map, size * 2);	/*Send the remaining data*/

	lv_flush_ready();

}
```

kisvegabor, 2018-11-06:
```c
	/*Column addresses*/
	st7735b_send_cmd(0x2A);
	data[0] = (x1 >> 8) & 0xFF;
	data[1] = x1 & 0xFF;
	data[2] = (x2 >> 8) & 0xFF;
	data[3] = x2 & 0xFF;
	st7735b_send_data(data, 2);
```

It should be `st7735b_send_data(data, 4);`, isn't it? And the same for the `y` coordinate.

MichaelVLV, 2018-11-07:
Yep, that was a problem.
Sometimes digging in datasheets forces to make such mistakes :worried:

closed 2018-11-07 by MichaelVLV

ihanhua, 2018-12-24:
static void ex_disp_flush(int32_t x1, int32_t y1, int32_t x2, int32_t y2, const lv_color_t * color_p)
{
    /*The most simple case (but also the slowest) to put all pixels to the screen one-by-one*/

    int32_t len = (x2-x1+1)*(y2-y1+1);
    if(len >= 32*128){
        st7735_Address_set(x1,y1,x1+x2,y1+y2);
        while(len--)
        {
            st7735_writedata16(color_p->full);
            color_p++;
        }
    }else{
        int32_t x;
        int32_t y;
        for(y = y1; y <= y2; y++) {
            for(x = x1; x <= x2; x++) {
                /* Put a pixel to the display. For example: */
                /* put_px(x, y, *color_p)*/
                LCD_DrawPoint_set(x,y,color_p->full);
                color_p++;
            }
        }
    }

    //printf("flush\n");
    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
    lv_flush_ready();
}

#define LV_VDB_SIZE         (32 * LV_HOR_RES)

ihanhua, 2018-12-24:
this code can solved the question