lvgl/lvgl #533: focus object issue once uses touch

Masi2017, 2018-11-05:
Hi All,
I am using both touch and hardware keyboard. I have a Start btn and once btn is pressed I want to display an mbox with tow btns on it (cancel and ok). I'd like to have cancel btn selected once mbox is created . This is fine when I pressed the Start btn using the hardware key  (mbox is created and cancel is selected) but once I use touch to press the Start btn, none of the mbox btns is selected. Also I have to freeze the mbox for a while otherwise it won't be the focused object while I added it to my group and focused it ( this is fine with keypad also) .

kisvegabor, 2018-11-07:
Hi,

Please send an example code to reproduce the issue.

Masi2017, 2018-11-14:
Hi, sure . I need to create a simple example to demonstrate that , probably I would find some time by weekend , please don't close the issue.

Masi2017, 2018-12-05:
Hi @kisvegabor 
Here is my code:

static lv_res_t mbox_apply_action(lv_obj_t * mbox, const char * txt)
{
    printf("Mbox button: %s\n", txt);

    return LV_RES_OK; /*Return OK if the message box is not deleted*/
}

static lv_res_t btn_click_action(lv_obj_t * btn)
{
    uint8_t id = lv_obj_get_free_num(btn);
    /*Called when a button is clicked*/

    lv_obj_t * mbox1 = lv_mbox_create(lv_scr_act(), NULL);
    lv_mbox_set_text(mbox1, "Default message box\n"
                            "with buttons");                    /*Set the text*/
    /*Add two buttons*/
    static const char * btns[] ={"\221Apply", "\221Close", ""}; /*Button description. '\221' lv_btnm like control char*/
    lv_mbox_add_btns(mbox1, btns, NULL);
    lv_mbox_set_action(mbox1, mbox_apply_action);
    lv_obj_set_width(mbox1, 250);
    lv_obj_align(mbox1, NULL, LV_ALIGN_IN_TOP_LEFT, 10, 10); /*Align to the corner*/
    lv_group_add_obj(activeGroup, mbox1);
    lv_group_focus_obj(mbox1);

    return LV_RES_OK; /*Return OK if the button is not deleted*/
}

void test_freez_problem() {

  mainMenuKeypadOpsInit();

  mainGroup = lv_group_create();
  activeGroup = mainGroup;
  lv_indev_set_group(kb_indev, mainGroup);

/*Create a normal button*/
lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);
lv_cont_set_fit(btn1, true, true); /*Enable resizing horizontally and vertically*/
lv_obj_align(btn1, NULL, LV_ALIGN_IN_TOP_MID, 0, 10);
lv_obj_set_free_num(btn1, 1);   /*Set a unique number for the button*/
lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, btn_click_action);
lv_group_add_obj(mainGroup, btn1);
/*Add a label to the button*/
lv_obj_t * label = lv_label_create(btn1, NULL);
lv_label_set_text(label, "1");

lv_obj_t * btn2 = lv_btn_create(lv_scr_act(), btn1);
lv_obj_align(btn2, btn1, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
lv_obj_set_free_num(btn2, 2);               /*Set a unique number for the button*/
lv_group_add_obj(mainGroup, btn2);
/*Add a label to the toggled button*/
label = lv_label_create(btn2, NULL);
lv_label_set_text(label, "2");

lv_obj_t * btn3 = lv_btn_create(lv_scr_act(), btn1);
lv_obj_align(btn3, btn2, LV_ALIGN_OUT_BOTTOM_MID, 0, 10);
lv_obj_set_free_num(btn3, 3);               /*Set a unique number for the button*/
lv_group_add_obj(mainGroup, btn3);                    /*Add to the group*/
/*Add a label to the toggled button*/
label = lv_label_create(btn2, NULL);
lv_label_set_text(label, "3");

lv_group_focus_obj(btn1);

}

When I press OK key on btn1 , mbox is focused while once I touch the btn1 , mbox is created but btn1 is focused again. Here is the screenshot:
[image]

kisvegabor, 2018-12-18:
Sorry no responding. Somehow I wasn't notified about your comment. 

What is `mainMenuKeypadOpsInit`? I don't see the implementation of this function in your example code.

Anyway, `lv_group_set_click_focus(group, false)` should solve the issue. By default. if you click an object it will be focused too.

kisvegabor, 2019-01-05:
As the proposed solution should solve the issue I close it. If it's not working, please reopen it and comment here.

closed 2019-01-10 by kisvegabor