lvgl/lvgl #534: [suggestion]  Whether to consider adding a T-9 input keyboard in lv_kb ？

qwert1213131, 2018-11-06:
@kisvegabor
it is friendly to small touch screens.
[image]

embeddedt, 2018-11-06:
Probably the simplest logical method to achieve this would be to get a long press action (without repeat) from `lv_btnm`, then in that action function create another `lv_btnm` object with the options.

This would only work, however, if LittlevGL's button matrix lets you slide over buttons to choose one.

simonchen007, 2020-04-22:
@qwert1213131 
have you done that ?

kisvegabor, 2020-04-23:
@simonchen007 
You can also open a topic in the Forum in "Feature request" category:
https://forum.littlevgl.com/c/feature-request

simonchen007, 2020-04-26:
@kisvegabor 

[image]

I have finish that type of keyboard, is that possible to add it into the main lib ?
if yes, I can share the code here.

kisvegabor, 2020-04-27:
@simonchen007 
> I have finish that type of keyboard, is that possible to add it into the main lib ?

Yes, but probably we'd have some comments so I suggest opening an issue to discuss the details.