lvgl/lvgl #537: Slow and erratic behavior with simple project

BlueIan, 2018-11-07:
**Hardware**:
320x240 display with ILI9341 controller connected with SPI (@8MHz) to a 120MHz M4 (NXP Kinetis). Touchpad controller is XPT2046 (SPI at 2MHz)
**Problem:**
I have built a small project that has only a slider on the display and the operation is very slow and intermittent. I can get the slider position to change eventually if I continually press and/or tap the screen for several seconds, and sometimes it shifts to the correct position but usually just moves closer to where I touched. I added a cursor to help debug the touchpad position.  When touched, the old cursor position isn't restored when a new cursor is drawn so multiple cursor images stay on the display.  These are eventually cleared every 10 seconds or so. 
It seems like the whole thing is running at 0.01x speed.
[image]
And finally, the lv_task_handler() function blocks while the display is touched.

I am using the sample drivers provided here for both the display and the touchpad and I have only made simple required changes to lv_conf.h

**Debugging so far:**
I put a printf in touchpad read function and I can see it is getting called regularly and returns the correct position.
I have put a printf in lv_task_exec to show 'elp' values of 10, 20 or 30

I expect this is something small I have missed or changed by mistake but I'm not sure where to start looking.  Any help would be appreciated. 

[lvgl_code.zip](https://github.com/littlevgl/lvgl/files/2558191/lvgl_code.zip)

int main(void)
{
	low_level_init();
	
	// Initialize the SPI, touchpad and display
	spiInit();
	xpt2046_init();		// Touchpad
	ILI9341_hw_init();	// Display
	
        lv_init();
	
	// Initialize lvgl driver structures
	lv_disp_drv_t disp_drv;                         /*Descriptor of a display driver*/
	lv_disp_drv_init(&disp_drv);                    /*Basic initialization*/
	
	// Set up the functions to access to your display
	disp_drv.disp_flush = ILI9341_flush;            /*Used in buffered mode (LV_VDB_SIZE != 0  in lv_conf.h)*/
	
	// Finally Register the driver
	lv_disp_drv_register(&disp_drv);

	/*************************
	* Input device interface
	*************************/
	/*Add a touchpad in the example*/
	
	lv_indev_drv_t inputDevice_drv;                 /*Descriptor of an input device driver*/
	lv_indev_drv_init(&inputDevice_drv);            /*Basic initialization*/
	inputDevice_drv.type = LV_INDEV_TYPE_POINTER;   /*The touchpad is pointer type device*/
	inputDevice_drv.read = xpt2046_read;            /*Library read  your touchpad via this function*/
	//lv_indev_drv_register(&inputDevice_drv);        /*Finally register the driver*/
	
	// Cursor
	// -------------------------------------------------------------------------
	lv_indev_t * touch_indev = lv_indev_drv_register(&inputDevice_drv);        // Finally register the driver
	lv_obj_t * cursor_obj =  lv_img_create(lv_scr_act(), NULL); 				//create object
	lv_img_set_src(cursor_obj, SYMBOL_POWER); 								//set its image
	lv_indev_set_cursor(touch_indev, cursor_obj); 							// connect the object to the driver

	
	lv_obj_t * scr = lv_page_create(NULL, NULL);
	lv_scr_load(scr);
	
	/****************
	* ADD A SLIDER
	****************/
	lv_obj_t * slider1 = lv_slider_create(scr, NULL);		//Create a slider
	lv_obj_set_size(slider1, 32, 180);   					// Set the size
	lv_obj_set_x(slider1, 30);                        /*Set the x coordinate*/
	lv_slider_set_value(slider1, 30);
	
  while(1)
  {
  	  lv_tick_inc(10);
	  lv_task_handler();
	  RED_LED_NegVal();
	  DELAY_Waitms(10);
  }

}

embeddedt, 2018-11-07:
I think `lv_tick_inc` should be called with a timer interrupt or another RTOS task. So you would have two loops:

Interrupt handler (timer should run at a consistent rate of at least 10 ms or better):
```c
lv_tick_inc(10);
```

Main `while(1)` loop:
```c
while(1)
{
lv_task_handler();
RED_LED_NegVal();
DELAY_Waitms(10);
}
```

Your current implementation assumes that `lv_task_handler` takes 0ms to execute, which is not going to be the case.

BlueIan, 2018-11-07:
Thanks for your comment embeddedt.
I did have lv_tick_inc(1) called in a 1ms timer ISR originally with the same results.  I just move the lv_tick_inc back to the timer ISR and (not surprisingly) still the same.

kisvegabor, 2018-11-07:
Are you sure that the `DELAY_Waitms(10)` really makes a 10 ms delay?
What happens if you set it to `DELAY_Waitms(1)`?

BlueIan, 2018-11-07:
Thanks for your help Gabor.  I have cleared up several of the problems - an interfering ISR causing slow/no response and XY swapped values from touch screen controller. Sorry about the false alarm.

But the slider still does not respond as expected.
1) The area on the screen to move the slider is ~40 pixels to the right of the slider.  This is quite obvious with the cursor showing,  You can touch directly on the slider and the cursor will show directly on top of the slider but there is no action.  If you touch to the right of the slider it will show the cursor where you touched and move to the corresponding vertical position on the slider.
2) The slider doesn't 'slide' when you touch the screen - instead it jumps to the position where you _released_ the touchscreen.  Is this the expected operation? I need a slider that will follow my finger while I slide it on the screen,
3) The lv_task_handler() function blocks while the screen is touched.
4) I still have multiple cursor 'artifacts' on the screen.  Occasionally they are cleaned up but I can't figure out why or when.   I don't intend to use the cursor but I wonder if the erratic behavior is an indication of another problem I am having?

embeddedt, 2018-11-07:
1. Really? Using `pc_simulator` I can click on the slider and it moves properly. This might be an issue with touchscreen calibration.
2. No, that is not expected. Using `pc_simulator` the slider moves with my "finger".
3. Not with `pc_simulator`.

I think maybe your touchscreen driver is still misbehaving.

BlueIan, 2018-11-07:
Thanks, I was expecting it was input related. I will carefully check the XPT206 driver code and look at some of the other implementations.  Issue #1 is an odd one - I can see that the touchpad is returning the correct coordinates based on printf output of x/y values and the placement of the cursor.
Do you have any thoughts on #4?

embeddedt, 2018-11-07:
@Bluelan how are you implementing the cursor? I can try it now on `pc_simulator`.

embeddedt, 2018-11-07:
Can I see your `indev_read` function?

BlueIan, 2018-11-07:
bool xpt2046_read(lv_indev_data_t * data)
{
    static int16_t last_x = 0;
    static int16_t last_y = 0;
    bool valid = true;
    uint8_t buf;

    int16_t x = 0;
    int16_t y = 0;

    uint8_t irq;
    // Read IRQ pin
    // ----------------------------------------------
    irq = TP_IRQ_GetVal();

    if(irq == 0) {

    	// Read X and Y data from TP controller via SPI
    	// ------------------------------------------------------------------
    	spiReadXPT2046_XY(&x, &y);

        /*Normalize Data*/
        x = x >> 3;
        y = y >> 3;
        xpt2046_corr(&x, &y);	// Convert to LCD coordinates

        xpt2046_avg(&x, &y);	// Filter (average)

        last_x = x;
        last_y = y;

    } else {
        x = last_x;
        y = last_y;
        avg_last = 0;
        valid = false;
    }

    data->point.x = x;
    data->point.y = y;
    data->state = valid == false ? LV_INDEV_STATE_REL : LV_INDEV_STATE_PR;

    // IMS: DEBUG
#if 0
    if(data->state == LV_INDEV_STATE_PR)
    	printf("X= %5d, Y=%5d \r\n",data->point.x, data->point.y);
#endif

    return valid;
}

BlueIan, 2018-11-07:
Made the change and it works now!

embeddedt, 2018-11-07:
@Bluelan Awesome! I was hoping it would! Basically you are supposed to return `true` if you need to read more data right then, which in most cases you don't. It's a bit confusing but in general you want to return `false` from those read functions.

Are all the issues on that list solved now?

BlueIan, 2018-11-08:
That makes sense, thank you. 
The multiple cursor artifacts are gone and the slider touch-position offset is mostly gone, I will check the screen calibration.
So yes, that solves all the problems. Thanks again.
Ian

closed 2018-11-08 by BlueIan