lvgl/lvgl #541: offline image converter bugs

nacyzhouw, 2018-11-09:
Hi,
   Recently I used the offline image converter, and I found some bugs.
1. when image cf equal to alpha, no output file is generated. And there is no warining or error.
  php img_conv_core.php "name=icon&img=bunny.png&format=c_array&cf=alpha"
2. cannot convert jpg image, there is error:
PHP Fatal error:  Uncaught Error: Call to undefined function imagecreatefromjpeg() in /GUI/lv_utils-master/lv_utils-master/img_conv_core.php:68
Stack trace:
#0 /GUI/lv_utils-master/lv_utils-master/img_conv_core.php(671): Converter->__construct('earthtest.jpg', 'earthtest.jpg', 'earthtest', 'enabled')
#1 {main}
  thrown in /GUI/lv_utils-master/lv_utils-master/img_conv_core.php on line 68
3.sometime cannot convert png image correctly.
before convert: 
[image]
after convert:
[image]

nacyzhouw, 2018-11-10:
Hi,

Using the online converter, there are no the first and sencond bugs mentioned above. But when the cf=LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED, the image also cannot convert correctly, as mention in the third bug.
I really hope the bugs of the offline image converter can be fixed soon.

kisvegabor, 2019-01-05:
@nacyzhouw 
I've checked the issues:

1. It was a typo in the readme. I replaced `alpha` with `true_color_alpha`
2. `imagecreatefromjpeg` was working for me with `PHP 7.0.30`.  According to [its documentation](http://php.net/manual/en/function.imagecreatefromjpeg.php) this function exists since PHP 4
3. I've downloaded your battery icon and it worked correctly for me with `LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED`.  See this: [battest.c.txt](https://github.com/littlevgl/lvgl/files/2729978/battest.c.txt)

closed 2019-02-03 by stale[bot]

nacyzhouw, 2019-03-19:
@kisvegabor 
I found the background of the downloaded battery icon  opaque, which can be worked correctly.
But when the background of PNG image is transparent, it can't convert correctly

kisvegabor, 2019-03-19:
Did you use `LV_IMG_CF_TRUE_COLOR_ALPHA`?

nacyzhouw, 2019-03-21:
I used LV_IMG_CF_TRUE_COLOR_CHROMA_KEYED

kisvegabor, 2019-03-22:
And is it working with LV_IMG_CF_TRUE_COLOR_ALPHA?

nacyzhouw, 2019-03-25:
I

kisvegabor, 2019-03-25:
Please send:
- a test image 
- screenshot about with your settings
- an test C program you used to display the iamge

nacyzhouw, 2019-03-26:
[image]

nacyzhouw, 2019-03-26:
I am sorry for decribing the question unclearly , and  the  test C program cannot be uploaded successfully.

- You can try this test image .
[image]

- my settings about  Color and Graphics :
/*Color settings*/
#define LV_COLOR_DEPTH     32                     /*Color depth: 1/8/16/24*/
#define LV_COLOR_16_SWAP   0                      /*Swap the 2 bytes of RGB565 color. Useful if the display has a 8 bit interface (e.g. SPI)*/
#define LV_COLOR_SCREEN_TRANSP        0           /*1: Enable screen transparency. Useful for OSD or other overlapping GUIs. Requires ARGB8888 colors*/
#define LV_COLOR_TRANSP    LV_COLOR_LIME          /*Images pixels with this color will not be drawn (with chroma keying)*/

/*Graphics feature usage*/
#define USE_LV_ANIMATION        1               /*1: Enable all animations*/
#define USE_LV_SHADOW           1               /*1: Enable shadows*/
#define USE_LV_GROUP            1               /*1: Enable object groups (for keyboards)*/
#define USE_LV_GPU              1               /*1: Enable GPU interface*/
#define USE_LV_REAL_DRAW        1               /*1: Enable function which draw directly to the frame buffer instead of VDB (required if LV_VDB_SIZE = 0)*/
#define USE_LV_FILESYSTEM       1               /*1: Enable file system (required by images*/

embeddedt, 2019-03-26:
@nacyzhouw If uploading isn't working you can copy and paste the C program into the post.

nacyzhouw, 2019-03-27:
test

nacyzhouw, 2019-03-27:
test

nacyzhouw, 2019-03-27:
@embeddedt I try but it still don't work, maybe so many words cannot support to post

kisvegabor, 2019-03-28:
Your image had "indexed color profile'. I've changed it to "RGB "in GIMP (Image->Mode->RGB) and now it works.
[image]

nacyzhouw, 2019-03-29:
Oh，I see.  Thank you very much for your answer!

nacyzhouw, 2019-04-02:
@kisvegabor  I am sorry for that I still has the similar problem. This PNG image is  RGB fomat, but it still cannot convert rightly. It seem the alpha cannot work.
[image]