lvgl/lvgl #542: Maintainer wanted!

kisvegabor, 2018-11-09:
Hi All,

I was so happy to see that how LittelvGL's community is growing. As time passed I saw that not only questions but answers arrive from you as well. And your help is really required. As LittelvGL is getting more popular, more Pull requests, more ideas, more bugfixes are being sent. I feel that LittelvGL is not a one-person project anymore but it's really shaped and developed by a community. Which is awesome! And thank you very much for your work, help and support.

To handle this great new situation more maintainers are required who have write access to the library to accepts Pull requests. 

And ideas, suggestions on this topic are also welcome! :)

turoksama, 2018-11-09:
@kisvegabor Me too, but a little bit concerned. :)

embeddedt, 2018-11-09:
@turoksama What is your concern?

kisvegabor, 2018-11-09:
@embeddedt 
Awesome! It'd be great if you could help! :)
I've just sent the invitation.

@turoksama 
I'd be also curious about your concerns.

turoksama, 2018-11-09:
It's about the efficiency. Because I am still learning the library.  ^.~

kisvegabor, 2018-11-10:
@embeddedt 
I thank you too! :smile: 

@turoksama 
Don't worry, you are doing great! :smile: 

I close this now. I'm sure the community will grow fast and other maintainers will be required too soon.

closed 2018-11-10 by kisvegabor

embeddedt, 2018-11-10:
@kisvegabo Would you like me to close stale issues when they get too old without activity? If so, what is your approximate timeframe for considering an issue to be old?

seyyah, 2018-11-12:
@kisvegabor I would be interested and willing to become a maintainer if needed.

kisvegabor, 2018-11-13:
@seyyah Thank you very much for your offering! I'll add you.

@embeddedt @seyyah 
To make the changes reviewable, I suggest that even 3 of us use Pull requests to add new code to the library, instead of directle push. Do you agree?

embeddedt, 2018-11-13:
@kisvegabor Sounds like a plan.

turoksama, 2020-12-28:
@kisvegabor @embeddedt 
Hello guys,
 It's been a while since we last met. Here I have a propose for joining the maintainer program.
 I would like to apply for maintainer since the very first conversation a long time ago, it was not available back then, but now it is. 
And I think I can help with subjects as widgets, libraries(lv_lib_xx), style, etc.
 What's your opinion?
Best wishes!

kisvegabor, 2020-12-29:
Hi @turoksama 

Thank you! A few days ago I was thinking about it'd be great to have a maintainer(s) for the new [lv_components](https://github.com/lvgl/lv_components) repository. And you just wrote that you are interested in being a maintainer. :slightly_smiling_face: 

In the `lv_components` repo we will have the non-essential LVGL widgets and we plan to keep this repo "very open" for contribution with user-created widgets. This repo is still under development and will be finished when v8 is released. (Still ~2 months.) Would you be interested in being the maintainer of that repo?

turoksama, 2020-12-31:
Hi @kisvegabor 
Thank you! Feeling great to paticiate in the LIBRARY, I would be happy to join you with the journey. 
Need any help with rework of the new engine? Perhaps need some time to catch up with you at the mean while since v6.

kisvegabor, 2020-12-31:
@turoksama 
We are discussing 2 main things for v8:
- **Better inheritance**  I suggest reading the [first comment](https://github.com/lvgl/lvgl/issues/1919#issue-748108794) and start reading from [here](https://github.com/lvgl/lvgl/issues/1919#issuecomment-752692654).
- **Better styles** I've just summarized what we have now [here](https://github.com/lvgl/lvgl/issues/1802#issuecomment-752630806)

I've started to rewrite the inheritance and styles of LVGL but it's rather like an experiment now. So feedback and ideas are very welcome.

turoksama, 2021-01-01:
@kisvegabor 
Yes, I am looking into it at the moment. Please let me know when we launch the process.

kisvegabor, 2021-01-04:
Great, thanks!

xiaoxiang781216, 2021-03-02:
@kisvegabor thanks for invitation, it's my honor to work with you as a team to improve LVGL in the future.

turoksama, 2021-03-02:
@kisvegabor 
Thanks for your invitation, I am very happy to join your team and let's make more progress in the future.

kisvegabor, 2021-03-02:
Spelndid! Thank you for both of you!

Feel free to review and comment on PRs, fix simple bugs on your own, join the conversations on the issues here, answer questions in the forum, or any other thing that you like. :slightly_smiling_face: 

If you are about adding new features or making large modifications,   please open an issue to discuss it.