lvgl/lvgl #543: Issue with organization of git branches

embeddedt, 2018-11-12:
Currently, the LittlevGL repository has the following branches:

- `dev-5.3`
- `dev-6.0`
- `master`
- `lua`
- `beta`

The problem I've noticed in several issues is that sometimes fixes are pushed to `master` (currently v5.2) and they don't propagate to v5.3 or v6.0. I propose that whenever a change is made to master it should also be immediately merged to `dev-5.3` and `dev-6.0`.

On another note, I have noticed that the `beta` and `lua` branches are very stale at this point. Should they be removed?

kisvegabor, 2018-11-12:
> I propose that whenever a change is made to master it should also be immediately merged to dev-5.3 and dev-6.0.

I agree. In the earlier versions after a release usually, a patch version came (e.g. v5.1.1). In the case of `v5.2` I tried the `v5.2-rc` instead of the inevitable bugfix release. So I merged the bugfixes to master and the new features to `dev-5.3`. As recently  there are no `v5.2` related bug reports I think we can turn `v5.2-rc` to `v5.2`

Earlier the for `v5.1.1` like releases was merged to `beta`.

The lua branch contains important data about some experimental Lua binding, but it really should exist in among my (kisvegabor user) repos and not among littlevgl's repos. I will move it there.

So I'm open to any ideas which way should we follow. Use the `rc` versions or not? What do you think?

embeddedt, 2018-11-12:
Thanks for the explanation! Here is my proposed setup:

Based on what I have observed over the last few months and what you said `master` holds the current stable LittlevGL version. That can stay the way it is. We could leave `rc` on the releases page and simply push bugfixes to `master`. However, those bugfixes would then have to be merged into `dev-5.3` as well.

At the same time, obviously, there are features that are not stable but are in the beta stage. Those could be put in `dev-5.3`.

This would eliminate the need for the `beta` branch.

> The lua branch contains important data about some experimental Lua binding, but it really should exist in among my (kisvegabor user) repos and not among littlevgl's repos.

That would resolve my questions about that branch.

I'm also curious to know the state of `v6.0`. I understand that major versions will break APIs and that is why it needs to be in its own branch. However, I am wondering what your plans are for `v6.0`. It looks like it has been around since 5.1 and will still be present at the release of 5.3.

kisvegabor, 2018-11-13:
So don't use bugfix versions such as `v5.2.1`?
 
My intention was to continuously merge the API breaking updates in `dev-6.0`.  And release it when it has enough updates. However it's true,`dev-6.0` is developing very slowly.  I have some ideas for `dev-6.0`  but I won't have enough time in the upcoming months to add all of them either.  
Probably an other approach is required.

kisvegabor, 2018-11-13:
BTW, do you agree to release `v5.2`?

embeddedt, 2018-11-13:
> So don't use bugfix versions such as `v5.2.1`?

You could still have bugfix versions. Just push the bugfixes to the `master` branch and later on if you deem it a suitable time to release `v5.2.1` it should be safe to do so using the `master` branch.

> BTW, do you agree to release `v5.2`?

In regards to the core library, yes, I think it is ready, as long as you don't mind releasing while #470 and #541 are still open.

> . . . `dev-6.0` is developing very slowly. . . . Probably an other approach is required. Any ideas?

If not many changes have been added to `dev-6.0`, maybe keep a record of those changes offline and drop the branch. I recommend that you eliminate the parallelism in development versions, so maybe once 5.3 is released the next version will be 6.0. The only problem there is that you said you won't have enough time to add all the ideas for 6.0.

kisvegabor, 2018-11-14:
> You could still have bugfix versions. Just push the bugfixes to the master branch and later on if you deem it a suitable time to release v5.2.1 it should be safe to do so using the master branch.

Sounds well. This way people who just pull the master branch will get the bugfixes instantly. The question is how to label these versions in `lvgl.h`. It's not `v5.2`. Can we call it `v5.2-bugfix`?
This way `beta` is not required, isn't it?

> In regards to the core library, yes, I think it is ready, as long as you don't mind releasing while #470 and #541 are still open.

These are related to the converters, and independent from `lvgl` itself.  So I think we can still release `v5.2`

 > If not many changes have been added to dev-6.0, maybe keep a record of those changes offline and drop the branch. I recommend that you eliminate the parallelism in development versions, so maybe once 5.3 is released the next version will be 6.0. The only problem there is that you said you won't have enough time to add all the ideas for 6.0.

I agree to avoid parallel development. I removed the `dev-6.0` and `lua` branches. There are two big updates in my mind for `v6.0`: 
- Multi-display support
- Improve style system with margin and padding similarly to CSS. #302

The other changes are not so time consumeing. I suggest working on `v5.3` and we will see how things are going.

embeddedt, 2018-11-14:
@kisvegabor 

> This way people who just pull the master branch will get the bugfixes instantly.

Exactly what I was thinking.

> The question is how to label these versions in `lvgl.h`. It's not `v5.2`. Can we call it `v5.2-bugfix`?

Hmm. Why not just keep the `.1` notation (i.e. `v5.2.1`)? 

```c
#define LVGL_VERSION_MAJOR   5
#define LVGL_VERSION_MINOR   2
#define LVGL_VERSION_PATCH   1
```

kisvegabor, 2018-11-14:
> Hmm. Why not just keep the .1 notation (i.e. v5.2.1)?

Then let's call it `v.5.2.1-dev` to see it's not stable and still get updates. What do you think?

embeddedt, 2018-11-14:
sounds good.

kisvegabor, 2018-11-14:
Ok, I've just released `v5.2` and started `v5.2.1-dev` in `master`.
I also deleted `beta` so now we only have `master` and `dev-5.3`.

embeddedt, 2018-11-16:
@kisvegabor I think that would work.

I would like to make a patch to README.md. How would you like me to go about making sure this change ends up in both `master` and `dev-5.3`? I assume I should make the change and then merge `master` into `dev-5.3`?

kisvegabor, 2018-11-17:
> I assume I should make the change and then merge master into dev-5.3?

Yes. The ideal case would be to merge `master` to `dev-5.3`. after every update.

embeddedt, 2018-11-20:
@kisvegabor How many collaborators should review a pull request before it's merged? I ask because sometimes I think a PR can be merged but I don't want to be adding code without your approval.

kisvegabor, 2018-11-21:
@embeddedt 
I'd prefer at least one verification about the update is really working, compiles and its quality is fine.
I always read all the issues and Pull requests but usually, I have time to answer only in a few days. If something has an undesirable approach I will comment on it faster. Is it okay?

embeddedt, 2018-11-21:
@kisvegabor No problem at all!

kisvegabor, 2018-11-21:
The features should be collected somewhere to make easier:
- creating a release note
- updating the documentation

I suggest creating a release note draft an collect the features there. Like this: https://github.com/littlevgl/lvgl/releases/tag/untagged-381801967e2defbbc152

I'm not sure who can see this draft.

What do you think about it?

embeddedt, 2018-11-21:
@kisvegabor Good idea. I can view the draft as a collaborator. I tested in another browser and you get a 404 if you're not signed in.

kisvegabor, 2018-11-21:
I think it's not a big problem if only we see the draft. The other approach would be to create a file in `docs/` folder but finally, only people with write access can write that file too (or send PR)

kisvegabor, 2018-11-21:
I updated the draft with the new features so far. It seems the link is changing when the draft is saved.

embeddedt, 2018-11-30:
@kisvegabor I was doing some thinking about the whole `v6.0` issue and I have a suggestion on how to approach it.

Can we start working on `v6.0` features but add new APIs while still keeping the older ones?

This would give the ability to continue releasing `5.x` releases while adding new functionality. As the features were added, we could update the documentation and explain that the older API is deprecated and will be removed in `v6.0`.

Do you like that suggestion?

kisvegabor, 2018-11-30:
@embeddedt I love this idea. There are some open issues which quite easy to implement but needs API change.  

I think it should be only an entry in `lv_conf.h` like: `LV_API_V6`.

kisvegabor, 2018-11-30:
> When v6.0 is released I assume that would be made a permanent define?

By permanent do you mean leaving the define in` lv_conf.h` after releasing `v6.0` to enable switching between teh APIs?

embeddedt, 2018-11-30:
@kisvegabor No, I was thinking it should be moved to `lvgl.h` because `v6.0` should permanently remove the old APIs.

kisvegabor, 2018-11-30:
I agree `v6.0` should remove the old API, but then why not delete the define and the old API when v6.0 is released?

embeddedt, 2018-11-30:
@kisvegabor Your solution would also work. The reason I suggested leaving a define is so that external code designed to work with both `v5.x` and `v6.0` can know which APIs to use.

embeddedt, 2018-11-30:
@kisvegabor Your solution is better. Please disregard my last comment.

kisvegabor, 2018-11-30:
@embeddedt Ok, then let's continue this way! :)

embeddedt, 2018-12-06:
@kisvegabor Congratulations! Would you like me to handle the issues and PRs myself during this time?

kisvegabor, 2018-12-07:
@embeddedt Thank you! :) 

> Would you like me to handle the issues and PRs myself during this time?

That would be great! Thank you for this too! :)

seyyah, 2018-12-07:
> @embeddedt
> Big things happened here! Our first baby was just born in this week.

Congratulations.

15 days ago, my 3rd child was born. The greatest miracle we have encountered in the world for us :pray:

kisvegabor, 2018-12-07:
@seyyah  Congratulations for you too! :) It's so good the hear some personal things about you and get to know each other a little bit better :)

kisvegabor, 2018-12-12:
@harteware07 Thank you, and congratulations for you too! We are really looking forward to these changes! :)

kisvegabor, 2018-12-17:
Already there is a lot of news in `dev-5.3` and as I see some big things are approaching which should go to `v6.0`. So I suggest releasing `v5.3` soon and continue with `dev-6.0`.
This way `LV_API_V6` finally wouldn't be required. 

Do you agree?

embeddedt, 2018-12-17:
@kisvegabor Good idea. So you are suggesting that we hold off on `architecture` issues until `v5.3` is released?

kisvegabor, 2018-12-18:
@embeddedt 
Yes, but I think we make the release very soon. It would be great to add the color picker object because it's almost ready.  (#565 )

kisvegabor, 2018-12-19:
@amirgon  I really liked your idea to add Micropython binding in `v5.3`. As I see it's already ready enough to make some experiments with it. Do you still want to add some feature before the release?

amirgon, 2018-12-19:
@kisvegabor I'm working on structs and styles now, probably will take a few more days, depending on how much free time I'll find to work on this.  
Depending on your `v5.3` schedule, you can either release it as-is or wait for structs and styles. There is no immediate change I want to add to it now.  

When you create `v5.3`, please make sure that the last thing you do is run `gen_mpy.py` to generate `lv_mpy.c`. This way Micropython Binding will be built based on your latest object interfaces, defines in the conf file etc.  

I use the following command line (run it from `lvgl/micropython` dir):

```bash
python gen_mpy.py -X anim -X group -I pycparser/utils/fake_libc_include ../lv_objx/*.h > lv_mpy.c
```

It should complete in a few seconds without any errors or warnings.

kisvegabor, 2018-12-20:
@amirgon @embeddedt 
I was thinking about the schedule. As Christmas time and New Year's Eve are approaching probably people will be more inactive in the upcoming days. So suggest making the release at the beginning of January.  

It will also give time to make some things ready.

How does it sounds to you?

kisvegabor, 2018-12-21:
@embeddedt Yes, June sounds reasonable!

kisvegabor, 2019-01-11:
Not stale.

kisvegabor, 2019-01-12:
@embeddedt I'm thinking about splitting the `lv_project` repository to more, separate repositories. It would be easier to clone and maintain the projects. The author of the project should have write access to the repo to allow its maintenance.

Do you agree with that?

embeddedt, 2019-01-12:
@kisvegabor Sounds like a good idea.

amirgon, 2019-01-29:
@kisvegabor I think so, but I can't check it right now. I can pull and test it in the evening.  
What happens after v5.3 is released? Is it merged with master?   
If there are any problems or changes, will can still push them (to master), right? I was thinking of doing some changes on how gc is used.

kisvegabor, 2019-01-29:
> What happens after v5.3 is released? Is it merged with master?

Yes `dev-5.3` will be merged to `master`

> If there are any problems or changes, will can still push them (to master), right?

After the release  `master` will receive only bugfixes.

> I was thinking of doing some changes on how gc is used.

If the current version works without any issue (not crashes) the improvements should go the `dev-6.0`

amirgon, 2019-01-29:
> After the release `master` will receive only bugfixes.

@kisvegabor What is the motivation to avoid adding small features to `master`? I understand it should be backward compatible, but why make the users wait for `dev-6.0` for some small feature if they can get it now on `master`?

embeddedt, 2019-01-29:
@amirgon We want our releases to be as stable as possible, and master always reflects the latest release, plus any bugfixes. For that reason, we avoid adding new features there.

To make up for that, I think that our development branch is much stabler in general than a bleeding-edge tree would be in another project. Therefore, if users want to try the newer features, they would normally be able to checkout `dev-5.x`, which (generally) has very few showstopping bugs.

This release is special because we are going to be changing a lot of architectural code in `v6.0`, which will mean that our development branch will be very unstable, and not ideal for users to work off of.

amirgon, 2019-01-29:
> This release is special because we are going to be changing a lot of architectural code in `v6.0`, which will mean that our development branch will be very unstable, and not ideal for users to work off of.

@embeddedt @kisvegabor For that reason I suggest that certain small features would still be pushed to master (after collaborating and verifying that this is ok).
I expect that when Micropython binding starts to be widely used, many small things would come up. It would be a shame to delay them for half a year.

embeddedt, 2019-01-30:
@kisvegabor @amirgon 

> Let's say we rename `_lv_scr_act` to `_dummy_lv_scr_act` and if the new feature uses this variable it would be difficult to merge and maintain the updates.

We could make a special exception for Micropython. We could get @amirgon to continue developing the binding off the `v5.3` codebase, and once we have stabilized the APIs for `v6.0`, simply migrate the binding to the newer version. It would allow for bugfixes, etc. to be implemented without having to worry about APIs breaking. To do that I suppose we would have to create a special branch, maybe something like `dev-mp`?

This would solve the problem of changing features in `master`, and avoid the issue of the unstable APIs in `v6.0`.

amirgon, 2019-01-30:
> To do that I suppose we would have to create a special branch, maybe something like `dev-mp`?

@embeddedt Do you suggest that every Micropython related change would go to both `master` and `dev-mp`? Or only to `dev-mp`?  
What about `lv_binding` - it has lvgl submodule. My thinking was that it would point to lvgl `master`. Do you suggest it would point to `master` or `dev-mp` branch? In either case it would require merging from one to the other every time.  

What is the benefit of having `dev-mp` vs. making the changes on `master`? (and possibly on `v6.0` as well). I'm talking about backward compatible changes that would not add risk to lvgl `master` branch.

amirgon, 2019-01-30:
@embeddedt 

> It should point to `dev-mp`.

I'm not sure that's a good idea. It's the same as making an lvgl fork and using it for Micropython binding. I thought the idea was to **integrate with lvgl**, not fork lvgl.  
Suppose you push some lvgl bug-fixes to master (unrelated to Micropython). Will you remember pushing them to `dev-mp` as well for every such bugfix?

> Can you guarantee that your changes aren't going to introduce new bugs into a stable release?

That depends on the change.  
I'll give you an example -  suppose we found some set of defines and would like to convert them to enum. A very simple change, C users would not be affected, Micropython users will benefit from another accessible enum in their scripts.

kisvegabor, 2019-01-30:
@amirgon 
> That depends on the change.
I'll give you an example - suppose we found some set of defines and would like to convert them to enum. A very simple change, C users would not be affected, Micropython users will benefit from another accessible enum in their scripts.
   
Finally, I agree with updating the master with very straight forward things. Not new features just easy-to-overview changes of the existing code. And the changes in master should be merged to `dev-5.3` too.
However, because of big API breaking changes, like the new action handling, merging between master and dev-6.0 might be difficult.  

Is it acceptable for you too?

kisvegabor, 2019-01-30:
> What is your position regarding dev-mp branch and the lvgl sub-module in lv-micropython? Do you agree it would point to master, and there is no need for dev-mp branch?

` lv_micropython` can include `master` but it (and `lv_binfing_micropython`) should have a branch where it uses `lvgl` `dev-6.0` to test the new features (e.g. the new action handling). This way `dev-mp` is not required.

embeddedt, 2019-01-30:
@amirgon @kisvegabor That sounds alright.

kisvegabor, 2019-01-30:
Great! Then only [this issue](https://github.com/littlevgl/lvgl/issues/776) needs to be solved to release v5.3.

amirgon, 2019-01-31:
> Great! Then only [this issue](https://github.com/littlevgl/lvgl/issues/776) needs to be solved to release v5.3.

And also #780.

kisvegabor, 2019-01-31:
> >Great! Then only this issue needs to be solved to release v5.3.

>And also #780.

#780 merged.

I'll release v5.3 today!

kisvegabor, 2019-02-03:
@embeddedt I'd like to have an other person too who have administrator right to the LittlevGL organization. It's important if because of any reasons I can't continue working here it should be somebody who can create repos, give write accesses etc. Would you accept it?

embeddedt, 2019-02-03:
@kisvegabor I'd be glad to! Thanks for the offer!

kisvegabor, 2019-02-03:
Awesome! Thank you very much. 
I sent the invitation. You will have the same right than me.

puzrin, 2019-02-08:
If it's not too late, i'd suggest a bit different branching approach:

- `master` - latest development. With mandatory CI to make sure it works somehow (in ideal world - with good tests coverage)
- `0.5.x` (numeric) - for patches

Reason - IMO it's common to expect edge code in master.

[7 more replies]

closed 2019-02-18 by kisvegabor