lvgl/lvgl #544: Is creating LVGL objects thread safe?

nameofuser1, 2018-11-12:
If I have two threads one that renders image using `lv_task_handler` and the other one that creates UI objects, let's say button with `lv_btn_create`, should I consider it to be thread safe?

embeddedt, 2018-11-12:
@nameofuser1 No, LittlevGL is not thread-safe in that sense. You can, however, get some degree of thread safety by doing the following:

1. In the thread that calls `lv_task_handler`, lock a mutex before the call and unlock it immediately after.
2. In all other threads that use any LittlevGL API functions, lock a mutex before calling the APIs, and then unlock it afterward.

This will ensure that at any given point in time only one of the threads in the system is utilizing the APIs offered by LittlevGL.

nameofuser1, 2018-11-12:
Thanks!

Do you think that it is worth to add locking support mechanism in LVGL by the means of preprocessor definitions for creating mutexes and (un)locking them?

embeddedt, 2018-11-12:
@nameofuser1 No. It would require a review of every public function and there would be too great of a potential to miss adding the necessary locking to every function. It is simpler to have the user handle the locking themselves.

kisvegabor, 2018-11-13:
I agree with @embeddedt . The proposed locking method is robust and very easy to implement.

You can read about this topic here: https://github.com/littlevgl/lvgl/wiki/Porting#using-with-an-operating-system

BrianPugh, 2018-11-15:
also keep in mind that callbacks will be executed in the main `lv_task_handler` loop, thus a callback handler doesn't need (will actually cause a deadlock) to take the mutex before creating/destroying/modifying lvgl objects.