lvgl/lvgl #545: Can littlevgl porting to android or linux?

zhanfeng06, 2018-11-13:
I want to set up a recorder machine based on android or linux system.Can littlevgl porting to linux?

embeddedt, 2018-11-13:
Yes, LittlevGL can run on Linux, by using SDL2. Check out https://github.com/littlevgl/pc_simulator.

zhanfeng06, 2018-11-14:
The recorder machine is an embedded machine, running on ARM. Can we run littlevgl directly without X11 or SDL ?

embeddedt, 2018-11-14:
@zhanfeng06 Yes. Have a look at https://github.com/littlevgl/lv_drivers/blob/master/display/fbdev.c. This is a driver that uses the Linux framebuffer directly. You will have to do some adaptation to use that with pc_simulator though.

zhanfeng06, 2018-11-14:
@embeddedt  There will be some application using this library(like camera、settings), so our recorder machine will use IPC. And will need to transfer the buffer between littevgl and application, How to adapt it? And is there some basic library dependency?

embeddedt, 2018-11-14:
@zhanfeng06 I'm not sure exactly what you mean, so please let me know if my interpretation of your questions is incorrect.

> And is there some basic library dependency?

If you mean "does LittlevGL depend on any external libraries", then no, it should work fine on any platform. It has been used on many embedded platforms without full-fledged C libraries (i.e. Arduino, STM32) as well as Linux. 

> our recorder machine will use IPC.

That would be something you would have to look into for your specific application. LittlevGL is designed to run within the context of one thread only; you would have to create your own message interface and handle all calls to the library yourself.

If you are asking whether LittlevGL provides out-of-the-box support for multiple applications, the answer would be no.

> And will need to transfer the buffer between littevgl and application.

Please clarify what you mean here.

zhanfeng06, 2018-11-14:
Ok, thanks very much. I will try to porting it over linux.

fallstool, 2018-11-28:
@zhanfeng06  you can decode the buffer as a image, and transfer the image as the lvgl that can display. In the arm, you can just port to **FBDEV_PATH** like /dev/fb0, you can set in the lv_drv_conf.h
#define USE_FBDEV           0
#if USE_FBDEV != 0
#define FBDEV_PATH          "/dev/fb0"
#endif

kisvegabor, 2018-12-18:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-12-18 by kisvegabor

ajsb85, 2021-06-11:
Seems interesting LVGL ported to Android.

kisvegabor, 2021-06-14:
I agree! :slightly_smiling_face: 
Do you have ideas about how it should be done?

ajsb85, 2021-06-14:
The NDK for sure.

Nice example:
https://github.com/zlgopen/awtk-android