lvgl/lvgl #547: setting up ESP32 + Arduino + ILI9341

dasBuero, 2018-11-13:
Hello,

I just found this beautiful piece of work and wanted to try this out with my ESP32 microcontroller (Arduino IDE) and an ILI9341 driven touch display. I'm a beginner, so any information bits about using this library with Arduino I have found so far are hard to understand for me. Could anybody help me with a working configuration setup in a beginner-friendly language?

Things I need to understand:

- importing this library into my Android IDE
- importing and setting up the appropriate display drivers for ILI9341 touch display running with an ESP32 (I have also a Atmega328 controller, if that is easier to work out)
- using one of the examples provided with this library

I know that maybe that's too much to ask, but any further information that would bring me a step further is very appreciated. My setup is very common (at least I think so), so it would help other beginners like me.

Thanks in advance!

kisvegabor, 2018-11-14:
Hi,

Actually, there is an example project for ESP32 + ILI9341, but it's not for Arduino. https://github.com/littlevgl/lv_projects/tree/master/esp32_ili9341

And there is an empty example project for Arduino: https://github.com/littlevgl/lv_projects/tree/master/arduino_library

In the README you will find some instructions: https://github.com/littlevgl/lv_projects/blob/master/README.md

From these projects, you can see how you can make it work.

dasBuero, 2018-11-14:
Thank you for the pointers! I will try to work out an appropriated setup and if it all works like it should I could write a beginner-friendly step-by-step tutorial for the wiki (if there is any interest).

kisvegabor, 2018-11-15:
@dasBuero  Yes, there is a big interest! :)

dasBuero, 2018-11-15:
I'll try my best but I'll definitely need some support of the more experienced users here.

kisvegabor, 2018-11-16:
What is the problem exactly?

embeddedt, 2018-12-10:
As there was no activity here for 1 month I will close this issue to keep the list from growing too large. Feel free to reopen the issue.

closed 2018-12-10 by embeddedt

vitormhenrique, 2019-02-24:
Hello, It looks like the "lv_projects" does not exist anymore. Are there any example configurations or projects?

kisvegabor, 2019-02-24:
We've split `lv_projects` to more individual repos. See:
* https://github.com/littlevgl/arduino
* https://github.com/littlevgl/esp32_ili9431

vitormhenrique, 2019-02-24:
Awesome! maybe add this to the "get started" documentation?
https://littlevgl.com/get-started

Really good documentation btw....

kisvegabor, 2019-02-24:
Ups, updated, thank you :)