lvgl/lvgl #548: Group style modifier for different style type (BG, INDIC, KNOB etc)

AloyseTech, 2018-11-13:
Hi,

When adding a style modifier to a group with `lv_group_set_style_mod_cb` or `lv_group_set_style_mod_edit_cb ` it is only possible to change the "base" style (LV_SLIDER_STYLE_BG for the slider example). It would be great if one could modify all the style type of an object (LV_SLIDER_STYLE_INDIC and LV_SLIDER_STYLE_KNOB for the slider example).

kisvegabor, 2018-11-14:
Hi,

this idea was already raised once but I don't find that issue. 

Anyway, at that time we didn't find a solution which was resource and user-friendly. The two thoughts were:
1.  Besides `LV_SLIDER_STYLE_INDIC` add `LV_SLIDER_STYLE_INDIC_FOCUSED` style to the slider. But this would mean the duplication of styles even if the FOCUSED styles are not required/used. 
2. Store e.g. 8 styles in `lv_group_t`, and add a new signal (e.g. `LV_SIGNAL_STYLE_FOCUS`). in this case, the user can add a custom signal function to handle this signal and modify the styles as he wants. But adding a custom signal functions in not user-friendly.

Do you have a better idea?

AloyseTech, 2018-11-15:
We could have an ext attribute which is a modifier callback pointer for each object with complex styling : 
```C
void cb_focus(obj, styletype, style);
void cb_edit(obj, styletype, style);

```
This could then be called in the design function for each type of style depending of the event (focus/edit)

kisvegabor, 2018-11-16:
Sounds reasonable, but one function should be enough with an extra parameter for `editing`.
And if `USE_LV_GROUP` is disabled in `lv_conf.h` this function should be disabled too.
Do you agree?

AloyseTech, 2018-11-16:
Sounds good!

kisvegabor, 2018-11-19:
As `lv_slider` and `lv_sw` is derived from `lv_bar` I suggest adding this extension to `lv_bar`. This way the other two objects also can use it.

kisvegabor, 2019-01-05:
@AloyseTech 
In `v6.0` will improve the `actions`. See [this](https://github.com/littlevgl/lvgl/issues/316#issuecomment-443499107)
Summary: Every object will have a common action function with an `lv_action_t` parameter to show the type of action (CLICK, PRESS, DRAG etc.)

In the new concept, the action handler will be called on focus and defocus too. This way you will have the opportunity to explicitly change the styles.

AloyseTech, 2019-01-06:
Sounds great! I like the idea that this will work for any object, it adds a lot of flexibility

kisvegabor, 2019-01-06:
Awesome! Then I close this issue. The implementation will be in v6.0 (v6.0 should be released in the summer)

closed 2019-01-06 by kisvegabor