lvgl/lvgl #549: Animations on switch control

embeddedt, 2018-11-13:
I'm interested in adding animation to the switch control, somewhat like what you see in this picture.

[image]

Would this be a viable feature for `dev-5.3`?

kisvegabor, 2018-11-14:
Absolutely yes! It looks great!

embeddedt, 2018-11-14:
@kisvegabor Great! Should I make it an option or just overwrite the existing code?

kisvegabor, 2018-11-14:
@embeddedt  I usually use an `uint16_t anim_time` variable in `lv_..._ext_t` structure. This way the user can set the `anim_time` to zero to disable this feature.

embeddedt, 2018-11-15:
@kisvegabor Ok. I'll make a pull request and begin working on this feature right now.

embeddedt, 2018-11-20:
The discussion will continue in #553.

PR #553 (2018-11-21): Add animations to switch control
This pull request adds support for an animated switch control. It resolves #549.
closed 2018-11-20 by embeddedt