lvgl/lvgl #550: Using littlevgl with a display in portrait mode.

jaythecelt, 2018-11-14:
Hi,

I'm a new littlevlg user.  We're using an 1024x600 LCD display that we want to use in portrait mode.  The display mode doesn't change dynamically - the display orientation is fixed.  

My idea was to rotate the VDB in my `disp_drv.disp_flush(...)` implementation right before it is written to the display.  However It looks like lvgl  is intended for landscape mode since `lv_refr_area_with_vdb(...)` calls `lv_refr_area_part_vdb(area_p)` by rows in the x direction.  

I'm not sure how to rotate the display in my driver without making changing `lv_refr_area_with_vdb(...)` - or always flushing the entire screen.  Neither seems practical - it seems like I'm off base here and am missing something.

What is the preferred way to use lvgl in portrait?

embeddedt, 2018-11-14:
@jaythecelt I don't remember the exact math to do it, but this is easily implemented by a combination of swapping and subtracting x and y values.

jaythecelt, 2018-11-14:
@embeddedt Thanks for the quick reply.  
In lv_conf.h I have `#define LV_VDB_SIZE (20 * LV_HOR_RES)` which means that `disp_drv.disp_flush(...)` is called 600/20=30 times to draw the entire screen.  

If I rotated each VDB before writing to my DMA buffer, I would have to wait until the 30th call to start the DMA transfer.  

Is there a better way to implement this?

embeddedt, 2018-11-14:
Hmm. I didn't think about doing this with DMA. I'm not sure but there must be a faster way to do it. I assume since you're using DMA that there isn't a framebuffer?

jaythecelt, 2018-11-14:
I'm writing to a frame buffer that's `LV_HOR_RES * LV_VER_RES` in size; as each VDB is flushed, that section of the frame buffer is transferred via DMA.  

I'm starting to think that DMA may not be they way to go - like you said maybe there is a better way to write it out, then I'm not limited by the DMA transfer process.

jaythecelt, 2018-11-14:
@embeddedt Thanks for the help!

jaythecelt, 2018-11-14:
@embeddedt Ah ... I understand your question better now ... (sorry - little slow - It's late here).

I'm correcting myself, I am NOT using a frame buffer but writing the VDB directly to the display via DMA.

embeddedt, 2018-11-14:
@jaythecelt 

> If I rotated each VDB before writing to my DMA buffer, I would have to wait until the 30th call to start the DMA transfer.

I'm not that familiar with DMA yet, but can't you start transferring data as soon as you get the first VDB?

kisvegabor, 2018-11-15:
@jaythecelt Do you have an external display controller? If yes, it's almost certain that you can set the scanning directions of its framebuffer. In other words, you can set to scan the framebuffer row-by-row or column-by-column. If you turn the display (portrait mode) you should choose the other scan direction.

jaythecelt, 2018-11-15:
@kisvegabor No, I'm using the display controller in the STM32F4.  I don't think there is a way to change the orientation, but I'll double check that.  Thanks.

jaythecelt, 2018-11-15:
@embeddedt I think I have it worked out to rotate the display in disp_flush and writing to the frame buffer in the controller directly rather than using DMA.  To speed up the display refresh, I plan on using two frame buffers A and B, then pointing the display controller to one or the other after updating.  I'm working on that now and hope to have it working very soon.

embeddedt, 2018-11-15:
@jaythecelt Sounds like a good plan. I hope it works for you!

embeddedt, 2018-12-10:
There has been no activity for 1 month so I will assume you got your speed improvements working. If you are still having problems feel free to reopen the issue and comment.

closed 2018-12-10 by embeddedt