lvgl/lvgl #554: Project structure - help needed

effeci, 2018-11-15:
Good morning! I've finally been able to get a working project for my f746 discovery board and I'm in the process of testing everything. I'm trying to build a speedometer which structure should be :
-Splash screen ( logo or gif logo ( if possible stored in the ad))
- main screen: variables display rpm speed temp hour tripmeter and so on
- menu: in which variables and interface can be set up
The first problem I'm finding is how to switch between these different interfaces and how, in your opinion, these interfaces should be organized considering that in the main screen will be a lot of callbacks for actually refreshing and storing variables.
Thank for your time,
Fabrizio

turoksama, 2018-11-15:
I think it would be a good idea to read the WIKI manual first, and you will have a better understanding of the library.

kisvegabor, 2018-11-16:
@effeci  As @turoksama said, please read the Wiki documentation first. Especially the section about [Objects](https://github.com/littlevgl/lvgl/wiki/Objects)

I close this issue. If something is not clear after reading the documentation, please open a new issue with a more specific question.

closed 2018-11-16 by kisvegabor