lvgl/lvgl #555: Combine multiple buttons on one screen with scroll

MichaelVLV, 2018-11-15:
Hi,

Is it possible to make analog of list _lv_list_create()_ but with buttons?

[image]

I want them can be scrolled, but now when I'm navigating down I go outside of screen.

```C
void list_of_btn()
{
	/*Create a scroll bar style*/
    static lv_style_t style_sb;
    lv_style_copy(&style_sb, &lv_style_plain);
    style_sb.body.main_color = LV_COLOR_BLACK;
    style_sb.body.grad_color = LV_COLOR_BLACK;
    style_sb.body.border.color = LV_COLOR_WHITE;
    style_sb.body.border.width = 1;
    style_sb.body.border.opa = LV_OPA_70;
    style_sb.body.radius = LV_RADIUS_CIRCLE;
    style_sb.body.opa = LV_OPA_60;
    style_sb.body.padding.hor = 3;          /*Horizontal padding on the right*/
    style_sb.body.padding.inner = 8;        /*Scrollbar width*/

	/*Create a page*/
    lv_obj_t * page = lv_page_create(lv_scr_act(), NULL);
    lv_obj_set_size(page, LV_HOR_RES+10, LV_VER_RES+10);
    lv_obj_align(page, NULL, LV_ALIGN_CENTER, 0, 0);

    lv_page_set_style(page, LV_PAGE_STYLE_SB, &style_sb);           /*Set the scrollbar style*/
    lv_page_set_sb_mode(page, LV_SB_MODE_AUTO);                     /*Show scroll bars is scrolling is possible*/

	// ---------- BUTTON 1
    lv_obj_t * btn1 = lv_btn_create(page, NULL);         /*Create a button on the currently loaded screen*/
    lv_obj_set_size(btn1, 100, 24);
    lv_obj_align(btn1, page, LV_ALIGN_IN_TOP_MID, 0, 0);

    /*Create a label on the button (the 'label' variable can be reused)*/
    lv_obj_t * label = lv_label_create(btn1, NULL);
    lv_label_set_text(label, "History");
    lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, my_setup_action);  // LV_BTN_ACTION_CLICK // LV_BTN_ACTION_PR
    lv_obj_set_free_num(btn1, 1);               /*Set a unique number for the button*/
    lv_group_add_obj(history_group, btn1);

    lv_obj_t * btn[10];

    for(uint8_t i = 0; i < 10; i++)
    {
    	btn[i] = lv_btn_create(lv_scr_act(), btn1);
    	lv_obj_align(btn[i], btn1, LV_ALIGN_OUT_BOTTOM_MID, 0, i*20);
    	label = lv_label_create(btn[i], NULL);
    	sprintf(buf, "History %d", i);
    	lv_label_set_text(label, buf);
        lv_obj_set_free_num(btn[i], i);               /*Set a unique number for the button*/
    	lv_group_add_obj(history_group, btn[i]);
    }
}
```

embeddedt, 2018-11-15:
Is your page too long for the screen?

kisvegabor, 2018-11-17:
It seems easier to use `lv_list` and modify the size of the list elements if required.
To do this:
```c
lv_obj_t * list_btn = lv_list_add(list1, NULL, buf, my_setup_action);
lv_btn_set_fit(btn, false, false);
lv_obj_set_size(btn, 100, 70);
```

mheranco, 2018-11-20:
Hello,
the problem with your first code example could be that you create the buttons with `lv_scr_act()` as parent.
Try
```btn[i] = lv_btn_create(page, btn1);```
instead of 
`btn[i] = lv_btn_create(lv_scr_act(), btn1);`
Then the buttons are created on the page and it should be scrollable.

kisvegabor, 2018-12-18:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2018-12-18 by kisvegabor