lvgl/lvgl #557: Feature suggestion: Micropython binding

amirgon, 2018-11-16:
[Micropython](https://github.com/micropython/micropython) is becoming mainstream on IOT and embedded development in general. It is available for quite a few of embedded architectures ([this](https://github.com/micropython/micropython/wiki/Boards-Summary) and [this](https://forum.micropython.org/viewforum.php?f=10), partial list), and can be ported to new ones.
Today there already exist several architecture-specific micropython display modules (for example [this](http://docs.micropython.org/en/latest/library/lcd160cr.html) and [this](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/wiki/display)), but none of them as feature rich and general as littlevgl.  
I think it could be useful to provide micropython API to littlevgl Objects and Styles.  
Main benefits are:
- Exposure to a larger audience in the embedded development world.
- Interactive developement. Use the Micropython interpreter to exercise littlevgl over a live embedded board. This would allow rapid GUI development without the build-flash cycle.
- Some would argue Python code is easier to develop and maintain (not sure I agree about this one...)

If it turns out there is an agreement about this feature, I might try sending PR if I find the time for it.

kisvegabor, 2018-11-18:
Hi,

I also see a big potential in it. Both reaching a larger audience and the benefits of Micropython sounds very well. Earlier I made some experiments with Lua binding but Micropython is more mainstream right now.

So this feature would be very welcome! 
How can this binding look like? One or more c file(s) with the interface functions?

embeddedt, 2018-11-18:
@turoksama There will definitely be overhead from using Python due to the need to convert objects back and forth. I guess the hope is that Python will be used on platforms which can afford the additional overhead.

amirgon, 2018-11-18:
Here is what I had in mind.  
The lvgl micropython module would provide a constructor for each object type, and each object would provide its methods.  
Here is the "Hello World" example:  

```python
import lvgl
label1 = lvgl.label(lvgl.get_scr_act())
label1.set_text("Hello world!")
label1.align(None, lvgl.ALIGN_CENTER, 0, 0)
```

Creating micropython extensions requires lots of boilerplate. I don't think we should write them manually. Writing them manually would be not only tedious to start with, but also hard to maintain when objects are changed and added.   
Instead I suggest creating these extensions automatically from the source code (the C headers) relying on the naming conventions and maybe some hints.  
During the make process a script will run, scan the lvgl object headers (with [pycparser](https://github.com/eliben/pycparser) for example) and generate the Micropython extensions in a C file that can be later compiled and linked on Micropython-enabled projects.

kisvegabor, 2018-11-19:
@amirgon  Sound good to me! I also vote for the automated way. 
From your example, we can see how to call LittelvGL from Micropython. But how to call Micropython from LittelvGL? It is required when clicking a  button and a user-defined action should be executed.

amirgon, 2018-11-19:
@kisvegabor Calling a Micropython function from C code is covered by Micropython.  
You can use `mp_call_function_*` defined [here](https://github.com/micropython/micropython/blob/57a7d5be9ae3d367982c0f25b9f0215f58029faa/py/runtime.h#L105), so we get this one for free.  
In case of a button (or any other) `set_action` micropython extension, we can give it a python function as a parameter since functions are first-class objects in python, can be passed around, used in lists dicts etc. so I think this should be pretty simple.

One thing we'll need to figure out is how to separate display/board specific code from generic lvgl code.  
We would need to provide Micropython extensions to register display driver, set IO pins, configure SPI, DMA, etc.

embeddedt, 2018-11-19:
@amirgon The only problem with your suggestion is that the performance overhead of handling the low-level display code through Python might be too high. What do you think?

amirgon, 2018-11-19:
@embeddedt It's true that Micropython is slower than C, and costs more resources (RAM, Flash). 
 However, I believe Micropython is still beneficial for platforms that can afford the extra resources (such as ESP32, for example):

- Micropython would only be used for the Objects/Styles API layer (for setting object properties), not for the graphics rending itself which remains in C.  Think about a button, for example. How many cycles are spent on creating the button object and setting its properties (action, state, style) vs. cycles spent on actually writing pixels to the display to render it? My guess is that rending would take most of the cycles (and rendering remains in C), while setting object properties would only consume a very small portion of the cycle budget.  
Also take into account that in most cases setting object properties is done once per GUI "screen" and rendering happens continuously while the user interacts with that certain "screen". 

- Micropython is optimized for architectures with limited resources (CPU and memory). More information can be found [here](https://docs.micropython.org/en/latest/reference/constrained.html). Micropython compiles code into bytecode and has an efficient VM that executes it. Micropython modules may also be implemented as frozen bytecode modules saving the need to compile them by the MCU before running them.

- Micropython support for lvgl would be optional, and you would need to explicitly opt in. If you do nothing you don't pay any cycles or memory. If you are developing for a platform where you want to save cycles and memory, this feature will not get in your way, and you can still write all your code in C as today and get the same performance.

kisvegabor, 2018-11-20:
@amirgon I agree, calling `lv_set_...` functions from Micropython would have only a minor overhead if the rendering remains in C. I suppose we can keep the `disp_fush` function in C as well, right?

amirgon, 2018-11-20:
@kisvegabor I agree, `disp_flush` should remain a low level C function.

kisvegabor, 2018-11-21:
@amirgon 
Did you already make some experiments with LittlevGL in Micropython?

kisvegabor, 2018-11-21:
@amirgon 
Great! I'm really waiting for that :)

kisvegabor, 2018-11-25:
Looks amazing! My old dream to make an interactive online GUI demo tool but didn't find the right way to do this.

I'm looking forward to try Micropython!

embeddedt, 2018-11-25:
@kisvegabor Have you considered [Emscripten](https://kripken.github.io/emscripten-site/#)? It converts C to JavaScript and might be faster than using Unicorn and MicroPython for a demo.

kisvegabor, 2018-11-25:
@embeddedt 
I tried Emscripten once but it was extremely slow and wasn't stable (sometimes not started properly). Probably it is worth an other try.

amirgon, 2018-11-26:
@embeddedt if you do have time to experiment with Emscripten, please try to build [Micropython](https://github.com/micropython/micropython) as part of lvgl sources.  
This [has been done before](https://github.com/micropython/micropython/issues/3474).

amirgon, 2018-11-26:
Not ready for a PR yet, but I have a working script.
It scans all `*.h` files in `lv_objx` dir and generates an Mpy module called `lvgl` which contains object for each lvgl object, method functions etc.  
It relies heavily on naming conventions. (For example, it assumes that an object is constructed by `lv_%s_create` with two specific arguments.  
Luckily, these naming conventions are followed very well on lvgl sources.

Here is a working example I just run on my device:

```
>>> import lvgl
ILI9341 initialization.
Enable backlight.
>>> label1 = lvgl.label(lvgl.scr_act())
>>> label1.set_text("Hello world!")
>>> label1.align(None, 0, 0, 0)
```

A nice thing about Mpy REPL is the automatic completion (start typing and press `tab`), the `help` function etc.  

For example:

```
>>> help(label1)
object lvgl label is of type label
  del -- <function>
  clean -- <function>
  invalidate -- <function>
  set_parent -- <function>
  set_pos -- <function>
  set_x -- <function>
  set_y -- <function>
  set_size -- <function>
  set_width -- <function>
  set_height -- <function>
  align -- <function>
  refresh_style -- <function>
  set_hidden -- <function>
  set_click -- <function>
  set_top -- <function>
  set_drag -- <function>
  set_drag_throw -- <function>
  set_drag_parent -- <function>
  set_opa_scale_enable -- <function>
  set_opa_scale -- <function>
  set_protect -- <function>
  clear_protect -- <function>
  refresh_ext_size -- <function>
  set_free_num -- <function>
  get_screen -- <function>
  get_parent -- <function>
  get_child -- <function>
  get_child_back -- <function>
  count_children -- <function>
  get_x -- <function>
  get_y -- <function>
  get_width -- <function>
  get_height -- <function>
  get_ext_size -- <function>
  get_hidden -- <function>
  get_click -- <function>
  get_top -- <function>
  get_drag -- <function>
  get_drag_throw -- <function>
  get_drag_parent -- <function>
  get_opa_scale -- <function>
  get_protect -- <function>
  is_protected -- <function>
  get_free_num -- <function>
  is_focused -- <function>
  set_text -- <function>
  set_array_text -- <function>
  set_static_text -- <function>
  set_long_mode -- <function>
  set_align -- <function>
  set_recolor -- <function>
  set_body_draw -- <function>
  set_anim_speed -- <function>
  get_text -- <function>
  get_long_mode -- <function>
  get_align -- <function>
  get_recolor -- <function>
  get_body_draw -- <function>
  get_anim_speed -- <function>
  ins_text -- <function>
  cut_text -- <function>

>>> help(lvgl)
object <module 'lvgl'> is of type module
  __name__ -- lvgl
  __init__ -- <function>
  obj -- <class 'obj'>
  arc -- <class 'arc'>
  cont -- <class 'cont'>
  btn -- <class 'btn'>
  label -- <class 'label'>
  bar -- <class 'bar'>
  btnm -- <class 'btnm'>
  cb -- <class 'cb'>
  line -- <class 'line'>
  chart -- <class 'chart'>
  page -- <class 'page'>
  ddlist -- <class 'ddlist'>
  lmeter -- <class 'lmeter'>
  gauge -- <class 'gauge'>
  img -- <class 'img'>
  kb -- <class 'kb'>
  led -- <class 'led'>
  list -- <class 'list'>
  mbox -- <class 'mbox'>
  preload -- <class 'preload'>
  roller -- <class 'roller'>
  slider -- <class 'slider'>
  sw -- <class 'sw'>
  win -- <class 'win'>
  tabview -- <class 'tabview'>
  ta -- <class 'ta'>
  font_init -- <function>
  anim_init -- <function>
  init -- <function>
  scr_act -- <function>
  layer_sys -- <function>
  disp_is_mem_fill_supported -- <function>
  tick_get -- <function>
  group_focus_obj -- <function>
  indev_enable -- <function>
  txt_ins -- <function>
  fs_init -- <function>
  fs_remove -- <function>
  fs_get_letters -- <function>
  fs_up -- <function>
  draw_aa_get_opa -- <function>

```

Next things to implement:

- Add Enums to lvgl module to allow the user specify a name instead of a number
- Callbacks (function pointers)
- Styles
- Structs (for example lv_color_t)
- Missing conversions

If you want to see the code, have a look at:

- [gen_mpy.py](https://github.com/amirgon/lvgl/blob/master/micropython/gen_mpy.py) - This script scans lvgl headers and generates the Mpy module.
- [lv_mpy.c](https://github.com/amirgon/lvgl/blob/master/micropython/lv_mpy.c) - This is the generated file. In the future it should not be committed, but generated by the Makefile instead. It provides the Mpy module `mp_module_lvgl` which should then be registered in Mpy `mpconfigport.h`.  
It's already quite large, about 9000 generated source lines.  

I'll be happy to receive comments.

kisvegabor, 2018-11-28:
@amirgon 
It looks amazing! 
Do you know how much flash space is required by `lv_mpy.c`?

I didn't use Micropython so far but I'd like to try it with my STM32F419 Discovery. I fond this: https://github.com/micropython/micropython/wiki/Board-STM32F429 
Do you think lv_mpy.c can be integrated into that project? Or is it possible for me who is not familiar with Micropython? :)

kisvegabor, 2018-11-30:
@amirgon 
I'm planning to try Micropytthon on Linux with framebuffer as a display. I started to put the project together. Some parts of the build process are already working I don't know what is best way to link `lvgl` to micropython.  Now I got a lot of similar errors:
```
build/../../../lvgl/mpy/lv_mpy.o: In function `mp_lv_txt_ins':
lv_mpy.c:(.text.mp_lv_txt_ins+0x31): undefined reference to `lv_txt_ins'
```

I suppose I need to add the source files of `lvgl` to the makefile but don't know what is the best to do it. Can you help with that?

amirgon, 2018-11-30:
@kisvegabor  Yes you need to add lvgl sources to the Makefile.
First, I configured lvgl as an optional component, the following way:
 - Edit `Kconfig.projbuild` and add the following section under `menu "Modules"`

```
	    config MICROPY_USE_LVGL
	       bool "Use LVGL module"
	       default n
	       help
	       Include LittlevGL Embedded GUI Library
```
- Run `make menuconfig` and enable `use LVGL module` under Micropython --> Modules. Alternatively you could edit `sdkconfig` directly and add `CONFIG_MICROPY_USE_LVGL=y`

- Edit `components/micropython/component.mk` and add lvgl sources like this:

```
ifdef CONFIG_MICROPY_USE_LVGL
SRC_C += <...lvgl C files..>
endif
```

Alternatively you could add the source files recursively, but make sure you also read [this](http://aegis.sourceforge.net/auug97.pdf). A better technique is something like [this](https://stackoverflow.com/a/10190866/619493).
On ESP32 all it takes is adding `component.mk` (even empty one!) on every directory that contains files you want to include in the build (as explained [here](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#component-makefiles)).  
If you build Micropython on Linux that is probably different.

amirgon, 2018-11-30:
@kisvegabor Please make sure you take the latest version when you try it out.
Apart from bug fixes, I added the following features:

- Object Inheritance. For example, when accessing a `btn` object, you can use `btn` methods, `cont` methods and `obj` methods, which are all the ancestors of `btn`. (The script deduces the heirarchy from the `..._ext_t` structs)
- Enums. Each enum is created as its own type, either as a child of an object type or of the lvgl module. For example, 

`ALIGN` is a member of lvgl

```
>>> help(lvgl.ALIGN)
object <class 'LV_ALIGN'> is of type type
  OUT_LEFT_BOTTOM -- 17
  OUT_TOP_LEFT -- 9
  IN_RIGHT_MID -- 8
  OUT_RIGHT_MID -- 19
  IN_BOTTOM_MID -- 5
  CENTER -- 0
  OUT_RIGHT_TOP -- 18
  OUT_BOTTOM_MID -- 13
  OUT_LEFT_MID -- 16
  IN_TOP_MID -- 2
  IN_BOTTOM_LEFT -- 4
  IN_TOP_RIGHT -- 3
  OUT_TOP_MID -- 10
  OUT_BOTTOM_LEFT -- 12
  OUT_RIGHT_BOTTOM -- 20
  OUT_LEFT_TOP -- 15
  IN_BOTTOM_RIGHT -- 6
  OUT_TOP_RIGHT -- 11
  OUT_BOTTOM_RIGHT -- 14
  IN_TOP_LEFT -- 1
  IN_LEFT_MID -- 7
```
but `STATE` is a member of `btn`:
```
>>> help(lvgl.btn.STATE)
object <class 'LV_BTN_STATE'> is of type type
  TGL_REL -- 2
  NUM -- 5
  TGL_PR -- 3
  PR -- 1
  INA -- 4
  REL -- 0
>>> 
```

kisvegabor, 2018-11-30:
@amirgon Thank you very much! I will try it on Sunday!

embeddedt, 2018-12-01:
@amirgon The issue you describe sounds similar to #316 and the solution would probably apply to both.

I think your idea of adding a context argument is probably a simple and clean solution. @kisvegabor and I recently had a brief discussion about changes that break compatiblity: https://github.com/littlevgl/lvgl/issues/543#issuecomment-443180096

amirgon, 2018-12-02:
@embeddedt while the problem sounds similar, the suggested solution of #316 is different. Adding something like `lv_action_id` is not general enough for callbacks.  
I would prefer the generic solution of context/user_data parameter on each callback. It's also simpler and clear for anyone who ever worked with callbacks on other libraries.  

Actually I noticed this is already used on lvgl on some places! For example, `lv_indev_data_t` contains `user_data` which was passed originally when registering the driver by `lv_indev_drv_init`.  
Why not generalize user_data for all callbacks?

kisvegabor, 2018-12-02:
@amirgon 

## Test results
I made it work on PC. Finally, I used SDL like to normal PC simulator. I also tested the new enums.
```python
label1 = lvgl.label(lvgl.scr_act())
label1.set_text("Hello world")
label1.align(lvgl.scr_act(), lvgl.ALIGN.CENTER, 0, 0)
```

However, I faced an issue. `lv_mp_init();` was not called during `import lvgl`.
To make it work I added `lv_mp_init();` to `lv_init()`.
And I added `lv_task_handler` to the monitor refresh tread.

## My Micropython project with SDL
I uploaded my SDL-Micropython project here: https://drive.google.com/file/d/16htsMz3L1knvfHWdPsmwg3YyiOtSbpF0/view?usp=sharing

Instructions:
1. Go to: `micropython/linux/micropython/ports/unix`
2. `make` (ignore the warnings :) )
3. `./micropython` it should start Micropython COmmand Line interface 
4. `import lvgl`
5. `lvgl.init()` it should open a new white window
6. `label1 = lvgl.label(lvgl.scr_act())`
7.  `label1.set_text("Hello world")`
8. ...continue testing... :)

## Actions
You can use `lv_obj_set_free_ptr(obj, &context_data)` and `lv_obj_set_free_num(obj, some_id)` to add an arbitrary pointer or number to an object which can be get in the action using `lv_obj_get_free_ptr()` and `lv_obj_get_free_num()`. It similar to the `user_data` at input devices.

amirgon, 2018-12-02:
### @kisvegabor I'm glad to hear you had a success with Micropython!  
Regarding `lv_mp_init()`, it's registered as `__init__` on `mp_module_lvgl` globals, so it should be called automatically (and is actually called on my application on ESP32).  
I'm not sure what the problem on your side is, but **please make sure you are using a relatively recent version of Micropython**. I think it was not called on older versions of Micropython.  
Another option to work around this it is to call `lvgl.__init__()` explicitly. Did you try this?

Currently I'm developing and testing Micropython integration with lvgl on ESP32, I think it's good you are trying this integration on another platform.

### Regarding Actions - 
Using `free ptr` is very much like option (3) I suggested [above](https://github.com/littlevgl/lvgl/issues/557#issuecomment-443455949) (using an existing `lv_obj_t` data member instead of adding a new one).  
The disadvantage of using `free ptr` instead of `context` parameter is that once the callback is called, I can know which **object** it relates to, but not which **specific callback** was registered.  
So if the same object has many callbacks, I have to define a function for each instead of using a single function to handle callbacks.  
As I explained above, implementing it this way would require more RAM, require more ROM (many callback functions in the code), and it would allow only one callback per object per callback type (for example you can't register multiple callbacks and call them all when the event happens. I don't know if you do this anywhere on lvgl).

kisvegabor, 2018-12-02:
I'm using `v1.9.4` which is the latest release. `lvgl.__init__()` is working. Strange.
In `mpconfigport.h` I registered lvgl like this 
`{ MP_OBJ_NEW_QSTR(MP_QSTR_lvgl), (mp_obj_t)&mp_module_lvgl },`
Is it correct?

With @embeddedt we are [talking about](https://github.com/littlevgl/lvgl/issues/316#issuecomment-443430583) having only one action callback per object but with an `enum` parameter about the action type, such as `LV_ACTION_CLICK`, `LV_ACTION_PRESS` etc. 

So the signature would be: 
- Register a action: `void lv_obj_set_action(lv_obj_t * obj, lv_action_t action);`
- The callback: `lv_res_t my_action(lv_obj_t * obj, lv_action_t action)`

This way the user has the opportunity to have a `free_prt` like this: 
```c
static void * contexts[LV_ACTION_NUM];
contexts[LV_ACTION_CLICK] = &x;
contexts[LV_ACTION_PRESS] = &y;

lv_obj_set_free_ptr(obj, contexts);
```

So we can decide now which direction to choose for `v6.0` where we can have API breaking changes. Please let us know what do you think about the actions.

amirgon, 2018-12-02:
@kisvegabor 
> In `mpconfigport.h` I registered lvgl like this
> `{ MP_OBJ_NEW_QSTR(MP_QSTR_lvgl), (mp_obj_t)&mp_module_lvgl },`
> Is it correct?

Looks ok. I do this in a similar way, see [here](https://github.com/amirgon/MicroPython_ESP32_psRAM_LoBo/blob/a759c0a330660a8673eac439168cd3995f913c41/MicroPython_BUILD/components/micropython/esp32/mpconfigport.h#L334).

kisvegabor, 2018-12-02:
I found it. I needed to add `#define MICROPY_MODULE_BUILTIN_INIT 1` to `mpconfigport.h`. This define was complately missing from the unix port.

kisvegabor, 2018-12-02:
The updated SDL project with mouse support is here: https://drive.google.com/open?id=1YjuaEFBrj1E0OVYZ_fX2wPvoaC_4Gw42

amirgon, 2018-12-02:
@kisvegabor 
> I found it. I needed to add `#define MICROPY_MODULE_BUILTIN_INIT 1` to `mpconfigport.h`. This define was complately missing from the unix port.

Nice. I wasn't aware of this macro.

What do think would be the minimal set of additional features Micropython-lvgl-binding should have for a PR?  

I was thinking of:

- Callbacks (`lv_action_t`)
- Styles
- Specific structs such as `lv_color_t`

Also, I think we should add `gen_mpy.py` script to the Makefile. It would generate different outputs depending on `v_conf.h` (and also any change in the lvgl obj H files), so I don't think we want `lv_mpy.c` commited in git, in the long run.

kisvegabor, 2018-12-03:
Thank you for the comments. I updated my previous post.

Yes, I agree if actions, styles, and structures would be added we can already add it to lvgl. It will allow getting feedback from people. I'll add it the PC simulator too in a cleaner way than my current SDL project :)

However, I think we can commit `lv_mpy.c` too. If the parts would be protected with `#if`s according to `lv_conf.h` we can leave it. Or why not leave it? Is there any disadvantage of leaving it in the code? IMO making it visible and instantly accessible is a good point.

amirgon, 2018-12-03:
@kisvegabor I think we can leave `lv_mpy.c` only as an example. The problem is that if the user changes `lv_conf.h` and removes some feature, `lv_mpy.c` will stop compiling as it assumes this feature is present.
Adding `#if`s according to `lv_conf.h` is hard to do automatically in `gen_mpy.py` . The reason is that the C parsing can only be done after C-preprocessing, after which these `#if`s are no longer present in the code.

kisvegabor, 2018-12-03:
@amirgon  I see. In this case, as you suggested, let's leave a full featured `lv_mpy.c` as an example.

embeddedt, 2018-12-03:
@kisvegabor @amirgon I think leaving an autogenerated file in Git is a bad idea. Personally I prefer to be able to compile LittlevGL+Micropython without having changes in my working tree.

https://stackoverflow.com/a/894042

amirgon, 2018-12-03:
@embeddedt I only meant leaving it there as an example, think of it as a text file for reference which is not meant to be compiled.  
Generated C files are not like obj files. You may want to browse `lv_mpy.c` to see which functions were generated, which were not and why (it contains many comments, for this purpose) even before you sync from git and build your project.   
I still think we should have a makefile and run `gen_mpy.py` as part of the build. You will not have changes in your working tree because the committed example will not be committed to the same directory where the file is generated.

C47D, 2018-12-03:
@amirgon How "easy" should be to also support [Circuitpython](https://github.com/adafruit/circuitpython)?

amirgon, 2018-12-04:
@C47D 
> @amirgon How "easy" should be also support [Circuitpython](https://github.com/adafruit/circuitpython)?

I'm not familiar with Circuitpython, but from what I read it is a derivative of Micropython. The differences from Micropython seem [small enough](https://circuitpython.readthedocs.io/en/latest/README.html#differences-from-micropython) such that you might be able to use lvgl Micropython binding **as is** without a change! (I didn't see any difference in the way new extensions are added).  

Why don't you try and let us know how it goes?

kisvegabor, 2018-12-06:
In case of actions `LV_RES_INV` means that the object is deleted in the action and it's invalid when the action returns. (Use case: a button is pressed to change screen -> the current screen is deleted along with the button, and the new screen is created in the action). However, as you pointed out if the action is called later the object will be still sure valid even if it's deleted later. 

The other functions might have a meaningful return value. `lv_anim_path_t` is a good example for that. However, I think it's not a big issue if custom animation paths or image decoder interfaces can't be declared in Micropython. IMO, only actions will be used in 99.9% of the cases. 

The ultimate solution would be to stop C code while the Micropython callback is executed but you said: " I don't think we want to do that". Do you concern because of the delay in execution? (I don't know how big it is)

By the way, how should we call (and/or protect) `lv_task_handler()` from the scheduled call of actions? I mean if an action LittlevGL related functions are executed while `lv_task_handler` also works with the same resources things will be messed up. In "normal" case you should use a mutex around `lv_task_handler` and your `lv_...` function calls. Or is Micropython thread-safe in this regard?

kisvegabor, 2018-12-06:
@amirgon 
I'll be off from work until next Friday. See https://github.com/littlevgl/lvgl/issues/543#issuecomment-445067815

Until Friday, maybe it would be more practical to deal with styles if too many questions arise about actions and callback. Styles seems more straightforward to me.

amirgon, 2018-12-06:
> I'll be off from work until next Friday. See [#543 (comment)](https://github.com/littlevgl/lvgl/issues/543#issuecomment-445067815)

Congratulations! Good luck to you and your wife with your new baby!

kisvegabor, 2018-12-06:
@amirgon Thank you :)

amirgon, 2018-12-14:
@kisvegabor 
> The ultimate solution would be to stop C code while the Micropython callback is executed but you said: " I don't think we want to do that". Do you concern because of the delay in execution? (I don't know how big it is)

In the general case, we can't assume the platform is multi-threaded. Micropython can run on a single thread and schedule its action using its internal scheduler. In such case you are not allowed to block ("stop") the C code.  
Even if you are in a multi-threaded platform, Micropython usually runs in a single thread. In such case you are not allowed to run a Micropython callback from another thread (AFAIK).

> By the way, how should we call (and/or protect) `lv_task_handler()` from the scheduled call of actions? I mean if an action LittlevGL related functions are executed while `lv_task_handler` also [...]

This is a general problem, not related only to Micropython, right?  
How do you protect it today, without Micropython?  
I'm using FreeRTOS and I call `lv_task_handler` periodically from another task (thread).  Should I protect every call to lvgl by mutex (as well as  lv_task_handler) to make sure they never preempts each other?

---

### So here is a quick update: I've added *limited* support for actions (callbacks).
Here is an example I just ran on my board:
```python

>>> import lvgl
ILI9341 initialization.
Enable backlight.
>>> b = lvgl.btn(lvgl.scr_act())
>>> b.align(None, lvgl.ALIGN.CENTER, 0, 0)
>>> def f(arg):
...       print('Button clicked')
...     
...     
... 
>>> b.set_action(b.ACTION.CLICK, f)
>>> 
>>> 
>>> Button clicked
Button clicked
Button clicked
Button clicked
Button clicked

```

Limitations:

- It uses the [internal Micropython scheduler](https://docs.micropython.org/en/latest/reference/isr_rules.html#using-micropython-schedule) (calls [`mp_sched_schedule`](http://docs.micropython.org/en/v1.9.3/pyboard/library/micropython.html#micropython.schedule) to schedule a callback).  
Apparently the Linux port of micropython does not enable the scheduler. I tried (naively) to enable it and the thing crashed. I'm not sure why and what is the effort to support this on the Linux port.

- Callbacks return **before** the action actually executes, so we assume actions never fail. (This is also related to the usage of the Micropython scheduler).

- It only supports callbacks with certain prototype. (need to return `lv_res_t`, first argument of callback function must be `lv_obj_t`, all arguments are convertible to python). Luckily, action callbacks (`lv_action_t`, `lv_btnm_action_t` , `lv_tabview_action_t`) use the convention and are supported.

- It only supports **one** callback per object instance. So don't try to register more than one action for the same button. This is in line with #316 for v6.0 where you plan single callback for each object instance anyway.

kisvegabor, 2018-12-14:
@amirgon 
> I'm using FreeRTOS and I call lv_task_handler periodically from another task (thread). Should I protect every call to lvgl by mutex (as well as lv_task_handler) to make sure they never preempts each other?

Yes, you should use a mutex. See: https://doc.littlevgl.com/#Porting (Using with an operating system
 section)

**Actions**
As the return value is used only to indicate that the object is deleted in the action, in `v6.0` I can add some improvements to automatically detect if the object was deleted. So the library will know if the object was deleted in the action and handle it as invalid. Would it solve the issue?

kisvegabor, 2018-12-14:
> To avoid a mutex, I would like to use Micropython scheduler same way as callbacks (call mp_sched_schedule to schedule a call to lv_task_handler every X milliseconds).

It sounds good but you said `mp_sched_schedule` is not always enabled. Like it's not enabled in the unix build either. 

> Now I'm always returning LV_RES_OK (when mp_sched_schedule succeeds) . How would that affect the user?

If the action callback is executed later i.e. not when you click the button then it still should work. However, you can try it if you call `lvgl.obj_del(btn)` in the action callback. You should set `LV_MEM_ADD_JUNK   1` in `lv_mem.c` to fill the freed memories with junk to be sure the system fails is it wants. :)

amirgon, 2018-12-15:
@kisvegabor 
> It sounds good but you said `mp_sched_schedule` is not always enabled. Like it's not enabled in the unix build either.

I think we must enable Micropython scheduler in order to use Actions. Since Action is a Micropython callback, it cannot be called from another thread or in the middle of another Micropython command, so it must be scheduled correctly. For that reason I think we should provide Micropython lvgl biding only under the assumption that Micropython scheduler **is enabled**.  
If we have the scheduler enabled for Actions, we might as well use it for `lv_task_handler` calls instead of using a mutex on every lvgl Micropython extension function.  

> BTW, in lack of `mp_sched_schedule` in the unix port I won't be able to test the actions on PC, right?

I'm not sure about this. When I tried enabling it (setting `MICROPY_ENABLE_SCHEDULER` macro) it crashed, but I didn't try to debug it much.  
On the other hand, I noticed that on the unix Micropython port there are has several mpconfigport.h files that can be used, [one of them](https://github.com/micropython/micropython/blob/39eef270831323b77445e0458c9357d38e23658a/ports/unix/mpconfigport_coverage.h#L37) defines `MICROPY_ENABLE_SCHEDULER` as 1, so it may be possible to enable the scheduler on the unix port after all. I think it's worth another try.

amirgon, 2018-12-15:
@kisvegabor 
### Good news!
I gave it another try, and the unix port supports the Micropython scheduler. I was able to run [my example of Action above](https://github.com/littlevgl/lvgl/issues/557#issuecomment-447175813) with your unix port!  
Here are the main issues I came across:

- The crash was not related to Micropython really. This was because in your code you changed `lv_mpy.c` and called `lv_init` in the generated file. Once I re-generated `lv_mpy.c`, `lv_init` was no longer called and that was the reason for the crash. So please.... don't change generated files....

- Needed to add a periodic call to `mp_sched_schedule` which runs `lv_task_handler`.

- Needed to change the read from stdin to nonblocking and add calls to `mp_handle_pending` which were missing in the unix port. Otherwise, Micropython is not called upon async events.

- I used the `mpconfigport_coverage.h` file for configuring Micropython with scheduling.

- I use a slightly older version of Micropython with my ESP32 board. To make `lv_mpy.c` compile I needed to change the calls to `mp_sched_schedule` (remove the last parameter). I'm planning to change `gen_mpy.py` to support both prototypes of `mp_sched_schedule`.

---

I suggest that instead of a zip file on google drive, add a project on github. Now I would like to show you my changes to your source code and I can't really use `git diff`. I'll try to manually paste the interesting parts, hopefully this would help you run it on your side:

Make command:
```Make
make CFLAGS_EXTRA='-DMP_CONFIGFILE="<mpconfigport_coverage.h>" -DMICROPY_USE_READLINE=1' DEBUG=1 -j4
```

Changes on `lvgl/mpy/lv_mpy_hal.c`:
```C
#include "py/runtime.h"
#include "py/obj.h"

STATIC mp_obj_t mp_lv_task_handler(mp_obj_t arg)
{
    lv_task_handler();
    return mp_const_none;
}

STATIC MP_DEFINE_CONST_FUN_OBJ_1(mp_lv_task_handler_obj, mp_lv_task_handler);

/**
 * A task to measure the elapsed time for LittlevGL
 * @param data unused
 * @return never return
 */
static int tick_thread(void * data)
{
    (void)data;

    while(1) {
        SDL_Delay(5);   /*Sleep for 5 millisecond*/
        lv_tick_inc(5); /*Tell LittelvGL that 5 milliseconds were elapsed*/ 
	mp_sched_schedule((mp_obj_t)&mp_lv_task_handler_obj, mp_const_none);
    }

    return 0;
}

...

void lv_mp_init(void)
{
    lv_init();
    hal_init();
}
```

Changes on `unix_mphal.c` (`inputAvailable` taken from [here](https://stackoverflow.com/a/1280659/619493))

```C
static bool inputAvailable()  
{
  struct timeval tv;
  fd_set fds;
  tv.tv_sec = 0;
  tv.tv_usec = 0;
  FD_ZERO(&fds);
  FD_SET(STDIN_FILENO, &fds);
  select(STDIN_FILENO+1, &fds, NULL, NULL, &tv);
  return (FD_ISSET(0, &fds));
}

int mp_hal_stdin_rx_chr(void) {
    unsigned char c;
#if MICROPY_PY_OS_DUPTERM
    // TODO only support dupterm one slot at the moment
    if (MP_STATE_VM(dupterm_objs[0]) != MP_OBJ_NULL) {
        int c;
        do {
             c = call_dupterm_read(0);
        } while (c == -2);
        if (c == -1) {
            goto main_term;
        }
        if (c == '\n') {
            c = '\r';
        }
        return c;
    } else {
        main_term:;
#endif
	while (!inputAvailable())
	{
	    mp_handle_pending();
	    usleep(5);   
	}
        int ret = read(0, &c, 1);
        if (ret == 0) {
            c = 4; // EOF, ctrl-D
        } else if (c == '\n') {
            c = '\r';
        }
        return c;
#if MICROPY_PY_OS_DUPTERM
    }
#endif
}
```

embeddedt, 2018-12-15:
@amirgon I'm not familiar with MicroPython's scheduler. Does it require the platform to have thread support?

embeddedt, 2018-12-16:
@amirgon 

> No. Micropython runs on a single thread no matter if MicroPython's scheduler is enabled or not.

No problem then. :)

amirgon, 2018-12-17:
@kisvegabor 

> I've created a repo here: https://github.com/kisvegabor/lv_mpy

Fixed and tested that Actions are working. See [PR](https://github.com/kisvegabor/lv_mpy/pull/1#issue-239283680).

kisvegabor, 2018-12-18:
Merged, thank you! 

I suggested releasing `v5.3` (#543) and start to work on `v6.0`. So we will have more freedom to customize lvgl if needed.

amirgon, 2018-12-18:
@kisvegabor 

> I suggested releasing `v5.3` (#543) and start to work on `v6.0`. So we will have more freedom to customize lvgl if needed.

How about releasing it on `v5.3` before I add support for styles and structs? (actually I'm going to add support for generic structs first, as it's the general case of styles, hopefully next week), or do you want to postpone Micropython for `v6.0`? What is the schedule for `v6.0` release?  

The advantage of releasing it early is letting the users know about this, and giving them the opportunity to try it out even if some features are still missing.

What do we do about Makefile/build script that would run `gen_mpy.py`? Or do you suggest the user would run it manually, not as part of an automatic build process?

turoksama, 2018-12-18:
I vote new branch for python implementation.

kisvegabor, 2018-12-18:
> How about releasing it on v5.3 before I add support for styles and structs? (actually I'm going to add support for generic structs first, as it's the general case of styles, hopefully next week), or do you want to postpone Micropython for v6.0?

Good point, I agree to add Micropython support in `v5.3` to allow people to try it. 

> What is the schedule for v6.0 release?

There is no strict deadline. But I'm sure it will take at least 3 months to add all the planned features.

> What do we do about Makefile/build script that would run gen_mpy.py? Or do you suggest the user would run it manually, not as part of an automatic build process?

The call of `gen_mpy.py` should be added to the Micropython's Makefile, right? So when you build Micropython for e.g. ESP32 or UNIX always get an up to date binding.

Another thing:
@rreilink added "normal" [Python binding to LittlevGL](https://github.com/rreilink/pylvgl). Now he is [creating an example](https://github.com/rreilink/pylvgl/issues/1) to test the binding.

So if we'd have more bindings I suggest having an `lv_bindings` folder where we can add the interface files and scripts.

kisvegabor, 2018-12-18:
@turoksama 
> I vote new branch for python implementation.

My view was to release Micropython binding with `v5.3` as it is and update it in `dev-6.0` after the release of `v5.3`. What would be the advantage of using a separate branch instead of keeping things in one place? Because a bug in the binding does not break LittelvGL itself.

turoksama, 2018-12-18:
It sounds good to bind as what you said.
One thing i concerned was integrity of the library, and it seems too much.

amirgon, 2018-12-18:
@turoksama 

> One thing i concerned was integrity of the library, and it seems too much.

What do you mean "integrity of the library"? Let's say there is some bug. lvgl is not affected since Micropython binding does not change anything in lvgl sources, it's just a python script that creates a `.c` file. So the worst thing that can happen is that someone who tries to use Micropython binding would see some problem (specific to Micropython), **and that's a good thing** as he would report this and we would fix that early.

amirgon, 2018-12-18:
> Another thing:
> @rreilink added "normal" [Python binding to LittlevGL](https://github.com/rreilink/pylvgl). Now he is [creating an example](https://github.com/rreilink/pylvgl/issues/1) to test the binding.

I wish we know about this earlier... :unamused:
I don't see a point in making the same work twice. What's the point in two different implementations that achieve the same goal?

@kisvegabor I don't think it's a good thing for lvgl to integrate two different implementations for Micropython. How would the user select one? Why maintain two scripts instead of one? Why provide two different python interfaces for the same library?

amirgon, 2018-12-18:
@kisvegabor 
Python and Micropython, from the user perspective, are not so different. At least not from the core language aspect.
A user that uses Python binding and wants to migrate to Micropython would have to change his code since the API looks different, things are implemented differently etc.  

Ideally we could think of **one** python script that could generate **both** Python and Micropython binding, and most of its code would be common for both.  
Benefits of such script are unified python API for lvgl from the user perspective, single source code to add features / fix bugs related to binding etc.

rreilink, 2018-12-18:
@amirgon, @kisvegabor 

> I wish we knew about this earlier... 
Yes, I do to 😒

> Ideally we could think of **one** python script that could generate **both** Python and Micropython binding, and most of its code would be common for both.
> Benefits of such script are unified python [...]

I totally agree.

I have not looked into the MicroPython code generation, I will do so this week, and propose a way to integrate the two bindings-generators to achieve what you propose: unified API and the most commonality in the source code.

kisvegabor, 2018-12-18:
@amirgon @rreilink If you guys believe that the same script can work for Micropython and Python too, of course, it would be the best! :smile:

[141 more replies]

closed 2019-02-25 by amirgon