lvgl/lvgl #558: [Question] How to use free_num with the buttons on a Message box?

mharizanov, 2018-11-17:
Hello,

I use multilanguage GUI, and I can't rely on button's text to identify the button being pressed, as it can have multiple different values based on the GUI language used.

What is the proper way to use lv_obj_set_free_num on the buttons on a message box, and then use that in the click action callback?

Thanks!

embeddedt, 2018-11-17:
@mharizanov `lv_obj_set_free_num` doesn't work on button matrices (which `lv_mbox` uses) because they are not really composed of multiple buttons. However, I read the area of `lv_btnm.c` that deals with button click handling, and this *might* work (untested):

```c
lv_btnm_ext_t *btnm_ext = lv_obj_get_ext_attr(btnm);
if(btnm_ext->btn_id_pr == 0) { } /* First button */
else if(btnm_ext->btn_id_pr == 1) { } /* Second button */
```

@kisvegabor If it works I think this would be a good API to add.

mharizanov, 2018-11-17:
Yes, that worked. Thanks! I think an API call makes sense for use cases like mine.

closed 2018-11-17 by mharizanov

kisvegabor, 2018-11-19:
I added the API extension here: #563
Please have a look at it.

mharizanov, 2018-11-20:
Thanks Gabor, you rock!