lvgl/lvgl #561: Cont fit and align cause hardfault

kl-cruz, 2018-11-18:
I am investigating this issue now, but I found it twice last time. I just want to keep it here.

First we have parent object called ctrl_dlyx. Now, creates button with label and align it:
```
    lv_obj_t * btn1 = lv_btn_create(ctrl_dlyx, NULL);
    lv_btn_set_style(btn1, LV_BTN_STYLE_REL, p_style_button);
    lv_btn_set_fit(btn1, true, true); /*Enable resizing horizontally and vertically*/
    lv_obj_set_free_num(btn1, 1);   /*Set a unique number for the button*/
    lv_btn_set_action(btn1, LV_BTN_ACTION_CLICK, btn_click_action);

    /*Add a label to the button*/
    label = lv_label_create(btn1, NULL);
    lv_obj_align(btn1, ctrl_dlyx, LV_ALIGN_IN_RIGHT_MID, 0, 0);
```
I am using encoder mode. Even first focus on this button causes hardfault. Style's text (default object's style, lv_style_pretty) configuration is overwritten.
When I disable fit option everything works well.

I created memory watch on lv_style_pretty's text struct. First picture shows situation when loop starts to write something over the memory region. Second one shows overwritten field in style. I've got similar issue with align function and (as I remember) lv_obj_set_size function (firstly set align then set size, otherwise, set_size and aligned - it worked). I'll try to reproduce it with simpler example and upload it here.
[image]
[image]

For now I know that for safety reasons it is the best idea to call align at the end of code which setups our control. Additionally, avoid to use fit and align together.
As I wrote, I'll investigate it deeply to eliminate my influence in application's execution.

turoksama, 2018-11-19:
please check style variable p_style_button if is static or global.

embeddedt, 2018-11-19:
Is the `lv_style_t` structure pointed to by `p_style_button` going out of scope? `lv_style_t` variables must either be:

1. declared `static` inside a function.
2. global variables.

kl-cruz, 2018-11-19:
I hope no. I am using global variables

kl-cruz, 2018-11-19:
Okay, I'll do it