lvgl/lvgl #571: Scrollable Grid Control

csteaderman, 2018-11-21:
A scrollable grid control would be a very useful addition.

embeddedt, 2018-11-21:
@csteaderman This could be built already with a combination of `lv_cont` and `lv_page` I think.

kisvegabor, 2018-11-22:
I'm not sure you are thinking about something like this but I'm working on a new object type, called `lv_tileview`. It looks like this now:

[image]

csteaderman, 2018-11-23:
Thanks for the very quick responses! They are giving me some good inspirations for how I might use LittlevGL. I'm afraid, though, that my initial question was not as precise as it should have been to describe exactly what I'm looking for. Since a picture is worth a thousand words, what I'm looking for is something like...

[image]

From using grid/table controls from other frameworks, I can imagine one of the biggest challenges being a way to dynamically load data when the number of rows and/or columns becomes large. For my initial implementation, I don't think that I'll need to embed controls in my grid, other than labels. All interaction will be static, only scrolling of the entire table's contents.

embeddedt, 2018-11-23:
@csteaderman I think you could implement a static version using my described method above. You would have to create checkbox and label objects inside the container.

kisvegabor, 2018-11-24:
I agree the suggestion of @embeddedt should work. And it's very flexible. Its only disadvantage is the memory usage.

To get something more efficient a new object type should be created where rows an columns are drawn from plain text.

csteaderman, 2018-11-24:
So, when I use the example from @embeddedt, setting the number of columns to three using the following code:
```C
void grid_create(void)
{
	static char buf[32];
	lv_obj_t *page = lv_page_create(lv_scr_act(), NULL);
	lv_obj_set_size(page, 500, 400);
	lv_obj_set_pos(page, 10, 10);
	lv_obj_t *cont = lv_page_get_scrl(page);
	lv_cont_set_layout(cont, LV_LAYOUT_GRID);
	for(int i = 0; i < 100; i++)
	{
		snprintf(buf, 32, "%d", i);
		lv_obj_t *btn = lv_btn_create(page, NULL);
		lv_style_t * style = lv_btn_get_style(btn, LV_BTN_STYLE_REL);
		lv_obj_set_size(btn, (lv_obj_get_width(cont) / 3) - (style->body.padding.hor), 30);
		lv_obj_t *label = lv_label_create(btn, NULL);
		lv_label_set_text(label, buf);
	}
}
```
I get:
[image]

I don't need or want buttons, so I changed the code to simply use labels with the following code:

```C
void grid_create_label(void)
{
	static char buf[32];
	lv_obj_t *page = lv_page_create(lv_scr_act(), NULL);
	lv_obj_set_size(page, 500, 400);
	lv_obj_set_pos(page, 10, 10);
	lv_obj_t *cont = lv_page_get_scrl(page);
	lv_cont_set_layout(cont, LV_LAYOUT_GRID);
	for(int i = 0; i < 100; i++)
	{
		snprintf(buf, 32, "%d", i);
		lv_obj_t *lbl = lv_label_create(page, NULL);
		lv_style_t * style = lv_label_get_style(lbl);
		lv_obj_set_size(lbl, (lv_obj_get_width(cont) / 3) - (style->body.padding.hor), 30);
		lv_label_set_text(lbl, buf);
	}
}
```

and I get:
[image]

Am I not able to set the size of a label?

embeddedt, 2018-11-24:
@csteaderman Try setting the size after setting the text.

csteaderman, 2018-11-24:
No difference.

kisvegabor, 2018-11-25:
@csteaderman By the default, the label's size will be overwritten by the library according to the size of the text. I suggest setting `lv_label_set_long_mode(label, LV_LABEL_LONG_BREAK);`

I modified @embeddedt's example:
```c

void grid_create_label(void)
{

    static lv_style_t scrl_style;
    static lv_style_t cell_style;

    lv_style_copy(&scrl_style, &lv_style_plain);
    scrl_style.body.padding.hor = 0;
    scrl_style.body.padding.ver = 0;
    scrl_style.body.padding.inner = 0;

    lv_style_copy(&cell_style, &lv_style_plain);
    cell_style.body.padding.hor = 0;
    cell_style.body.padding.ver = 0;
    cell_style.body.padding.inner = 0;
    cell_style.body.border.width = 1;
    cell_style.body.border.color = LV_COLOR_BLACK;

    lv_obj_t *page = lv_page_create(lv_scr_act(), NULL);
    lv_page_set_style(page, LV_PAGE_STYLE_SCRL, &scrl_style);
    lv_page_set_style(page, LV_PAGE_STYLE_BG, &lv_style_transp_tight);
    lv_obj_set_size(page, LV_HOR_RES, LV_VER_RES);
    lv_page_set_scrl_layout(page, LV_LAYOUT_GRID);

    lv_coord_t lbl_w = (lv_page_get_fit_width(page) / 3) - scrl_style.body.padding.inner * 2;

    static char buf[32];
    for(int i = 0; i < 100; i++)
    {
        snprintf(buf, 32, "%d", i);
        lv_obj_t *lbl = lv_label_create(page, NULL);
        lv_label_set_long_mode(lbl, LV_LABEL_LONG_BREAK);
        lv_obj_set_size(lbl, lbl_w, 30);
        lv_label_set_text(lbl, buf);
        lv_label_set_style(lbl, &cell_style);
        lv_label_set_body_draw(lbl, true);
        if(i % 3 == 0) lv_label_set_align(lbl, LV_LABEL_ALIGN_LEFT);
        if(i % 3 == 1) lv_label_set_align(lbl, LV_LABEL_ALIGN_CENTER);
        if(i % 3 == 2) lv_label_set_align(lbl, LV_LABEL_ALIGN_RIGHT);
    }
}

```
[image]

csteaderman, 2018-11-25:
Very nice! At one point I tried LV_LABEL_LONG_BREAK, but I had other things wrong so it didn't work.

csteaderman, 2018-11-25:
I'm thinking about possibly creating a formal Data Grid control. Do you think there would be interest?

kisvegabor, 2018-11-25:
@csteaderman It really would be great to have a dedicated gridview object. It was already asked a lot of times. Would you be interested in adding this new object type?

csteaderman, 2018-11-29:
@kisvegabor I believe that I am willing to try to implement a Grid/Table view. Since the control will first be used on a project that I am developing, I will limit the features to the minimum that I need for that project. Then I will consider adding more features and customizations. I was thinking about starting with a preliminary API and functional description that I would present for feedback. Since I am very new to LittlevGL, I will need some help with some of the finer points, as long as that is acceptable. 

For my initial implementation, I may need to signal when the user selects a row. One thing that I already think I will need is a click callback for the label control. Is it available?

kisvegabor, 2018-11-29:
@csteaderman Glad to hear that you are interested! Actually, it turned out that I also need a table view in once my project. So I've just added the basics of table view into the `table` branch. It is not scrollable but you can place it on a page.
It looks like this now:

The current features are:
- set row and column count
- set cell content
- merge cell with the cell on the right
- align the label horizontall to left/center/middle
- set column width

Still missing:
- align the labels vertically to center
- add style setting API (background, cell, first column, first row)
- API to add set column widths

[image]

Test code:
```c
    lv_obj_t * table = lv_table_create(lv_scr_act(), NULL);
    lv_obj_set_size(table, 200, 200);
    lv_table_set_col_cnt(table, 3);
    lv_table_set_row_cnt(table, 4);
    lv_table_set_cell_format(table, 0, 0, LV_TABLE_CELL_ALIGN_RIGHT);
    lv_table_set_cell_value(table, 0, 0, "as");
    lv_table_set_cell_value(table, 2, 2, "xy\nab");
    lv_table_set_cell_format(table, 2, 2, LV_TABLE_CELL_ALIGN_CENTER);

    lv_table_set_cell_value(table, 1, 1, "merged in");
    lv_table_set_cell_merge_right(table, 1, 1, true);
    lv_table_set_cell_format(table, 1, 1, LV_TABLE_CELL_ALIGN_CENTER);
```
You need to enable the table in `USE_LV_TABLE    1`

Selecting a row or cell is not supported right now but we can add it! I think it would be a good starting point for you to add this feature! How does it sounds to you?

kisvegabor, 2018-11-29:
Se this PR: https://github.com/littlevgl/lvgl/pull/592

kisvegabor, 2018-12-13:
I updated the table object. Now it works but clicking and selecting rows/cells are not supported. 

If you'd like to add these improvements, please comment here and reopen this issue.

closed 2018-12-13 by kisvegabor

CaptainSimbo, 2019-09-16:
Hi,

I'm using the Table object (within a page) and it's doing a great job of presenting data from a database query but I could really use a callback on cell select (e.g. in my case to toggle whether or not a database point should be included in further processing or potentially to enter a new value for a cell). 

Following this thread has anyone made any progress on a feature like this? 

My guess would be that the table would almost have to be 're-rendered' to find out the regions occupied by each cell and see if that cell is under the pointer. (Alternatively I guess each render could remember the row offsets after each render and use that data with the column widths to locate the cell.)

Cheers,

Simon.

embeddedt, 2019-09-16:
> anyone made any progress on a feature like this?

The latest development version of LittlevGL still doesn't have support for receiving events when cells are selected, so unless someone has done it in a private project, it's likely that no one has implemented it thus far.

> I guess each render could remember the row offsets after each render and use that data with the column widths to locate the cell.

This would be the better way of implementing it, in my opinion. The other method might harm performance as a result of needing to recalculate the table cells every time someone clicks it.

kisvegabor, 2019-09-16:
> This would be the better way of implementing it, in my opinion. The other method might harm performance as a result of needing to recalculate the table cells every time someone clicks it.

I'd vote to calculate the clicked cell on the fly instead of storing additional data because clicks shouldn't happen so often to really affect the performance. `get_row_height` should help to find the clicked row. The column widths are stored directly in `col_w[]` in `lv_table_ext_t`.

@CaptainSimbo let us know if you are interested in implementing it.

CaptainSimbo, 2019-09-16:
Ah ha ! I didn't realise that there was a 'get_row_height'  ( I didn't see it in the online API documentation and typing 'lv_get_row_height' drew an intellisense blank too.  :-)

I'll experiment with that to see if I can get somewhere. 

One fly in the ointment might be that the table is in a page - I'll need to find the (relative) click coordinate in the page view and work out how far the table is currently scrolled.  ( so I can find the click coordinate relative to the origin of the table.).

kisvegabor, 2019-09-17:
> I'll experiment with that to see if I can get somewhere.

Great!

I can imagine functions like 
```c
lv_table_get_row_on_point(table, y);
lv_table_get_col_on_point(table, x);
```
Where `x` and `y` are realtive to the table. So it in the practice it would work like:
```c
row = lv_table_get_row_on_point(table, cliked_y - table->coords.y1);
col = lv_table_get_col_on_point(table, cliked_x - table->coords.x1);
```

What do you think?

CaptainSimbo, 2019-09-17:
Yep that would do the job and I can see that rather than always getting the row/column as a pair there would be many uses for just selecting a row or column.

The difficulty I'm having is trying to get 'clicked_x' and 'clicked_y' relative to the table origin when that table is hosted within a 'page' . 

I think it fair to say that, apart from a few cases, a table is often going to be larger than the available screen space and so will need to be 'scrollable'.

To raise the various pointer events ( I've attached a callback to the table object and it's nicely getting called) presumably littleVGL has had to examine the pointer position and see if it lies within the bounds of the table. 

I can see how the origin of the table changes as it's scrolled within the page changes (e.g. 'y1' goes negative when you scroll down the table) so that origin will have to be factored in to the coordinates I get get back from my query of the pointer device -

 **lv_indev_get_point(mindev, &pos);**

To break the problem down a little bit I'm going to try and pick up and map the click coordinates on a 'simple' i.e. not hosted by a page table.

Hopefully the problem then will be just one of incorporating the active scroll offsets.

kisvegabor, 2019-09-18:
> I can see how the origin of the table changes as it's scrolled within the page changes (e.g. 'y1' goes negative when you scroll down the table) so that origin will have to be factored in to the coordinates I get get back from my query of the pointer device -

The code example I have shown is not working?
```
row = lv_table_get_row_on_point(table, cliked_y - table->coords.y1);
col = lv_table_get_col_on_point(table, cliked_x - table->coords.x1);
```

Note that `table->coords.x1` is the absolute coordinate on the screen and not relative to the parent. `lv_indev_get_point` also gives absolute coordinates.

CaptainSimbo, 2019-09-23:
(Sorry for the delay)

>The code example I have shown is not working?

Yes it is! :-) and I found out why I was having problems - for some reason (I blame Brexit) I was re-scaling the input coordinates that I picked up from the pointer and this was resulting in erroneous 'clicked_x' & 'clicked_y' . 

Will you be exposing the 'get_row_height' function (much like 'lv_table_get_col_width') in a later update? (Perhaps there's a good reason not to?)

kisvegabor, 2019-09-23:
> Yes it is! :-) and I found out why I was having problems - for some reason (I blame Brexit) I was re-scaling the input coordinates that I picked up from the pointer and this was resulting in erroneous 'clicked_x' & 'clicked_y' .

Fortunately, you outsmarted Brexit :smile:

> Will you be exposing the 'get_row_height' function (much like 'lv_table_get_col_width') in a later update? (Perhaps there's a good reason not to?)

I saw no reason to make this function public.  Can you mention a use-case when it'd be useful? 
`lv_table_get_col_width` is public because it has `lv_table_set_col_width` pair.