lvgl/lvgl #572: LVGL ARM Linux Frame Buffer configuration issue.

martincarra, 2018-11-21:
Hi all,
Thanks in advance for reading this. 
I am trying to use the library in a custom TI Jacinto-6 board to do some basic operations with the display.   Turns out that the driver compiles seamless but the "hello world" shows in the display totally messed up.
I am 100% positive sure that the display driver is working as expected since using different tools (like fbsplash from busybox) are working perfectly.

Looks like the configuration is incorrect. In case anyone wants to take a peek I am including a photo of how the PoC app is being shown at the display and the configuration files. 
The display is a 800 x 480px @133dpi 24-Bits 
Any clue?

Thanks in advance,
Martín
[lv_config_files.zip](https://github.com/littlevgl/lvgl/files/2605318/lv_config_files.zip)
[image]

embeddedt, 2018-11-21:
Here are my thoughts:

1. Check that color depths match between LittlevGL and the linux frame buffer.
2. Possibly your orientation is wrong (i.e. x/y values swapped?)

kisvegabor, 2018-11-21:
Hi,

LittlevGL doesn't support 24 bit color depth, only 32 bit.  To use a 24 bit framebuffer you need to convert the color array in your `disp_flush` function. Although the color array is `const` you can use it to write back the converter pixels.

martincarra, 2018-11-22:
Solved! 
The problem was related to the stride of the frame buffer.
fbdev_init
    // Figure out the size of the screen in bytes
    screensize = finfo.line_length * vinfo.yres;

In fbdev_flush
/*32 or 24 bit per pixel*/
    if(vinfo.bits_per_pixel == 32 || vinfo.bits_per_pixel == 24) {
        uint32_t * fbp32 = (uint32_t *)fbp;
        int32_t x;
        int32_t y;
        for(y = act_y1; y <= act_y2; y++) {
            for(x = act_x1; x <= act_x2; x++) {
                **location = (x + vinfo.xoffset) + (y + vinfo.yoffset) * finfo.line_length / 4 ;**
                fbp32[location] = color_p->full;
                color_p++;
            }

            color_p += x2 - act_x2;
        }
    }

that appears to solve all the problems related to the mapping to the fb memory

kisvegabor, 2018-11-23:
@martincarra 
So `vinfo.xres` should be replaced with `finfo.line_length / 4` in:
`location = (x + vinfo.xoffset) + (y + vinfo.yoffset) * vinfo.xres;`

martincarra, 2018-11-23:
To summarize the modifications:
In the function **fbdev_init**:
    The **screensize** calculation shall be:
    **screensize = finfo.line_length * vinfo.yres;**
In the function **fbdev_flush**
     **vinfo.xres** should be replaced with **finfo.line_length / 4** in:
     **location = (x + vinfo.xoffset) + (y + vinfo.yoffset) * vinfo.xres;**

Best regards,
Martín

closed 2018-11-23 by martincarra

kisvegabor, 2018-11-24:
@martincarra 
Thank you. Updated based on you suggestion: https://github.com/littlevgl/lv_drivers/commit/6ff8f3818359a55a349e39363fc4fc88e8937830