lvgl/lvgl #573: Storing images in flash memory

jaythecelt, 2018-11-21:
Hi,

We have several png images to display on our STM32F4 board (2MB of flash / 8MB external RAM) and have run out of flash memory to store them.  We're thinking of writing a compression utility that takes the .c file output from the [online image converter](https://littlevgl.com/image-to-c-array) and generates a similar .c source file but with the compressed image data in the array.  That .c file would be compiled with our app and at runtime the image would be decompressed into external RAM.  

We've looked at the encode/decode function, but we don't have a file system or memory to store the original image.

Is there any support for compressing images in littlevgl?  How have others dealt with this problem?

embeddedt, 2018-11-21:
@jaythecelt Is there any possibility of using a lower color depth for the images? Indexed (palette) images take up less space.

jaythecelt, 2018-11-22:
Yes, we may be able to on some, that's a good suggestion.  The largest ones are small photographic images - but they may look ok as indexed image.

kisvegabor, 2018-11-22:
Have look at this: https://littlevgl.com/blog/50/use-png-images-in-littlevgl section **Test the PNG decoder library with a PNG file stored in ROM**

embeddedt, 2018-12-18:
There was no activity here for 1 month so this issue has been closed. If you continue to have problems please reopen the issue.

closed 2018-12-18 by embeddedt