lvgl/lvgl #574: uint64_t used as pointer address type

embeddedt, 2018-11-21:
I think the use of `uint64_t` as an address on this line and in other locations is nonportable and should be avoided. Maybe `uintptr_t` should be used instead.

https://github.com/littlevgl/lvgl/blob/2c8cf7d7ea341ff524b0d021d69b229357bdc06a/lv_themes/lv_theme.c#L76

C47D, 2018-11-21:
+1 for using `uintptr_t`, AFAIK it's available since C99 tho.

kisvegabor, 2018-11-22:
I agree we can use `uintptr_t`. It's really more elegant.

embeddedt, 2018-11-22:
@C47D I think that most modern embedded compilers support C99, and in the worst case the user can simply `typedef` `uint64_t` themselves.

kisvegabor, 2018-11-22:
@embeddedt 
Will you create a Pull request?

embeddedt, 2018-11-22:
@kisvegabor I will, but it will be later today.

kisvegabor, 2018-11-22:
Great, thank you!

C47D, 2018-11-22:
I can't find the issue were me, another user and @kisvegabor were discussing about using C99 features and leave C90 behind, i guess it's time to do so (if everybody agree).

kisvegabor, 2018-11-23:
@C47D 
I remember that conversation. For maximal compatibility, we should use only the most important and most widely adopted features of C99. For example: `for(int i = 0; i < 10; i++)` is still not working in a lot of compilers.
Anyway, using `uintptr_t` is quite reasonable. And in case of incompatibility, it can be fixed with a typedef (as @embeddedt mentioned)

@embeddedt 
Thank you for the fix!

PR #578 (2018-11-23): Replace uint64_t with uintptr_t in lv_theme.c
This resolves #574.
commit d277e80d (2018-11-23): Replace uint64_t with uintptr_t in lv_theme.c

This resolves #574.
closed 2018-11-29 by kisvegabor