lvgl/lvgl #575: Could you please clarify the meaning of the return values for mbox callback

TMarkivULCRobotics, 2018-11-21:
Hello,
I have a little misunderstanding regarding mbox callbacks.

1) I am confused by return values meanings. In the documentation I found the following text:
[image]

I did not understand this, return ok if mbox is not deleted by whom? In which cases I have to return OK and INV?

2) I think there is obsolete description of the return values in the lv_btnm.h, line 50.
`return LV_ACTION_RES_INV if  the button matrix is deleted else LV_ACTION_RES_OK*/`
must be replaced with :
`return LV_RES_INV if  the button matrix is deleted else LV_RES_OK*/``

Thanks,
Taras

turoksama, 2018-11-22:
Hi,
"return LV_ACTION_RES_INV if the button matrix is deleted else LV_ACTION_RES_OK*/"
As it says.For example, mbox has 2 buttons looks like: OK , Cancel.
Gerneral speaking, when clicked, mbox should be delected by applying ```lv_obj_del(mbox);```
Your action callback function may look like this:
```
lv_res_t mbox_action_callback(lv_obj_t *mbox, const char *txt)
{
  if( strcmp(txt, "OK")==0)
  {
     /*TODO:usr action*/

  }
  else
  {
    /*TODO:usr action*/

  }

  lv_obj_del(mbox);
  return LV_RES_INV;
}
```
Ohterwise, no "lv_obj_del(mbox);" and return LV_RES_OK;

kisvegabor, 2018-11-22:
@turoksama Exactly!