lvgl/lvgl #579: ESP32 & ILI9341

sv1eia, 2018-11-23:
Hi,
Trying to have lvgl working with my ESP32.

Got working ESP32 & ILI9341 with the exact same wiring and pin use for

A) in Arduino IDE with Bodmer's TFT_eSPI and
[https://youtu.be/_MmUxUFjpXg](url)

B) in espressif-idf with Loboris's ESP32_TFT_library
[https://youtu.be/Zarjzt4YLeU](url)

but with littlevGL (git snapshot on 22/Nov) when I tried to run the esp32_ili9341 project from lv_projects got this issue
[https://youtu.be/u4zyP3dpeFY](url)

Is the ESP32 with littlevGL project working or am I missing something?
Let me know if you need any details or files from my setup.

kisvegabor, 2018-11-24:
Hi,

Please attach you `lv_conf.h` and you `disp_flush` function. What is resolution of you display?

sv1eia, 2018-11-24:
Hi,

I have not touched the project's files, nor changed anything but what was needed in order to compile (*.mk) as indicated in your project setup instructions.
So I placed the .mk files, 
then I changed the ili9341.h file to suit my lcd pins
#define ILI9341_HOR_RES	320
#define ILI9341_VER_RES	240

#define ILI9341_DC   2
#define ILI9341_RST  4
#define ILI9341_BCKL 0

and the disp_spi.h file

#define DISP_SPI_MOSI 23
#define DISP_SPI_CLK  18
#define DISP_SPI_CS    5

The LCD is 240x320 2.2"

After that I tried to compile and flash.
The LCD did came up but the output on its screen is garbled, I do see that something is getting changed every couple of seconds but as you see it has a problem.

The lv_conf.h is the one already in your project
[https://github.com/littlevgl/lv_projects/blob/f84fa324db1c78cdbdeb8e173e50c0f9abf3cacd/esp32_ili9341/lv_conf.h#L1](url)

and the disp_flush is the same unchanged from the project
[https://github.com/littlevgl/lv_projects/blob/f84fa324db1c78cdbdeb8e173e50c0f9abf3cacd/esp32_ili9341/main/main.c#L35](url)

So as you see, I did no changes to the ofiginal project it is the same as it resides in the lv_projects, just changed the pins to use for the LCD.
That is why I am asking if this project is still working ok with ESP32 and ILI9341.

amirgon, 2018-11-24:
@sv1eia if you changed the SPI pins on the example, you are using the IOMUX.
In such case you need to initialize these pins. Something like:

    gpio_pad_select_gpio(DISP_SPI_MISO);
    gpio_pad_select_gpio(DISP_SPI_MOSI);
    gpio_pad_select_gpio(DISP_SPI_CLK);

    gpio_set_direction(DISP_SPI_MISO, GPIO_MODE_INPUT);
    gpio_set_pull_mode(DISP_SPI_MISO, GPIO_PULLUP_ONLY);
    gpio_set_direction(DISP_SPI_MOSI, GPIO_MODE_OUTPUT);
    gpio_set_direction(DISP_SPI_CLK, GPIO_MODE_OUTPUT);

    gpio_pad_select_gpio(DISP_SPI_CS);

Also make sure you use hald duplex (`.flags=SPI_DEVICE_HALFDUPLEX`)

sv1eia, 2018-11-25:
@amirgon thank you for the hints.
I have not seen all the code and initialization functions so far.
Your remarks about the SPI speed made me to search and find the relevant setting, it was on 80MHz and I know that none of my tests so far work with this high frequency, they all use mainly ~27MHz.
So, I changed this to a lower number
https://github.com/littlevgl/lv_projects/blob/f84fa324db1c78cdbdeb8e173e50c0f9abf3cacd/esp32_ili9341/drv/disp_spi.c#L54
and the lcd now works just fine!
I did not use the mentioned commands for pin initialization nor the SPI half duplex flag.
I only did the lowering of the frequency and it worked just fine.

Thanks.

embeddedt, 2018-11-25:
Since the issue is solved I will close this issue. If you have questions or continue to have problems feel free to reopen it.

closed 2018-11-25 by embeddedt