lvgl/lvgl #580: the win header problem

TangoZhu, 2018-11-24:
I use LV_WIN_STYLE_HEADER to set win head title as chinese But I found That The Close symbol is missing, only when i put my mouse on the close symbol the close symbol will appear

the following is my code 
three are two problem 
1:the first image is not align 
2:the close symbol is missing 
[image]
[image]

extern lv_font_t MYFONT_24;

char *menu_name[] = {"录像查询","系统信息","系统设置","高级选项","文件备份","关闭系统"};

lv_res_t main_menu_create(lv_obj_t *activebtn)
{

    lv_obj_t * scr = lv_scr_act();

	int16_t distance_scr =64;//图标与屏幕边框的间距
	int16_t distance_obj=64;//图标之间的间距
	int16_t img_width = 104;
    uint8_t j,i;
    
	//SCREEN
	lv_obj_t * btn[6];
	lv_obj_t * img[6] ;
	lv_obj_t * label[6] ;

    close_former_win(activebtn);

	lv_obj_t * win = lv_win_create(scr,NULL);

	static lv_style_t win_style;
    lv_style_copy(&win_style, &lv_style_transp);
    win_style.body.padding.hor = LV_DPI / 4;
    win_style.body.padding.ver = LV_DPI / 4;
    win_style.body.padding.inner = LV_DPI / 4;
    
    lv_win_set_style(win, LV_WIN_STYLE_CONTENT_SCRL, &win_style);
    lv_win_set_layout(win, LV_LAYOUT_OFF);
    lv_win_set_style(win, LV_WIN_STYLE_BTN_REL, &win_style);
   // lv_win_set_style(win, LV_WIN_STYLE_CONTENT_BG, &win_style);
 //   lv_win_set_style(win, LV_WIN_STYLE_CONTENT_SCRL, &win_style);
 //   lv_win_set_style(win, LV_WIN_STYLE_BG, &win_style);
 //   lv_win_set_btn_size(win, LV_DPI / 3);

//	lv_obj_align(win,scr,LV_ALIGN_CENTER,0,0);

    lv_obj_set_size(win, 600, 480);

    static lv_style_t header;
    lv_style_copy(&header, &lv_style_pretty);
    header.body.main_color = LV_COLOR_HEX(0x524752);
    header.body.grad_color = LV_COLOR_HEX(0x4d4d4d);
    header.body.padding.inner = 0;
    header.text.color = LV_COLOR_WHITE;
    header.text.font = &MYFONT_24;

    lv_win_set_title(win, "主菜单");
    lv_win_set_style(win, LV_WIN_STYLE_HEADER, &header);

	lv_win_add_btn(win, SYMBOL_CLOSE, lv_win_close_action);
    lv_obj_set_pos(win, (LV_HOR_RES-600)/2, (LV_VER_RES-480)/2);
    lv_obj_set_top(win, false);

	static lv_style_t btn_style_rel,btn_style_pr;
	lv_style_copy(&btn_style_rel,&lv_style_transp);
	lv_style_copy(&btn_style_pr,&lv_style_plain_color);
	btn_style_pr.body.opa = 80;
	
	
	
	static lv_style_t label_style1; 
	lv_style_copy(&label_style1, &lv_style_pretty_color); 
	label_style1.text.font = &MYFONT_24; /*Set the base font whcih is concatenated with the others*/ 	 
//	label_style1.text.color = LV_COLOR_RED;

	for(j=0;j<2;j++)
	{
		for(i=0;i<3;i++)
		{
			img[i+j*3] = lv_img_create(win, NULL);
			lv_obj_set_size(img[i+j*3],104,104);
			
			lv_obj_set_pos(img[i+j*3], distance_scr+ img_width*i+distance_obj*i,distance_scr+ img_width*j+distance_obj*j);
            printf("after set pos x:%d,y:%d\n",lv_obj_get_x(img[i+j*3]),lv_obj_get_y(img[i+j*3]));

			btn[i+j*3] = lv_btn_create(win, NULL);
			lv_obj_set_size(btn[i+j*3],104+15,104+15);
			lv_obj_align(btn[i+j*3],img[i+j*3],LV_ALIGN_CENTER,0,0);
			lv_btn_set_style(btn[i+j*3],LV_BTN_STYLE_REL,&btn_style_rel);
			lv_btn_set_style(btn[i+j*3],LV_BTN_STYLE_PR,&btn_style_pr);
			
			label[i+j*3] = lv_label_create(win, NULL);
			lv_label_set_style(label[i+j*3], &label_style1); 
			lv_label_set_text(label[i+j*3] , menu_name[i+j*3]);/*Use ASCII and Cyrillic letters together*/
			lv_obj_align(label[i+j*3],btn[i+j*3],LV_ALIGN_OUT_BOTTOM_MID,0,0);

		}
	}

	lv_img_set_src(img[0], &img_search);
	lv_img_set_src(img[1], &img_info);
	lv_img_set_src(img[2], &img_config);
	lv_img_set_src(img[3], &img_advanced);
	lv_img_set_src(img[4], &img_backup);
	lv_img_set_src(img[5], &img_exit);

	lv_btn_set_action(btn[0],LV_BTN_ACTION_CLICK,btn_setting_action);
	lv_btn_set_action(btn[1],LV_BTN_ACTION_CLICK,btn_sd_action);
	lv_btn_set_action(btn[2],LV_BTN_ACTION_CLICK,btn_alarm_action);
	
    return LV_RES_OK;

}

turoksama, 2018-11-24:
I think you should check the style of that button.

TangoZhu, 2018-11-24:
> I think you should check the style of that button.

when i change the     win_style.body.padding.ver = LV_DPI / 4; the first button align will change,
if i dont call lv_win_set_style(win, LV_WIN_STYLE_CONTENT_SCRL, &win_style); the first button will align more obvious but the other button are align 
[image]

kisvegabor, 2018-11-25:
It's a little bit tricky thing. You used `lv_style_transp` as the button's released style. The built in transparent styles have `glass = 1` property which means those styles can't be inherited. What does it mean?
1.  `lv_style_transp.text.font` by default knows the close symbol but it won't be inherited
2. Instead, the header's style will be used for the close symbol which font is `MYFONT_24` which don't know the close symbol.

What can you do?
The most simple thing is to create a custom style for the button from `lv_style_plain`  and set `rel_style.body.opa = LV_OPA_TRANSP`

TangoZhu, 2018-11-26:
> It's a little bit tricky thing. You used `lv_style_transp` as the button's released style. The built in transparent styles have `glass = 1` property which means those styles can't be inherited. What [...]

OK ,I Solve my Frist problem 3Q
at lv_win_set_style function I add some code to set title’s style

        case LV_WIN_STYLE_TITLE:
            lv_label_set_style(ext->title, style); 
            lv_win_realign(win);
            break;

@kisvegabor  ### And  What about my second problem
The First Button Is not align
[image]

kisvegabor, 2018-11-27:
Have a look a [this](http://doc.littlevgl.com/#Page) to see how the positioning is working on a page. (the Notes section)

Or you can use `lv_win_set_layout(win, LV_LAYOUT_PRETTY);` instead of positioning the items manually.

TangoZhu, 2018-11-28:
> Have a look a [this](http://doc.littlevgl.com/#Page) to see how the positioning is working on a page. (the Notes section)
> 
> Or you can use `lv_win_set_layout(win, LV_LAYOUT_PRETTY);` instead of positioning the items manually.

according to the (the Notes section) i add the following code   
  lv_win_ext_t * ext = lv_obj_get_ext_attr(win);
    lv_page_set_scrl_fit(ext->page, false, false);
    lv_page_set_scrl_width(ext->page,580);
    lv_page_set_scrl_height(ext->page, 400);
and the align is ok
[image]
but if i call lv_win_set_layout(win, LV_LAYOUT_PRETTY); the display is in a mess 

[image]
the following is my code
`lv_res_t main_menu_create(lv_obj_t *activebtn)
{

    lv_obj_t * scr = lv_scr_act();

	int16_t distance_scr =64;//鍥炬爣涓庡睆骞曡竟妗嗙殑闂磋窛
	int16_t distance_obj=64;//鍥炬爣涔嬮棿鐨勯棿璺?
	int16_t img_width = 104;
    uint8_t j,i;
    
	//SCREEN
	lv_obj_t * btn[6];
	lv_obj_t * img[6] ;
	lv_obj_t * label[6] ;

    close_former_win(activebtn);

	lv_obj_t * win = lv_win_create(scr,NULL);

	static lv_style_t win_style;
    lv_style_copy(&win_style, &lv_style_plain);
    win_style.body.padding.hor = LV_DPI / 4;
    win_style.body.padding.ver = LV_DPI / 4;
    win_style.body.padding.inner = LV_DPI / 4;
    win_style.body.opa = LV_OPA_60;
    win_style.body.main_color = LV_COLOR_ORANGE;
    lv_win_set_style(win, LV_WIN_STYLE_CONTENT_SCRL, &win_style);

    lv_obj_set_size(win, 600, 480);

    lv_win_set_title(win, "涓昏彍鍗?);

	lv_win_add_btn(win, SYMBOL_CLOSE, lv_win_close_action);
    lv_obj_set_pos(win, (LV_HOR_RES-600)/2, (LV_VER_RES-480)/2);
    lv_obj_set_top(win, true);

	static lv_style_t btn_style_rel,btn_style_pr;
	lv_style_copy(&btn_style_rel,&lv_style_transp);
	lv_style_copy(&btn_style_pr,&lv_style_btn_pr);
	btn_style_pr.body.opa = LV_OPA_40;
	
	
	
	static lv_style_t label_style1; 
	lv_style_copy(&label_style1, &lv_style_pretty_color); 
	label_style1.text.font = &MYFONT_24; /*Set the base font whcih is concatenated with the others*/ 	 
    lv_win_set_style(win, LV_WIN_STYLE_TITLE, &label_style1);
    lv_win_set_title_align(win, LV_ALIGN_CENTER, 0,0);

   // lv_win_ext_t * ext = lv_obj_get_ext_attr(win);
  //  lv_page_set_scrl_fit(ext->page, false, false);
  //  lv_page_set_scrl_width(ext->page,580);
  //  lv_page_set_scrl_height(ext->page, 400);

    lv_win_set_layout(win, LV_LAYOUT_PRETTY);

	for(j=0;j<2;j++)
	{
		for(i=0;i<3;i++)
		{
			img[i+j*3] = lv_img_create(win, NULL);
			lv_obj_set_size(img[i+j*3],104,104);
	//		lv_obj_set_pos(img[i+j*3], distance_scr+ img_width*i+distance_obj*i,distance_scr+ img_width*j+distance_obj*j);
            //printf("after set pos x:%d,y:%d\n",lv_obj_get_x(img[i+j*3]),lv_obj_get_y(img[i+j*3]));
			btn[i+j*3] = lv_btn_create(win, NULL);
			lv_obj_set_size(btn[i+j*3],104+15,104+15);

			lv_obj_align(btn[i+j*3],img[i+j*3],LV_ALIGN_CENTER,0,0);
			lv_btn_set_style(btn[i+j*3],LV_BTN_STYLE_REL,&btn_style_rel);
			lv_btn_set_style(btn[i+j*3],LV_BTN_STYLE_PR,&btn_style_pr);
			
			label[i+j*3] = lv_label_create(win, NULL);
			lv_label_set_style(label[i+j*3], &label_style1); 
			lv_label_set_text(label[i+j*3] , menu_name[i+j*3]);
			lv_obj_align(label[i+j*3],btn[i+j*3],LV_ALIGN_OUT_BOTTOM_MID,0,0);

		}
	}

	lv_img_set_src(img[0], &img_search);
	lv_img_set_src(img[1], &img_info);
	lv_img_set_src(img[2], &img_config);
	lv_img_set_src(img[3], &img_advanced);
	lv_img_set_src(img[4], &img_backup);
	lv_img_set_src(img[5], &img_exit);

	lv_btn_set_action(btn[0],LV_BTN_ACTION_CLICK,btn_setting_action);
	lv_btn_set_action(btn[1],LV_BTN_ACTION_CLICK,btn_info_action);
	lv_btn_set_action(btn[2],LV_BTN_ACTION_CLICK,btn_alarm_action);
	lv_btn_set_action(btn[3],LV_BTN_ACTION_CLICK,btn_setting_action);
	lv_btn_set_action(btn[4],LV_BTN_ACTION_CLICK,btn_sd_action);
	lv_btn_set_action(btn[5],LV_BTN_ACTION_CLICK,btn_exit_action);
    return LV_RES_OK;

}`

fallstool, 2018-11-28:
Hi @TangoZhu 
I encounter the same problem, if you want to layout the object in your setted the position when in a page container, the only way is set the function **lv_page_set_scrl_fit(ext->page, false, false);**.

kisvegabor, 2018-11-29:
To use a layout you need to add the realted obejct (icon, name) to a common parent. Now the layout is applied to the icon and name too. 
I suggest using `lv_page_set_scrl_fit(page, false, false);.` as you already did.

TangoZhu, 2018-11-30:
3Q   littlevgl is a great project

kisvegabor, 2018-11-30:
Thank you! :)

It seems your issue is solved so I close it.

closed 2018-11-30 by kisvegabor