lvgl/lvgl #582: [Bug]Calendar action.

turoksama, 2018-11-24:
Hi,
 I've noticed that when calendar created and will act at once under the following conditions:
 I used a button to show(create) calendar, and the calendar's action configures DATE.
 When I touch that button, the calendar will show up, but if its position is under the finger and an action(if defined) will be mis-trigged at once whatever it is (LV_CALENDAR_ACTION_CLICK or LV_CALENDAR_ACTION_PR).
```
//calendar action callback
lv_res_t cfg_sp_dtm_cb (lv_obj_t *calendar)
{
	lv_calendar_date_t *date_sel = lv_calendar_get_pressed_date(calendar);
	lv_calendar_set_today_date( calendar, date_sel);
}
//btn action callback
lv_res_t cfg_sp_btn_dtm_cb (lv_obj_t *btn)
{
	lv_obj_t *date = lv_calendar_create(scr, NULL);
	lv_calendar_set_day_names(date, day_names_char);
	lv_calendar_set_month_names(date, month_names_char);
	lv_obj_set_size(date,440,400);
	lv_obj_align(date,scr,LV_ALIGN_CENTER, 0, 0);
	lv_calendar_set_action( date, LV_CALENDAR_ACTION_PR, cfg_sp_dtm_cb);
}
```

kisvegabor, 2018-11-25:
Hi,
In `cfg_sp_btn_dtm_cb` try: `lv_indev_wait_release(lv_indev_get_act())`.
The indev will do nothing until the next release. It is designed especially for cases like this.

turoksama, 2018-11-25:
Yes, exactly what i am thinking.

kisvegabor, 2018-11-25:
Great! :)

If it's working please close the issue!

closed 2018-11-26 by turoksama