lvgl/lvgl #589: Is three any way to get the top level window

TangoZhu, 2018-11-27:
AT my application i will create multi-level window,is three any way that i can get the top level windows,and then i can just call one function to get the top windows obj and close it

BrianPugh, 2018-11-27:
In my project (JoltOS), I use groups and button-navigation. I create all my windows as separate children of the lv_scr_act() object, and all of the elements inside of a window are children of the window. In my application, the back button closes a window, but I didn't want to close the primary menu. The following function is probably close to what you want.

```
lv_action_t jolt_gui_scr_del() {
    /* jolt_gui_store.group.main is the lvgl group handle */
    lv_obj_t *scrn = lv_group_get_focused(jolt_gui_store.group.main);
    if( NULL == scrn ) {
        MSG("Nothing in focus\n");
        return 0;
    }
    /* jolt_gui_store.main_menu is a lv_obj* to my main menu */
    if( scrn == jolt_gui_store.main_menu ) {
        MSG("Can't exit main menu\n");
        return 0;
    }

    // Disable any focus callback
    lv_group_set_focus_cb(jolt_gui_store.group.main, NULL);
    lv_obj_t *parent = scrn;
    lv_obj_t *tmp = scrn;
    while( (tmp = lv_obj_get_parent(tmp)) ) {
        if( tmp != lv_scr_act() ) {
            parent = tmp;
        }
        if( tmp == jolt_gui_store.main_menu ) {
            return 0;
        }
    }
    MSG("deleting %p\n", parent);
    lv_obj_del(parent);
    return 0;
}
```

TangoZhu, 2018-11-28:
3Q
your "the back button closes a window"give me an idea,maybe i can just use right mouse to close window,and i use a global point which point to the top window,when click right mouse just close the global point

kisvegabor, 2018-11-28:
You can use `lv_obj_get_child()` and  `lv_obj_get_child_back()` functions to get the children of a parent. They will give the children according to the current order.

embeddedt, 2018-12-18:
There was no activity here for 1 month so I assume you managed to solve this issue. If you continue to have problems please reopen the issue.

closed 2018-12-18 by embeddedt